mu-editor/mu #1053: ESP32 MicroPython Library installer

KKawase0104, 2020-06-02:
I wanted to make it easier to install the MicroPython library (^^

d2fd587349e3a2bb032668819d2e51f86d01d5dc

[image]

dybber, 2020-06-10:
Great idea, I often need all my students to install one library on their ESP Device, but it's a hurdle to first get them all online to be able to run upip.install(). I usually end up installing it for them, before I hand out the devices.

Some immediate feedback: 
 - perhaps it would be better to name it "MicroPython packages", just to avoid confusion with the other tab named "Third Party Packages"?
 - Why does the button say "Connection test" and not "Connect"?
 - Why does the button say "Start"? Could it say "Update and install" or similar?

I will return with more feedback when I've tested it

dybber, 2020-09-18:
@KKawase0104, could you send this in as a pull-request instead?

/Martin

KKawase0104, 2020-09-27:
@dybber 
Thank you for your comment.
I will rebase and pull-request.