mu-editor/mu #1055: Tab height on fresh start vs restored session

carlosperate, 2020-06-09:
Whenever Mu starts up (in macOS Mojave and Mu from master commit e3b7191e223a02fbff296fa0ecc834382e16e2b4) and re-opens tabs saved from the last session, it configures the tab height to be larger than they used to be.

I can't remember if this was changed on purpose (I think i like the larger tabs better), but the tab height is not the same in these two situations:
- When no files are opened and a new tab is opened (this is the default state when Mu is installed, or when all tabs are closed)
- When mu re-open tabs from the previous session

[image]

[image]

```
$ pip list
Package                       Version    
----------------------------- ----------
alabaster                     0.7.12
appdirs                       1.4.4
appnope                       0.1.0
arrow                         0.15.6
attrs                         19.3.0
Babel                         2.8.0
backcall                      0.1.0
binaryornot                   0.4.4
biplist                       1.0.3
black                         19.10b0
bleach                        3.1.5
boto3                         1.13.24
botocore                      1.16.24
briefcase                     0.2.9
certifi                       2020.4.5.2
chardet                       3.0.4
click                         7.1.2
cookiecutter                  1.7.2
coverage                      5.1
decorator                     4.4.2
dmgbuild                      1.3.3
docutils                      0.16
ds-store                      1.1.2
flake8                        3.8.2
Flask                         1.0.2
idna                          2.9
imagesize                     1.2.0
importlib-metadata            1.6.1
ipykernel                     5.3.0
ipython                       7.15.0
ipython-genutils              0.2.0
itsdangerous                  1.1.0
jedi                          0.17.0
Jinja2                        2.11.2
jinja2-time                   0.2.0
jmespath                      0.10.0
jupyter-client                6.1.3
jupyter-core                  4.6.3
keyring                       21.2.1
mac-alias                     2.0.7
MarkupSafe                    1.1.1
mccabe                        0.6.1
more-itertools                8.3.0
mu-editor                     1.1.0a2
nudatus                       0.0.4
numpy                         1.18.5
packaging                     20.4
parso                         0.7.0
pathspec                      0.8.0
pexpect                       4.8.0
pgzero                        1.2
pickleshare                   0.7.5
pip                           20.1.1
pkginfo                       1.5.0.1
pluggy                        0.13.1
poyo                          0.5.0
prompt-toolkit                3.0.5
ptyprocess                    0.6.0
py                            1.8.1
pycodestyle                   2.5.0
pyflakes                      2.1.1
pygame                        1.9.6
Pygments                      2.6.1
pyparsing                     2.4.7
PyQt5                         5.12.1
PyQt5-sip                     4.19.19
PyQtChart                     5.12.0
pyserial                      3.4
pytest                        5.4.3
pytest-cov                    2.9.0
pytest-faulthandler           2.0.1
pytest-random-order           1.0.4
python-dateutil               2.8.1
python-slugify                4.0.0
pytz                          2020.1
pyzmq                         19.0.1
QScintilla                    2.11.1
qtconsole                     4.4.3
readme-renderer               26.0
regex                         2020.6.8
requests                      2.23.0
requests-toolbelt             0.9.1
s3transfer                    0.3.3
semver                        2.10.1
setuptools                    47.1.1
six                           1.15.0
snowballstemmer               2.0.0
Sphinx                        3.0.4
sphinxcontrib-applehelp       1.0.2
sphinxcontrib-devhelp         1.0.2
sphinxcontrib-htmlhelp        1.0.3
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.3
sphinxcontrib-serializinghtml 1.1.4
text-unidecode                1.3
toml                          0.10.1
tornado                       6.0.4
tqdm                          4.46.1
traitlets                     4.3.3
twine                         3.1.1
typed-ast                     1.4.1
urllib3                       1.23
voc                           0.1.6
wcwidth                       0.2.4
webencodings                  0.5.1
Werkzeug                      1.0.1
wheel                         0.34.2
zipp                          3.1.0
```

dybber, 2020-06-10:
I can reproduce this. The error doesn't occur in the latest alpha, so I did a git bisect, which shows it's this commit that introduced the problem:
https://github.com/mu-editor/mu/commit/57d8e974ee266e976241521c27d06602e5183aaa

carlosperate, 2020-06-10:
Thanks @dybber, looks like this was introduced with PR https://github.com/mu-editor/mu/pull/899 then.
Did you replicate in macOS as well with the same PyQt versions? Or a different platform and/or versions? Might be a bug in PyQt.

dybber, 2020-06-10:
Same platform, Mac OS X Mojave, PyQt 5.12.1, so not that much new info I'm bringing.

dybber, 2020-06-11:
Found a way to fix this, see details in the pull request. My best guess is that it is due to some Qt oddity. However, it could be fixed be reordering when set_theme was called.

tjguk, 2020-06-12:
Fixed by #1062

closed 2020-06-12 by tjguk

carlosperate, 2020-06-14:
Great, thanks @dybber and @tjguk!