mu-editor/mu #1060: Devise some means of alerting before UI starts

tjguk, 2020-06-11:
As things stand a crashing bug in any piece of code which runs before the UI becomes available essentially looks like: I clicked on Mu and nothing happened.

Ideally we'd have -- at the least -- some kind of alert to indicate that an error occurred and to highlight the location of the log file

tjguk, 2020-06-11:
There is a catch-all solution referred to in https://github.com/mu-editor/mu/issues/1061#issuecomment-642665841

which writes a log file from the launcher into %APPDATA%

tmontes, 2020-06-14:
Agree very much. Linking #964, addressing more generic "crash scenarios", where I share an fun/odd experiment. :)

carlosperate, 2020-06-14:
>Ideally we'd have -- at the least -- some kind of alert to indicate that an error occurred and to highlight the location of the log file

I'm sure you have already consider this, but I just wanted to mention that, if I remember correctly, a Windows executable has to define itself as a console or GUI app, and when launching as a GUI I think it's unable to print to console.
So to be able to show some info to the user on Windows we might have to do a GUI for error reporting, as suggested by @tmontes.

tjguk, 2020-06-15:
Thanks all who've commented on this. My focus has been on helping people -- and, indeed, myself when testing -- who are running from the installed version on Windows where basically "Nothing happens" is the only symptom you get.

In that situation, the pynsist mechanism does actually log into %APPDATA%\<name-of-launcher>.log which gives us, on Windows, a basic fallback. Even if we have to instruct someone: "Start > Run > %APPDATA% and then look for a file called `mu.launch.pyw.log` and email that file" or whatever.

But (a) that's only on Windows via the installer; (b) it's clunky at best. While I quite like the kind of thing @tmontes demo-ed over in #964, I'm reluctant to add yet another moving part. I don't think we need to rush into a solution for this. In terms of Milestone 1.1 -- which we're planning towards now -- I regard this is a "could have" rather than a "must have" or even a "should have".

tmontes, 2020-06-16:
> But (a) that's only on Windows via the installer; (b) it's clunky at best. While I quite like the kind of thing @tmontes demo-ed over in #964, I'm reluctant to add yet another moving part. I don't [...]

Agreed. Thinking it through, the kind of (non-trivial) solution I demo-ed only works *after* Mu successfully starts. Contrarily, I suppose common "I clicked and nothing happens" scenarios mean Mu did not start successfully/at all!

I'm not sure how Mu can tell users "something's wrong" when itself cannot start. Unless, of course, we bring in (yet!) another moving/fragile piece of code that could act as a launcher/monitor of some kind - it would need to be very very simple (so as to have as few moving/failing parts as possible), launch Mu and, when/if Mu exits unsuccessfully, display a minimal UI/dialog box (maybe Tk based?) guiding the user to the log file.

Other than that, I suppose all bets are off... (very non trivial problem this one) Thoughts.