mu-editor/mu #1066: Upgrade briefcase to 0.3.0

dybber, 2020-06-13:
Briefcase have been revamped, to be more of an independent tool, rather than a plugin for setuptools. It would probably make sense for us to upgrade, here's some instructions for how to upgrade:
https://briefcase.readthedocs.io/en/latest/how-to/upgrade-from-v0.2.html

It seems this will require some effort, e.g. dependencies now also has to be described in a `pyproject.toml` file, and perhaps we could then rework setup to read the dependencies from that file, or perhaps more of setup.py can be moved to the `pyproject.toml` file, and setup.py simplified

tmontes, 2020-06-14:
(added this note on PR #1067, but should have shared it here, I suppose -- here's some duplication, for the sake of "having the right things in the right place"!) :)

Hey @dybber, I see you're pushing packaging things forward... Nice!

WRT to `briefcase`, you may want to read through my recent 0.3 exploration, something I did when I joined the `briefcase` PyCon US sprint a while back. Here are a few issues I raised which are directly extracted from the experience of packaging Mu with it:

* Fails if dependencies include .pth files | https://github.com/beeware/briefcase/issues/381
* User installable MSI packages? | https://github.com/beeware/briefcase/issues/382
* Packaged Standard Library missing modules | https://github.com/beeware/briefcase/issues/383
* Windows MSI - Append version number to name? | https://github.com/beeware/briefcase/issues/385

While most of these issues are Windows-related, some will apply to macOS too, I suspect. Regardless of that, my vision would be to have `briefcase` address all of Mu's application packaging needs across the supported platforms.

This could simplify the whole packaging process substantially (once it works!), let us keep a single source of truth WRT dependences (all in `pyproject.toml`, no more `setup.*`, no more `pynsist`, no more extra building/packaging scripts to maintain, etc).

Thoughts?

dybber, 2020-06-14:
Hi @tmontes, thanks for the input and pointers to your write ups. I hadn't discovered that briefcase doesn't bring in pip, venv, and tkinter through its support package. This also affects macOS packaging, and e.g. makes the "Third Party Packages" tab unusable.

I agree the end goal should be to package for all platforms using briefcase, but as I see it now, it will not happen right now, due to the issues you've found in briefcase itself. My suggestion before reading through your writeups, would have been to start switching to briefcase 0.3 for macOS packaging now, and then over time also switch over to use it for the other platforms.

I still think the right approach is to switch over one platform at a time, since I guess none of us are experts on all three platforms. For example, I haven't used Windows for around 15 years. I will be able to help with the Mac and Linux part, but the Windows packaging issues I'm not comfortable with. This will of course mean that in that transition period, we would need to have both pyproject.toml and setup.* for packaging, which is why I was concerned with how to read pyproject.toml from setup.py: https://stackoverflow.com/questions/62362693/

dybber, 2020-06-14:
We might be able to solve the issue with the support package by specifying our own, however, it seems that briefcase 0.2.9 and briefcase 0.3.* excepts the different directory structures within the support_package, so we can't just use the one we currently use for 0.2.9, we will have to rearrange its directory structure.

carlosperate, 2020-06-14:
I would also love if briefcase could be be single packaging resource we use (also because I am a big fan of pyproject.toml). As you have pointed out, I fear we are not quite there yet, but we don't know until we try so thanks @dybber and @tmontes for all the groundwork.

> I still think the right approach is to switch over one platform at a time...

This definitely sounds like the best approach, at the moment PyNsist is still working for us for Windows (I'm making a somewhere uninformed claim here, @tjguk would be able to confirm this or not), and as we are not yet creating Linux builds (that would be a different discussion I'd rather not include in this thread to keep the scope limited), then we can continue using the latest briefcase for the macOS builds.

> This will of course mean that in that transition period, we would need to have both pyproject.toml and setup.* for packaging, which is why I was concerned with how to read pyproject.toml from setup.py: https://stackoverflow.com/questions/62362693/

Just to clarify, this is only because we'd like to have a "single source of truth", right? Not because PyNsist needs or uses setup.py?

I think we can still generate (and gitignore) a simple pyproject.toml that uses setup.py (so having doesn't affect normal packaging installation), and thankfully as the toml file forma is very simple (and the available packages are small as well) so we can have a small script like the windows one to add the dependencies to that.

> We might be able to solve the issue with the support package by specifying our own, however, it seems that briefcase 0.2.9 and briefcase 0.3.* excepts the different directory structures within the [...]

Not sure I completely understand this, what directory structure? The repository directory structure?

carlosperate, 2020-06-14:
Oh right, this support package? https://github.com/mu-editor/mu_portable_python_macos
In that case I think we want to go back to the briefcase "official" support packages, as they deal with statically linking OpenSSL correctly. Last time I looked at this there were patches applied to the Python source code to achieve this, and I'd rather not go that route if we can contribute and use the briefcase ones.

dybber, 2020-06-14:
Just a few comments now, will probably have more to say on this tomorrow:

 - Yes it was only to keep a single source of truth, and I've also dabbled with the idea of generating pyproject.toml, even though that is against the intention of the pyproject.toml file

 - Thanks for the link to the support package repo , yes, I believe it's that (see my comment at #1067), however, as @tmontes have identified, the briefcase official support packages, doesn't bring in pip, venv, tkinter (perhaps others), which means that for instance the Third Party packages tab will not work due to lack of pip

dybber, 2020-06-15:
I've committed a solution to the "single source of truth" issue https://github.com/mu-editor/mu/pull/1067/commits/9a43c5dd46491088fbc2aaa81bef203cbf28e4bd, which makes setup.py read pyproject.toml. Perhaps we could work towards moving more stuff from setup.py to pyproject.toml (e.g. development dependencies as well), what do you think about that solution?

To get started working on the issue on support_package, I have also added a commit that reintroduces that non-working support package https://github.com/mu-editor/mu/pull/1067/commits/1b9ec3f67b86a3c31805d8162f4897e769f2f714, to make it visible what's wrong with the directory layout. Anyone up for looking at that part?

dybber, 2020-06-15:
The error I'm getting: `line 5: Support/bin/python3: No such file or directory`, which seems caused by the directory structure in the support package has an extra level:
```
Support/
├── README
└── python
    ├── bin
    ├── include
    ├── lib
    ├── share
    └── version.txt
```

First attempt of how to fix this by doing a `mv Support/python/* Support/`, to get this directory structure:
```
Support/
├── README
├── bin
├── include
├── lib
├── python
├── share
└── version.txt
```

After that, I'm then getting this error:
```
Traceback (most recent call last):
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 183, in _run_module_as_main
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 142, in _get_module_details
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 109, in _get_module_details
  File "/Users/dpr964/Development/mu/macOS/mu-editor/mu-editor.app/Contents/Resources/app/mu/__init__.py", line 4, in <module>
    from PyQt5.QtCore import QLocale
ModuleNotFoundError: No module named 'PyQt5.sip'
```

dybber, 2020-06-15:
Could this part just be caused by PyQt5 version was updated, and that the support package somehow should be updated as well? 

But that just doesn't seem right, as the version of PyQt5 and support package we use now, works fine with briefcase 0.2.9

tmontes, 2020-06-16:
Sharing a somewhat tangential note regarding `briefcase`'s support package:

* As you have noted, @dybber, it does not include a few Mu required things.
* Then, Mu's custom support package for `briefcase` 0.2.x is, unfortunatelly, in an incompatible layout.
* Additionally, IIRC it has been identified that Mu's custom support package is still linking to non system things, like `brew`'s openssl or some such.

What I tried successfully, in my previous exploration:

* Grabbed one of the very nice [Python Standalone Builds from here](https://github.com/indygreg/python-build-standalone/releases)...
* ...unpackaged and repackaged it into the `briefcase`'s support package expected layout...
* ...while trimming it down along the way (IIRC I brought it down to ~30 MB or so).

From https://python-build-standalone.readthedocs.io/en/latest/:

> This project produces self-contained, highly-portable Python distributions. These Python distributions contain a fully-usable, full-featured Python installation as well as their build artifacts (object files, libraries, etc).

Maybe we could build Mu's updated support package for `briefcase` based on that, instead of reinventing that wheel... :) (and then, who knows, follow the same model for a future `briefcase` packaged Mu Windows distribution)

dybber, 2020-06-16:
Good find. I think for Mu, the simplest solution, would then be if briefcase supported using that Python standalone distribution as an alternative support package. That is, it would be nicer to fix the issue on the briefcase side, rather than on a hack that would have to be maintained and update everytime we want to do a Mu release.

tmontes, 2020-06-16:
> Good find. I think for Mu, the simplest solution, would then be if briefcase supported using that Python standalone distribution as an alternative support package. That is, it would be nicer to fix [...]

...agreed. I have shared that idea with @freakboy3742, in at one of the issues I linked to, above. Actually, I'd rather focus all (most?) packaging efforts on `briefcase` and have Mu "just" depend on that. :)

dybber, 2020-06-16:
Great! I just read through the thread where you suggested it (beeware/briefcase#383). We could try go about it the following way:
 - suggest to the dev(s) behind the Python Standalone Builds, that we reduce the size of the image,  or they also release "stripped" editions. E.g. they currently include 22MB of test-cases, and also some build scripts, etc., which I guess could be removed  for most use cases
 - suggest to the briefcase dev's to support using those standalone builds

freakboy3742, 2020-06-17:
@dybber For the record: happy to help however we can to support any of these goals. 

Putting another option on the table that may help address the size issue: we have been [considering the possibility of shipping a "full" build, and then providing a mechanism to mark what you want to strip out as a post-filtering step](beeware/briefcase#413). That would allow us to include Tk in the support package, but then strip it out when it's not needed (e.g., for Toga builds). It would also allow apps to determine if they do/don't need test folders, specific standard library modules, or any other pieces of the support package.

dybber, 2020-06-17:
Great, and just adding another reason why Tk and `tkinter` is important for Mu: the `turtle` library uses `tkinter` for drawing, and `turtle` is quite popular for teaching Python.

Let us know if we can help in anyway.

tjguk, 2020-06-17:
From a Windows perspective: for the moment, I'm going to continue to focus on pynsist because that's a solution we have working right now and I don't want to jeopardise a 1.1 release. That said, I'd be very happy with a unified Briefcase-all-the-things approach in the future. It's just that I don't want to divert resource and/or add risk at this point.

Very happy to take part in discussions -- and even to have my mind changed :)

dybber, 2020-06-17:
@tjguk - that's also how I see it - see my previous comment at https://github.com/mu-editor/mu/issues/1066#issuecomment-643770836

dybber, 2021-01-26:
I will close this, as we're instead moving towards pup for packaging

closed 2021-01-26 by dybber