mu-editor/mu #1073: unable to import third party modules

hpm007, 2020-07-02:
i was trying to import pyperclip module into mu,which i am sure i installed with cmd.
later i tried importing other python modules (surely downloaded) which are present at C:\Users\admin\AppData\Local\Programs\Python\Python38\Lib\site-packages but none of them can be imported.
i tried copying the modules from site-packages to Lib(the location from where mu accesses it's modules)but to no avail. Already present modules work perfectly fine.
On the other hand these modules can be accessed with IDLE. so the only problem i can think of has to do with mu.
I am using mu 1.0.3 on windows 10 pro.
I was running mu with python 3 mode.
[image]

logs:
2020-07-02 10:29:23,105 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.3
2020-07-02 10:29:23,120 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-QPP2PUP', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 9, GenuineIntel')
2020-07-02 10:29:23,120 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\admin\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-07-02 10:29:23,120 - root:115(run) INFO: Language code: en_US
2020-07-02 10:29:24,171 - mu.logic:543(__init__) INFO: Setting up editor.
2020-07-02 10:29:24,171 - mu.logic:562(__init__) INFO: Settings path: C:\Users\admin\AppData\Local\python\mu\settings.json
2020-07-02 10:29:24,171 - mu.logic:563(__init__) INFO: Session path: C:\Users\admin\AppData\Local\python\mu\session.json
2020-07-02 10:29:24,171 - mu.logic:564(__init__) INFO: Log directory: C:\Users\admin\AppData\Local\python\mu\Logs
2020-07-02 10:29:24,171 - mu.logic:565(__init__) INFO: Data directory: C:\Users\admin\AppData\Local\python\mu
2020-07-02 10:29:24,256 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2020-07-02 10:29:29,111 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\admin\mu_code
2020-07-02 10:29:29,132 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\admin\AppData\Local\python\mu\session.json
2020-07-02 10:29:29,132 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['C:\\Users\\admin\\mu_code\\dictionaries.py', 'C:\\Users\\admin\\Manupulating strings.py', 'C:\\Users\\admin\\mu_code\\start.py', 'C:\\Users\\admin\\mu_code\\updating a dictionary.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2020-07-02 10:29:29,132 - mu.logic:715(_load) INFO: Loading script from: C:\Users\admin\mu_code\start.py
2020-07-02 10:29:29,142 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-07-02 10:29:29,143 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-07-02 10:29:29,143 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\r\n'
2020-07-02 10:29:29,143 - mu.logic:789(_load) DEBUG: import selenium
2020-07-02 10:29:29,489 - mu.logic:715(_load) INFO: Loading script from: C:\Users\admin\mu_code\dictionaries.py
2020-07-02 10:29:29,525 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-07-02 10:29:29,525 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-07-02 10:29:29,525 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2020-07-02 10:29:29,525 - mu.logic:789(_load) DEBUG: allGuests = {'Alice': {'apples': 5, 'pretzels': 12},
'Bob': {'ham sandwiches': 3, 'apples': 2},
'Carol': {'cups': 3, 'apple pies': 1}}
def totalBrought(guests, item):
    numBrought = 0
    for k, v in guests.items():
        numBrought = numBrought + v.get(item, 0)
    return numBrought
print('Number of things being brought:')
print(' - Apples         ' + str(totalBrought(allGuests, 'apples')))
print(' - Cups           ' + str(totalBrought(allGuests, 'cups')))
print(' - Cakes          ' + str(totalBrought(allGuests, 'cakes')))
print(' - Ham Sandwiches ' + str(totalBrought(allGuests, 'ham sandwiches')))
print(' - Apple Pies     ' + str(totalBrought(allGuests, 'apple pies')))
2020-07-02 10:29:29,592 - mu.logic:715(_load) INFO: Loading script from: C:\Users\admin\mu_code\updating a dictionary.py
2020-07-02 10:29:29,627 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-07-02 10:29:29,627 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-07-02 10:29:29,627 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\r\n'
2020-07-02 10:29:29,627 - mu.logic:789(_load) DEBUG: import docx
2020-07-02 10:29:29,677 - mu.logic:715(_load) INFO: Loading script from: C:\Users\admin\Manupulating strings.py
2020-07-02 10:29:29,691 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-07-02 10:29:29,691 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-07-02 10:29:29,691 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2020-07-02 10:29:29,691 - mu.logic:789(_load) DEBUG: def printPicnic(itemsDict, leftWidth, rightWidth):
    print('PICNIC ITEMS'.center(leftWidth + rightWidth, '-'))
    for k, v in itemsDict.items():
        print(k.ljust(leftWidth, '.') + str(v).rjust(rightWidth))

picnicItems = {'sandwiches': 4, 'apples': 12, 'cups': 4, 'cookies': 8000}
printPicnic(picnicItems, 12, 5)
printPicnic(picnicItems, 20, 6)
2020-07-02 10:29:29,709 - mu.logic:649(restore_session) INFO: Loaded files.
2020-07-02 10:29:29,709 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2020-07-02 10:29:29,709 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2020-07-02 10:29:29,749 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\admin\mu_code
[... 232 lines cut ...]
2020-07-02 11:17:34,076 - mu.logic:1178(autosave) INFO: Autosave detected and saved changes in C:/Users/admin/mu_code/start.py.
2020-07-02 11:17:39,030 - mu.logic:871(save_tab_to_file) INFO: Saving script to: C:/Users/admin/mu_code/start.py
2020-07-02 11:17:39,031 - mu.logic:872(save_tab_to_file) DEBUG: import pyperclip

2020-07-02 11:17:39,121 - mu.logic:1178(autosave) INFO: Autosave detected and saved changes in C:/Users/admin/mu_code/start.py.
2020-07-02 11:17:41,335 - mu.logic:871(save_tab_to_file) INFO: Saving script to: C:/Users/admin/mu_code/start.py
2020-07-02 11:17:41,335 - mu.logic:872(save_tab_to_file) DEBUG: import pyperclip

2020-07-02 11:17:42,270 - mu.modes.python3:202(stop_script) DEBUG: Stopping script.
2020-07-02 11:17:42,853 - mu.interface.panes:651(start_process) INFO: Running script: c:\users\admin\mu_code\start.py
2020-07-02 11:17:42,854 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2020-07-02 11:17:42,854 - mu.interface.panes:656(start_process) INFO: Command args: []
2020-07-02 11:17:42,855 - mu.interface.panes:683(start_process) INFO: Python paths set via C:\Users\admin\AppData\Roaming\Python\Python36\site-packages\mu.pth
2020-07-02 11:17:42,855 - mu.interface.panes:714(start_process) INFO: Working directory: C:/Users/admin/mu_code
2020-07-02 11:17:42,855 - mu.interface.panes:719(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\admin\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-07-02 11:23:01,641 - mu.modes.python3:202(stop_script) DEBUG: Stopping script.
2020-07-02 11:23:02,679 - mu.logic:1107(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2020-07-02 11:24:21,450 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\admin\AppData\Local\python\mu\Logs\mu.log
2020-07-02 11:24:41,965 - mu.interface.panes:651(start_process) INFO: Running script: c:\users\admin\mu_code\start.py
2020-07-02 11:24:41,965 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2020-07-02 11:24:41,965 - mu.interface.panes:656(start_process) INFO: Command args: []
2020-07-02 11:24:41,965 - mu.interface.panes:683(start_process) INFO: Python paths set via C:\Users\admin\AppData\Roaming\Python\Python36\site-packages\mu.pth
2020-07-02 11:24:41,969 - mu.interface.panes:714(start_process) INFO: Working directory: C:/Users/admin/mu_code
2020-07-02 11:24:41,970 - mu.interface.panes:719(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\admin\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-07-02 11:24:45,031 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\admin\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2020-07-25:
Hi @hpm007,
Mu runs it's own isolated copy of Python and packages, so it is not able to access packages installed using cmd.
If you install the latest alpha version of Mu there is a new option in the settings that allows you install 3rd party packages inside this isolated environment:

[image]

hpm007, 2020-07-26:
Thanks,the problem was that my creates it's own folder to keep site packages in the roaming directory. So I copied the relevant packages in roaming/python36/site packages

closed 2020-07-26 by hpm007