mu-editor/mu #1076: Strange behaviour on Windows - PygameZero mode + fonts + special polish letters...

abixadamj, 2020-07-14:
Hey, I wrote a book about Python https://kodujwpythonie.pl/ and I'm using mu-editor as editor in this book. The repository with code is public: https://github.com/kodujwpythonie/koduj-w-pythonie.

But there is strange thing under Windows with coding Polish special chars....
[image]
Screen is from youtube, where one of readers is trying to run it
https://youtu.be/77uTYgry3AE
Under Linux everythnig is OK.
[image]
I have asked about checking in Windows in "standard" pygamezero... user answered everything is OK (polish):
"Gra zadziała poprawnie poza Mu-editor. VSC, Python 3.8 i pgzero 1.2 zainstalowane w systemie"
Translation: "The game will work correctly outside of Mu-editor. VSC, Python 3.8 and pgzero 1.2 installed on the system"
I think there must be something in mu-editor, can somebody try this code in mu-editor in Windows English version?

ZanderBrown, 2020-07-14:
Congrats on discovering Windows support for unicode is interesting....

Could you elaborate on "checking in Windows in "standard" pygamezero."? Since it seems the problem is in the pygame window it's almost certainly a pygame issue not mu

Mu on Windows bundles all it's dependencies it's quite possible this has been fixed in newer pygame (hence works on mint)

imiolek-ireneusz, 2020-07-14:
I can confirm the issue exists under Windows in a non-polish locale (English UK). However, I do believe it is not an issue in any of the software mentioned here (mu, pygame, etc.) other than the Windows itself. The users of Windows using the system in a language other than Polish may need to be advised to either change the locale or enable UTF-8 support. The latter seems to be a beta feature but worked fine for me and the game gets rendered as expected after applying the change and restarting the system.

There may be a shorter way to get to the setting, but here's my route:
Settings -> Time & Language -> Region -> Additional data, time & regional settings -> Region -> Administrative tab -> Change system locale... -> Tick box - Beta: Use Unicode UTF-8 for worldwide language support.

tmontes, 2020-07-15:
Hey @abixadamj,

Short feedback:

* Reproduced your issue in a Windows 10 VM with English settings.
* Digged a little bit: the culprit is `pgzero`, same underlying issue *and fix* as #746.
* AFAICT: pending an updated release of `pgzero`, current 1.2 does not include the fix that has already been merged to `master` (see https://github.com/lordmauve/pgzero/issues/173).

Hacking it away:

* Edit `runner.py` in your virtual environment's `pgzero` install (or bundled Mu Python environment).
* Find the `with open` in line 76.
* Add `mode='rb'` as an additional argument to the `open` call.
* Your program should now run fine! :)

tmontes, 2020-07-15:
PS: One other thing I did was adding a [PEP 263](https://www.python.org/dev/peps/pep-0263/) encoding declaration at the top of your `program_10.py`: `# -*- coding: utf8 -*-`. Might be relevant/required. ;)

carlosperate, 2020-11-08:
Thanks for the report @abixadamj and for the amazing debugging @tmontes.
Since this issue is already captured in #746 I'll close this one, but please feel free to reopen if this is not the case.

closed 2020-11-08 by carlosperate