mu-editor/mu #1077: Checking code

BestdogShadow, 2020-07-16:
If you are reporting a bug, we would like to know:

* What you were trying to do, I was checking my code using the check feature
* What steps you took to make this happen, clicking the check feature
* What you expected to happen, to see what was wrong
* What actually happened, i could only see the start and the rest was off the screen making it impossible to read the text that follows the start
* Why this difference is problematic (it may not be a bug!), i cant see the full reason whats wrong with my code
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. i was running the latest version and i have windows

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.
[image]

Thank you for contributing to Mu! :-)

ZanderBrown, 2020-07-16:
Hello @BestdogShadow!

This is the place for reporting problems with Mu itself, but you seem to be having a problem with code you've written (other forums exist for this :-) )

But since you are here:

Your code looks so be something like so

```py
value = 0

def addtimetotimer():
    value = value + 1

addtimetotimer()
```

Now if you run it you get an error:
```
Traceback (most recent call last):
  File "1077.py", line 6, in <module>
    addtimetotimer()
  File "1077.py", line 4, in addtimetotimer
    value = value + 1
UnboundLocalError: local variable 'value' referenced before assignment
```

The problem is when you set `value` here, that's in the global scope
```py
value = 0
```

But here your using it in `addtimetotimer`'s local scope
```py
    value = value + 1
```

Now python can be a little funky, when using a variable it will look for it both locally and globally. However setting a variable only sets it in the local scope (unless it's been declared to be global) hence your problem. Python sees `value =` and knows your trying to set a local variable called `value` to something, it then moves on to the expression `value + 1`.

So now it must lookup the current value of `value`, but oops we haven't set one yet! our local `value` variable shadows the global one. Since "undefined" + 1 doesn't make any sense python gives up.

The solution is simple, mark `value` as being global:
```py
value = 0

def addtimetotimer():
    global value   # <====
    value = value + 1

addtimetotimer()
```

carlosperate, 2020-07-25:
Thank you @ZanderBrown for providing such a detailed answer to @BestdogShadow's code issue. This is why the Mu community is so awesome. ✨ 

As it's been 10 days and this is not a problem with Mu I'll close this issue.

closed 2020-07-25 by carlosperate