mu-editor/mu #1078: Ubuntu 18.04: Sometimes, Highlighting Code and Switching Window Removes Code from Mu

aaronistheman, 2020-07-20:
* What you were trying to do: highlight code in the Mu editor and then switch window.
* What happened: I highlighted the code, switched window, and then when I look back at the Mu editor, I see that the code I had highlighted has disappeared. This happens regardless of whether I switch windows by clicking or with Alt+Tab. The bug does not always occur; it seems more likely to occur if any of the following happen, but it might be coincidence: a) I try to copy/paste code first, b) I only highlight a part of a line instead of entire series of lines, c) the window that I switch to is a terminal. I am able to replicate this bug repeatedly and have encountered it on multiple different days.
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running: Mu 1.0.3, Ubuntu 18.04, Python 3.6.9.
* The related actions do not seem to appear in the log, i.e. highlighting code and switching window are not registered.

carlosperate, 2020-07-25:
Thanks for the report @aaronistheman. Does the selected text get replaced by anything? Like a space or tab? Could you also try the latest alpha version to see if the problem is still persistent there? https://github.com/mu-editor/mu/releases/tag/1.1.0-alpha.2

I haven't been able to replicate in v1.0.3 macOS Mojave, so if there are other Ubuntu users that could try to replicate this it would be quite useful.

abixadamj, 2020-07-28:
I can say - it is often in 1.03 version. I have this often. Linux Mint 19.3
I can reproduce  this even in 1.1.0_alpha2. - rare, but it happens.
Video on my Vimeo: https://vimeo.com/442310555/c495c8172d

sunnyyakima, 2020-08-09:
Hi
I tried to install mu 1.1.0 to my ubuntu 18 by:
python3 setup.py install --user
it seems mu-editor is installed. But I got an error when running it:
Traceback (most recent call last):
  File "/home/leo/.local/bin/mu-editor", line 11, in <module>
    load_entry_point('mu-editor==1.1.0a2', 'console_scripts', 'mu-editor')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/leo/.local/lib/python3.6/site-packages/mu_editor-1.1.0a2-py3.6.egg/mu/__init__.py", line 4, in <module>
    from PyQt5.QtCore import QLocale
ModuleNotFoundError: No module named 'PyQt5.sip'

Can anyone help me on this? Thanks

carlosperate, 2020-08-09:
Hi @sunnyyakima,
Thanks for the report, but it looks like this might be a different problem, would you mind opening a new issue to avoid mixing different conversations here?
In the meantime you can find installation instructions here: https://codewith.mu/en/howto/1.1/

carlosperate, 2020-08-09:
> I can say - it is often in 1.03 version. I have this often. Linux Mint 19.3
> I can reproduce this even in 1.1.0_alpha2. - rare, but it happens.
> Video on my Vimeo: https://vimeo.com/442310555/c495c8172d

Thanks for the video @abixadamj, that's really useful to visualize the issue.
One off-topic thing, I wasn't aware the Ctrl+alt+i shortcut opened IDLE and I don't have the same in macOS. I had a quick look at the source code and I am not sure where that comes from? It was a very quick scan though, so I probably didn't search hard enough. Do you get this issue with alt-tab as well? Or just with this IDLE shortcut?

abixadamj, 2020-08-10:
@carlosperate - ctrl+alt+i is my own shortcut to run IDLE. This issue is with alt-tab, because if I open IDLE with mouse (desktop icon) and then copy/alt+tab/paste - the same problem occurs (but sometimes, not every time).

carlosperate, 2021-08-04:
This was also replicated by user @ajoh504 in gitter minimising the window in Ubuntu 20.04:

> I'm using Mu on Ubuntu 20.04.1, and I've encountered a glitch where if I minimize Mu while all my code is still highlighted, it deletes all of the code. Has anyone encountered this problem? 

I was able to replicate in an ubuntu 20.04 VM with v1.1.0-beta.5 by simply changing window focus.

https://user-images.githubusercontent.com/4189262/128229651-9bc26c14-2f6c-4b0a-a848-410dc84e75e2.mp4

It seems like it happens only once until the file is edited again. So if a text is selected, the window focus changed, that erases the text, but it doesn't happen again until the file is edited:

@ntoll can you replicate this on Debian?

carlosperate, 2022-04-25:
Looks like this was an issue in QScintilla that will be fixed in the next release! 🎉 
https://www.riverbankcomputing.com/pipermail/qscintilla/2022-April/001539.html