mu-editor/mu #1079: make check error on Linux Mint 19.3 with 1.1.0_alpha2

abixadamj, 2020-07-28:
I am trying to make devel version as described https://mu.readthedocs.io/en/latest/.
When I run `make check`, an error occured:

```
FAILED tests/modes/test_python3.py::test_python_on_kernel_start - AssertionError: Expected call: show_status_message('REPL started.')
FAILED tests/modes/test_python3.py::test_python_on_kernel_stop - AssertionError: Expected call: show_status_message('REPL stopped.')
FAILED tests/modes/test_python3.py::test_python_run_toggle_on - AssertionError: Expected call: setText('Stop')
FAILED tests/modes/test_python3.py::test_python_run_toggle_off - AssertionError: Expected call: setText('Run')
================================================== 46 failed, 789 passed, 19 warnings in 11.81s ==================================================
Makefile:57: recipe for target 'coverage' failed
make: *** [coverage] Error 1

I have:
(mu_venv) adasiek@mint-desktop:~/python_projekty/mu$ coverage
Code coverage for Python, version 5.2.1 with C extension.  Use 'coverage help' for help.
Full documentation is at https://coverage.readthedocs.io
```
Any idea? Maybe fact, that I have pl_PL locales ?

carlosperate, 2020-09-27:
Thanks for the report @abixadamj, could you try to run `LANG=en_GB.utf8 make coverage` that way we could confirm if the issues you are seeing are related to the locale.

carlosperate, 2021-04-11:
Thanks for the report @abixadamj 

This has been fixed in https://github.com/mu-editor/mu/pull/1151, so I'll close this one,  but please do reopen it if it is not fully fixed.

closed 2021-04-11 by carlosperate