mu-editor/mu #1082: Issues with third-party modules in Mu

ximeng86, 2020-07-29:
Hi all,

I just installed version 1.1.0.alpha2 of the Mu editor and would like to start adding Python 3 packages to it, but I am not able to.
I tried to follow the instructions in Automate the boring stuff with Python, but that did not work, I tried to add the modules name in the tab "Third party packages" but then I got this message:

> pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
> Collecting pyperclip
>   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
>   Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
>   Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
>   Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
>   Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
>   Could not fetch URL https://pypi.python.org/simple/pyperclip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.python.org', port=443): Max retries exceeded with url: /simple/pyperclip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping
>   Could not find a version that satisfies the requirement pyperclip (from versions: )
> No matching distribution found for pyperclip
> 
> FINISHED

Since I am new to coding in the meanwhile I will have to stick with IDLE, but I would've loved to try MU

abixadamj, 2020-07-29:
Can you write down the content of packages tab? What packages did you wrote? Because I have tried 
`pyperclip` and:

```
Collecting pyperclip
  Downloading pyperclip-1.8.0.tar.gz (16 kB)
Building wheels for collected packages: pyperclip
  Building wheel for pyperclip (setup.py): started
  Building wheel for pyperclip (setup.py): finished with status 'done'
  Created wheel for pyperclip: filename=pyperclip-1.8.0-py3-none-any.whl size=8691 sha256=0fd2e1beeea89fe9e0e8635622e455bfda0b1f6ed79e3ca868bf81be18095c0a
  Stored in directory: /home/adasiek/.cache/pip/wheels/26/30/fe/92e2d4b1301ba74c07ea09c9e4c08f5bf12bae9c30319d74c5
Successfully built pyperclip
Installing collected packages: pyperclip
Successfully installed pyperclip-1.8.0

FINISHED
```
[image]

abixadamj, 2020-07-29:
Please provide log, for example:
```
2020-07-29 16:05:41,641 - mu.logic:771(restore_session) INFO: Loaded files.
2020-07-29 16:05:41,641 - mu.logic:776(restore_session) INFO: User defined environment variables: []
2020-07-29 16:05:41,641 - mu.logic:782(restore_session) INFO: Minify scripts on micro:bit? False
2020-07-29 16:05:41,665 - mu.logic:1383(change_mode) INFO: Workspace directory: /home/adasiek/mu_code
2020-07-29 16:05:59,528 - mu.logic:1336(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'debugger', 'pygamezero']
2020-07-29 16:06:02,301 - mu.logic:1342(select_mode) INFO: New mode selected: python
2020-07-29 16:06:02,333 - mu.logic:1383(change_mode) INFO: Workspace directory: /home/adasiek/mu_code
2020-07-29 16:06:04,123 - mu.logic:1270(show_admin) INFO: Showing admin with logs from /home/adasiek/.cache/mu/log/mu.log
2020-07-29 16:06:04,124 - mu.logic:200(installed_packages) INFO: Packages found: ['/home/adasiek/.local/share/mu/site-packages/websockets-8.1.dist-info', '/home/adasiek/.local/share/mu/site-packages/PyTechBrain-0.7.2.dist-info', '/home/adasiek/.local/share/mu/site-packages/pyserial-3.4.dist-info', '/home/adasiek/.local/share/mu/site-packages/pymata_aio-2.33.dist-info']
2020-07-29 16:06:31,183 - mu.logic:1318(sync_package_state) INFO: Synchronize package states...
2020-07-29 16:06:31,183 - mu.logic:1319(sync_package_state) INFO: Old: {'pyserial', 'pymata-aio', 'websockets', 'pytechbrain'}
2020-07-29 16:06:31,184 - mu.logic:1320(sync_package_state) INFO: New: {'pyperclip', 'pytechbrain', 'pyserial', 'websockets', 'pymata-aio'}
2020-07-29 16:06:31,184 - mu.logic:1324(sync_package_state) INFO: To add: {'pyperclip'}
2020-07-29 16:06:31,184 - mu.logic:1325(sync_package_state) INFO: To remove: set()
2020-07-29 16:06:31,184 - mu.logic:1326(sync_package_state) INFO: Site packages: /home/adasiek/.local/share/mu/site-packages
2020-07-29 16:06:31,190 - mu.interface.dialogs:503(run_pip) INFO: /home/adasiek/python_venvs/mu_venv/bin/python -m pip install pyperclip --target /home/adasiek/.local/share/mu/site-packages
2020-07-29 16:08:40,286 - mu.logic:1270(show_admin) INFO: Showing admin with logs from /home/adasiek/.cache/mu/log/mu.log
2020-07-29 16:08:40,286 - mu.logic:200(installed_packages) INFO: Packages found: ['/home/adasiek/.local/share/mu/site-packages/websockets-8.1.dist-info', '/home/adasiek/.local/share/mu/site-packages/pyperclip-1.8.0.dist-info', '/home/adasiek/.local/share/mu/site-packages/PyTechBrain-0.7.2.dist-info', '/home/adasiek/.local/share/mu/site-packages/pyserial-3.4.dist-info', '/home/adasiek/.local/share/mu/site-packages/pymata_aio-2.33.dist-info']

```
[image]

ximeng86, 2020-07-29:
Hi,

Here is what happens:

pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting pyperclip
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
Could not fetch URL https://pypi.python.org/simple/pyperclip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.python.org', port=443): Max retries exceeded with url: /simple/pyperclip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping
Could not find a version that satisfies the requirement pyperclip (from versions: )
No matching distribution found for pyperclip

FINISHED

abixadamj, 2020-07-29:
Log !!!!

ximeng86, 2020-07-29:
Which screenshot do you need?

Here is the log:

2020-07-29 17:15:31,316 - mu.logic:706(restore_session) INFO: Loaded files.
2020-07-29 17:15:31,316 - mu.logic:710(restore_session) INFO: User defined environment variables: []
2020-07-29 17:15:31,316 - mu.logic:714(restore_session) INFO: Minify scripts on micro:bit? False
2020-07-29 17:15:31,410 - mu.logic:1266(change_mode) INFO: Workspace directory: /Users/Simone/mu_code
2020-07-29 17:15:37,343 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/Simone/Library/Logs/mu/mu.log
2020-07-29 17:15:37,343 - mu.logic:163(installed_packages) INFO: Packages found: []
2020-07-29 17:16:11,149 - mu.logic:1192(show_admin) INFO: No admin settings changed.
2020-07-29 17:16:16,851 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/Simone/Library/Logs/mu/mu.log
2020-07-29 17:16:16,851 - mu.logic:163(installed_packages) INFO: Packages found: []
2020-07-29 17:16:25,466 - mu.logic:1203(sync_package_state) INFO: Synchronize package states...
2020-07-29 17:16:25,467 - mu.logic:1204(sync_package_state) INFO: Old: set()
2020-07-29 17:16:25,468 - mu.logic:1205(sync_package_state) INFO: New: {'pyperclip'}
2020-07-29 17:16:25,468 - mu.logic:1209(sync_package_state) INFO: To add: {'pyperclip'}
2020-07-29 17:16:25,468 - mu.logic:1210(sync_package_state) INFO: To remove: set()
2020-07-29 17:16:25,468 - mu.logic:1211(sync_package_state) INFO: Site packages: /Users/Simone/Library/Application Support/mu/site-packages
2020-07-29 17:16:25,470 - mu.interface.dialogs:456(run_pip) INFO: /Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3 -m pip install pyperclip --target /Users/Simone/Library/Application Support/mu/site-packages
2020-07-29 17:16:42,050 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/Simone/Library/Logs/mu/mu.log
2020-07-29 17:16:42,050 - mu.logic:163(installed_packages) INFO: Packages found: []

abixadamj, 2020-07-29:
Any Screenshots, just to look around...
OK, what about Teminal App and command:
```
/Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3 -m pip install pyperclip --target /Users/Simone/Library/Application Support/mu/site-packages
```
Can you try it and show output?

ximeng86, 2020-07-29:
Output is:

pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Invalid requirement: 'Support/mu/site-packages'
It looks like a path. Does it exist ?

abixadamj, 2020-07-29:
I do not use Mac, so I cannot help more. Maybe @tmontes will know better?

tmontes, 2020-08-02:
@ximeng86,

If I understand correctly what's going on, you're hitting a *severe limitation* with the currently distributed macOS Mu 1.1.0-alpha2... Technically it is due to the fact that it does not include all the bits needed for encrypted communication with PyPI, which is required to grab packages from there -- the jargon is TLS (aka SSL, which shows up a lot in the messages you pasted -- thanks!).

To confirm the limitation, I suggest you launch Mu in Python mode, bring up the REPL pane, and try:

```
In [1]: import ssl
```

What output do you get? I suspect it will fail, printing out some kind of error. If nothing is printed, it means it worked, but I'd venture it doesn't.

After that, please try this variation (note the leading `_` in `_ssl`):

```
In [2]: import _ssl
```

What output do you get? Again, I suspect it will fail, probably with a similarly phrased error.

**If the latter works**, (again, odds are low) then please try out:

```
In [3]: _ssl.OPENSSL_VERSION
```

Please share your results here and maybe we can figure out a work-around for people like you to apply until we can publish an updated better/shinier/happier Mu version that works as we all want! :-)

PS: Thanks @abixadamj for the calling! :-)

carlosperate, 2020-08-05:
If you follow @tmontes instructions and find out ssl is not working, you can try this workaround: https://github.com/mu-editor/mu/issues/910#issuecomment-558845231

This is assuming you haven't installed OpenSSL with Homebrew in the past. If you have installed it before, then I'd recommend not to revert to an old version as indicated in that link. To check this you can excute `brew list openssl` and if the output is `Error: No such keg: /usr/local/Cellar/openssl` then you should be good to go.

dybber, 2020-08-20:
I'm also experiencing this now, after I updated my Python versions earlier in the month.

```python
Traceback (most recent call last):
  File "/Users/dpr964/mu_code/blah.py", line 1, in <module>
    import ssl
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/ssl.py", line 101, in <module>
ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload/_ssl.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/openssl/lib/libssl.1.0.0.dylib
  Referenced from: /Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload/_ssl.cpython-36m-darwin.so
  Reason: image not found
>>> import _ssl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload/_ssl.cpython-36m-darwin.so, 2): Library not loaded: /usr/local/opt/openssl/lib/libssl.1.0.0.dylib
  Referenced from: /Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload/_ssl.cpython-36m-darwin.so
  Reason: image not found
>>> 
```

However, the current ssl-version is openssl 1.1.

carlosperate, 2020-11-08:
As this is the same issue as https://github.com/mu-editor/mu/issues/910, I'll close this ticket to keep things tidy, but please feel free to continue the conversation over there.

closed 2020-11-08 by carlosperate