mu-editor/mu #1084: Importing openpyxl in mu

annieacres, 2020-07-30:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

annieacres, 2020-07-30:
Hi, 

I have been trying to import the openpyxl module in python on my macOS catalina 10.15.6. I have successfully used pip install openpyxl in the terminal, but when I come to use >>>import openpyxl in python an error message appears saying 'openpyxl' is not defined. I have also looked through the modules in mu and openpxyl does not seem to be there. 

I'd be very grateful to receive some help!

carlosperate, 2020-08-05:
Hi @annieacres,

Mu runs it's own isolated copy of Python and packages, so it is not able to access packages installed using pip in a normal terminal.
If you install the latest alpha version of Mu there is a new option in the settings that allows you install 3rd party packages inside this isolated environment:

[image]

carlosperate, 2020-09-27:
Hi @annieacres 
As it's been a while without a response I'll close this issue as resolved, but please feel free to reopen if you had any issues installing packages inside Mu.

I've also created this ticket to ensure we remember to document this feature for the next release: https://github.com/mu-editor/mu/issues/1107

closed 2020-09-27 by carlosperate