mu-editor/mu #1085: æøå ÆØÅ becoming Ã¦Ã¸Ã¥ ÃÃÃ in serial in adafruit mode

jatmega, 2020-08-03:
Hi, 
I am new to this, and dont know if this should be its own issue, or if it fits under: https://github.com/mu-editor/mu/issues/797 

I was trying to write some Norwegian text in my code.py on my CircuitPython based microcontroller, that I would like to print back to me.

The code is `print("test av bokstaver: æøå ÆØÅ")` 

I did expect t

In the serial in Adafruit mode I,
expected to get:
   get "test av bokstaver: æøå ÆØÅ
but instead i got 
   test av bokstaver: Ã¦Ã¸Ã¥ ÃÃÃ

This is an issue for me atleas, and probably for some other Norwegians trying to use Mu to develope code.

I have tried to enable UTF-8 support in windows, but it did not change anything(several restarts), and broke some of my other software.

mu version: 1.0.3
CircuitPython 5.2.0
Windows 10 : 10.0.18363 Build 18363

My current workarround is using PuTTY for serial, it works, but it would be much more pleasant to be able to use the mu editor and its serial capability

dybber, 2020-08-03:
Thanks for the error report. I'm Danish, so I'd also really like to get this fixed soon! There's actually already a fix for it at #1044, which is waiting to get reviewed and merged.

However, we're currently in the process of getting some other issues solved and merged, which we believe it's best to get merged first. When these other pull requests are merged, we'll continue with the PR's on the ESP/CircuitPython/microbit modes. List of other related upcoming fixes: https://github.com/mu-editor/mu/pulls?q=is%3Aopen+is%3Apr+label%3A%22ESP+Mode%22

carlosperate, 2020-08-03:
Yes, I was able to have a look at this recently, but I didn't finish up writing some of tests needed, so I didn't really push anything. I'll update the PR this week with my current progress if it's not quite finishe.

@jatmega I hope you don't mind I'll close this ticket in favour of https://github.com/mu-editor/mu/issues/797 as I'll try to unify all the open issues for this problem there.

closed 2020-08-03 by carlosperate

jatmega, 2020-08-05:
@carlosperate Works great for me. Nice to see that this is an issue you are working on. Thanks for your work and contribution to this project, I really appreciate it.