mu-editor/mu #1086: install mu-editor 1.1.0 in ubuntu 18.0.4

sunnyyakima, 2020-08-09:
Hi I am trying to install mu 1.1.0 in ubuntu.

here are command I used:
python3 setup.py  install --user 

mu-editor is installed. But I got an error when running it:

Traceback (most recent call last):
  File "/home/leo/.local/bin/mu-editor", line 11, in <module>
    load_entry_point('mu-editor==1.1.0a2', 'console_scripts', 'mu-editor')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/leo/.local/lib/python3.6/site-packages/mu_editor-1.1.0a2-py3.6.egg/mu/__init__.py", line 4, in <module>
    from PyQt5.QtCore import QLocale
ModuleNotFoundError: No module named 'PyQt5.sip'

Can anyone help me on this? Thanks

sunnyyakima, 2020-08-09:
my python is 3.6.9

carlosperate, 2020-08-09:
I just noticed you had already opened this ticket after commenting in https://github.com/mu-editor/mu/issues/1078#issuecomment-671099341, thanks for that and sorry for the confusion!

As mentioned there, different installation instruction can be found in https://codewith.mu/en/howto/1.1/ .
If you'd like to install Mu from source the developer instructions can be found in https://mu.readthedocs.io/en/latest/setup.html .

carlosperate, 2020-11-11:
Installing from source in Ubuntu with Python 3.6.9 should be working with pip as indicated in https://mu.readthedocs.io/en/latest/setup.html.

I'll close this issue as resolved, but please feel free to reopen if you are still having issues.

closed 2020-11-11 by carlosperate