mu-editor/mu #1087: Add sparkfun samd51 thing plus - circuitpython

nezra, 2020-08-15:
If you are reporting a bug, we would like to know:

* What you were trying to do,

Use the sparkfun samd51 thing plus with circuitpython installed

* What steps you took to make this happen, 

Attempted to connect via repl, also via circuit python
* What you expected to happen,

Works.

* What actually happened,

No device attached error message
* Why this difference is problematic (it may not be a bug!),

Doesn't work.
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

In short, the modes definition under circuitpython is missing the device. It needs the following added:

PID: 0x0016
VID: 0x1b4f

Spark fun thing plus SAMD51

dybber, 2020-09-05:
Thanks for reporting this, it's not easy to keep tab on all the new devices which are manufactured.

I've added it to #1094, and should be working in the next version of Mu. Here's a direct link to the commit: https://github.com/mu-editor/mu/pull/1094/commits/47f8bbeff812a822598c44f5381257be500df12a

carlosperate, 2020-09-15:
I'll close this issue as https://github.com/mu-editor/mu/pull/1094 has been merged 🎉

closed 2020-09-15 by carlosperate