mu-editor/mu #1089: Installation fails: sdl-config not found

kjcole, 2020-08-27:
I was attempting to install mu-editor on Ubuntu 20.04.

```
$ pip3 install --user mu-editor
[...several downloads later...]
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-axggm6m2/pygame/setup.py'"'"'; __file__='"'"'/tmp/pip-install-axggm6m2/pygame/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-1m7d8te9
         cwd: /tmp/pip-install-axggm6m2/pygame/
    Complete output (12 lines):
    
    
    WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
    Using UNIX configuration...
    
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    
    Hunting dependencies...
    WARNING: "sdl-config" failed!
    Unable to run "sdl-config". Please make sure a development version of SDL is installed.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
I'd read that PyGame and Python 3.8 might be an issue but:

```
$ sudo apt install python3-pygame
$ python3
Python 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pygame
pygame 2.0.0.dev10 (SDL 2.0.12, python 3.8.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
>>>
```

I also added `libsdl2-dev` and `python3-sdl2` based on the error. No joy.

```
$ dpkg --get-selections | grep sdl
libsdl-image1.2:amd64                           install
libsdl-mixer1.2:amd64                           install
libsdl-ttf2.0-0:amd64                           install
libsdl1.2debian:amd64                           install
libsdl2-2.0-0:amd64                             install
libsdl2-dev:amd64                               install
libsdl2-gfx-1.0-0:amd64                         install
libsdl2-image-2.0-0:amd64                       install
libsdl2-mixer-2.0-0:amd64                       install
libsdl2-ttf-2.0-0:amd64                         install
python3-sdl2                                    install
$
```

kjcole, 2020-08-27:
Forging ahead:

```
$ sudo apt install libsdl1.2-dev
$ pip 3 install --user mu-editor
...
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] $
         cwd: /tmp/pip-install-psc9otk1/pygame/
    Complete output (18 lines):

    WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
    Using UNIX configuration...

    
    Hunting dependencies...
    SDL     : found 1.2.15
    FONT    : not found
    IMAGE   : not found
    MIXER   : not found
    PNG     : found
    JPEG    : not found
    SCRAP   : found
    PORTMIDI: not found
    PORTTIME: not found
    FREETYPE: found 23.1.17
    Missing dependencies
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

victordiaz, 2020-08-29:
Hi, I'm having the same issue in arch linux with python 3.8.
I really want to try Mu :) 

Is there a way to overcome the problem?

soltysh, 2020-09-10:
I installed (Fedora specific packages) SDL_mixer-devel, SDL_image-devel, SDL_ttf-devel. I also needed gcc-c++. Hope that helps.

kjcole, 2020-09-10:
> I installed (Fedora specific packages) SDL_mixer-devel, SDL_image-devel, SDL_ttf-devel. I also needed gcc-c++. Hope that helps.

Nope. I already had C++. 

```
apt install libsdl2-image-2.0-0 libsdl2-ttf-2.0-0 libsdl2-mixer-dev
```
brought no joy. (I now have all the `libsdl2*-dev` packages installed.)

soltysh, 2020-09-10:
@kjcole what's the current output from running `$ pip3 install --user mu-editor` it should have less dependencies still required.

kjcole, 2020-09-10:
```
$ pip3 install --user  mu-editor
Collecting mu-editor
  Using cached mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
Collecting qtconsole==4.3.1
  Using cached qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
Collecting pycodestyle==2.4.0
  Using cached pycodestyle-2.4.0-py2.py3-none-any.whl (62 kB)
Collecting guizero>=1.1.0
  Using cached guizero-1.1.0-py3-none-any.whl (41 kB)
Collecting PyQtChart==5.14.0
  Using cached PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
Collecting gpiozero>=1.4.1
  Using cached gpiozero-1.5.1-py2.py3-none-any.whl (138 kB)
Collecting qscintilla==2.11.4
  Using cached QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
Collecting pyqt5==5.14.1
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
Collecting nudatus>=0.0.3
  Using cached nudatus-0.0.4.tar.gz (4.5 kB)
Collecting matplotlib==2.2.2
  Using cached matplotlib-2.2.2.tar.gz (37.3 MB)
Requirement already satisfied: requests>=2.19.1 in /usr/lib/python3/dist-packages (from mu-editor) (2.22.0)
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting semver>=2.8.0
  Using cached semver-2.10.2-py2.py3-none-any.whl (12 kB)
Requirement already satisfied: appdirs>=1.4.3 in /home/kjcole/.local/lib/python3.8/site-packages (from mu-editor) (1.4.4)
Requirement already satisfied: Pillow>=5.2.0 in /home/kjcole/.local/lib/python3.8/site-packages (from mu-editor) (7.2.0)
Collecting pgzero==1.2
  Using cached pgzero-1.2-py3-none-any.whl (69 kB)
Collecting pigpio>=1.40.post1
  Using cached pigpio-1.46-py2.py3-none-any.whl (39 kB)
Collecting pyflakes==2.0.0
  Using cached pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
Collecting jupyter-client>=4.1
  Using cached jupyter_client-6.1.7-py3-none-any.whl (108 kB)
Requirement already satisfied: pygments in /home/kjcole/.local/lib/python3.8/site-packages (from qtconsole==4.3.1->mu-editor) (2.6.1)
Collecting ipykernel>=4.1
  Using cached ipykernel-5.3.4-py3-none-any.whl (120 kB)
Collecting jupyter-core
  Using cached jupyter_core-4.6.3-py2.py3-none-any.whl (83 kB)
Requirement already satisfied: traitlets in /home/kjcole/.local/lib/python3.8/site-packages (from qtconsole==4.3.1->mu-editor) (4.3.3)
Requirement already satisfied: ipython-genutils in /home/kjcole/.local/lib/python3.8/site-packages (from qtconsole==4.3.1->mu-editor) (0.2.0)
Requirement already satisfied: PyQt5-sip<13,>=12.7 in /home/kjcole/.local/lib/python3.8/site-packages (from PyQtChart==5.14.0->mu-editor) (12.8.0)
Collecting colorzero
  Using cached colorzero-1.1-py2.py3-none-any.whl (33 kB)
Requirement already satisfied: numpy>=1.7.1 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (1.17.4)
Requirement already satisfied: cycler>=0.10 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (2.4.6)
Requirement already satisfied: python-dateutil>=2.1 in /home/kjcole/.local/lib/python3.8/site-packages (from matplotlib==2.2.2->mu-editor) (2.8.1)
Requirement already satisfied: pytz in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (2019.3)
Requirement already satisfied: six>=1.10 in /home/kjcole/.local/lib/python3.8/site-packages (from matplotlib==2.2.2->mu-editor) (1.15.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (1.0.1)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7v9sh75m/pygame/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7v9sh75m/pygame/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-po1e_9g_
         cwd: /tmp/pip-install-7v9sh75m/pygame/
    Complete output (12 lines):
    
    
    WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
    Using UNIX configuration...
    
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    
    Hunting dependencies...
    WARNING: "sdl-config" failed!
    Unable to run "sdl-config". Please make sure a development version of SDL is installed.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
$
```

soltysh, 2020-09-10:
Looks like you're missing `SDL-devel` (Fedora-specific) package name.

kjcole, 2020-09-10:
> Looks like you're missing `SDL-devel` (Fedora-specific) package name.

No such beastie in the Ubuntu repositories.

I yield. (I was only trying it out to see if it was worth recommending to a beginner. I use emacs. Since I'm having this much trouble getting it going, I know my students will have at least as much trouble. I don't want to have to recommend PPDs or other repositories, or building from source. Maybe sometime in the future I'll give it a go again.)

closed 2020-09-10 by kjcole

tjguk, 2020-09-10:
@kjcole Just jumping in to say: thanks for trying to that extent that you did.

We definitely need a better story on Linux -- which is slightly ironic as the lead developer @ntoll is Linux (in fact Ubuntu)-based. But our main focus has always been on producing a packaged version with extra batteries as that's where our perceived target audience (secondary schools, code clubs etc.) sits, whether on Windows or on Mac.

If anyone does come up with a good set of instructions for current Ubuntu, I'll post them here so at least there visible for any future attempts :)

carlosperate, 2020-09-10:
I gave this a go in Ubuntu 20.04 and there is definitely quite a few issues with dependencies (old versions pinned for Mu) not supporting Python 3.8 and having to build from source (and some even using Python standard library deprecated functions), so at the moment PyPI Mu 1.0.3 is only up to 3.7 compatible.

For Ubuntu that means using something like deadsnakes PPA:

```
$ sudo add-apt-repository ppa:deadsnakes/ppa
$ sudo apt-get update
$ sudo apt-get install python3.7
$ python3.7 -m pip install mu-editor
$ python3.7 -m mu
```