mu-editor/mu #1090: File Menu Graphical Glitch on Raspbian Desktop?

alex-engelmann, 2020-09-01:
Hi!

I just set up a USB drive with Raspbian desktop, went to try out Mu (as it's included) and realized that the file menu doesn't seem to work.

When I try to Load or Save I see a menu that almost looks right but is unusable.  I've included screenshots as it's hard to explain:

Load
[image]

Save
[image]

This effectively makes it impossible for me to load and save files, which severely limits my use of the editor.  The Save one is especially bad as I have to hit Esc to close it, since it doesn't have a top bar to close it from.

Mu version 1.0.2
OS - Linux raspberrypi 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
I can't produce log files, as the issue seems to be popping up there when click on the Gear icon:

[image]

I'm not using any themes, my system is default colors and settings as I've barely started using it.  I'm not seeing this happen on any other programs I've tried so far (LibreOffice, Chromium, GIMP).  Any help would be appreciated.

carlosperate, 2020-09-03:
Thanks for the report @alex-engelmann.
As you might have guessed we are only requesting the prompts from PyQt, so not much else we could be doing within Mu to get it glitch like this.
Do you have any other Qt-based programs installed via deb file (so that it is using the same Qt library than PyQt is using), it'd be good to see if those have a similar issue.

As this is an x86_64 OS you could also create a new Python virtual environment and install Mu via pip, that should download the wheels from PyPI with their own Qt libs.

alex-engelmann, 2020-09-13:
HI @carlosperate thanks for getting back to me.  Yes I ultimately found it on other apps (VLC player for example) and believe it's a scaling issue with QT and 4k monitors.  Bug does not occur on actual Raspberry Pi OS (on the Pi 4 I mean), so doesn't ultimately matter as much.  Will go ahead and close this.

closed 2020-09-13 by alex-engelmann