mu-editor/mu #1098: Dash/Underline Characters Don't Appear at All Zoom Levels in Serial Monitor/REPL

robertgallup, 2020-09-10:
Under MacOS Catalina, in the serial monitor/REPL pane, dash and underline characters are not visible at all zoom levels. If you zoom out as far as you can, then zoom in two steps, dash/underline characters are not visible. If you zoom one more level in, the characters will appear again.

This behavior does not appear in Windows.

It seems like it might be related to font size interpolation and/or anti-aliasing.

dybber, 2020-10-07:
Thanks for reporting this. I can replicate it on the current master with Mac OS Mojave, but not in the most recent official release (1.1 alpha 2). Which version did you use @robertgallup?

Also, this seems to be an issue only affecting the MicroPython REPL, not the Python 3 QtConsole-REPL.

dybber, 2020-10-07:
My current thought is that this is due to some change in PyQt, and not something we have introduced. 

As I mentioned 1.1 alpha 2, doesn't seem to have this problem on my machine. However, when I checkout that version (efb7c70) in my development environment it *does* in fact have the issue. I'm then reasoning that the only difference is in the version of installed packages between what's included in 1.1 alpha 2 and my development environment.

I didn't get us much further, however we probably aren't going to solve this by looking for an old commit introducing this problem.

dybber, 2020-12-05:
I looked a bit more into this:

As the issue is not present in other text fields, I'm reasoning we're doing something wrong with the specific text area used for the REPL. It doesn't happen in QsciScintilla editor or in the Jupyter editor. 

What's more interesting, it also doesn't happen in the log viewer in the admin console. The main difference it seems, is that the log viewer is based on QPlainTextEdit, where as the REPL is a QTextEdit. I think we should try using QPlainTextEdit for the REPL as well. I don't have more time on my hands today, but will take a look at it tomorrow!

dybber, 2020-12-06:
Further investigation:
 - changing to a QPlainTextEdit didn't solve the problem
 - the reason the log viewer in the admin console works, is because it uses another non-monospace font (that, might be another bug, actually)

dybber, 2020-12-06:
I next tried to create a minimal Qt application, to see if I could make a minimal example replicating this behavior. However, the problem didn't appear. Here's the code:

```python
from PyQt5.QtWidgets import (QApplication, QMainWindow, QPlainTextEdit)
from PyQt5.QtGui import (QFontDatabase, QFont)
import sys

def main():
    # Initialize application
    app = QApplication(sys.argv)

    mainwindow = QMainWindow()
    textedit = QPlainTextEdit(mainwindow)
    mainwindow.setCentralWidget(textedit)
    mainwindow.show()

    # Create font database w. Source Code Pro
    font_filename = "mu/resources/fonts/SourceCodePro-Regular.otf"
    fontDB = QFontDatabase()
    fontDB.addApplicationFont(font_filename)

    # Change QPlainTextEdit font
    textedit.setFont(QFont("Source Code Pro", 8))
    
    app.exec_()
    sys.exit(0)

if __name__ == '__main__':
    main()
```

dybber, 2020-12-06:
Here's a minimal example where the problem occurs. Might have something to do with how we load the fonts.

```python
from PyQt5.QtWidgets import (QApplication, QMainWindow, QPlainTextEdit)
from mu.interface.themes import Font
import sys

def main():
    # Initialize application
    app = QApplication(sys.argv)
    
    mainwindow = QMainWindow()
    textedit = QPlainTextEdit(mainwindow)
    textedit.setPlainText("Example _-_-_ |il!")
    
    mainwindow.setCentralWidget(textedit)
    mainwindow.show()

    font = Font().load()
    textedit.setFont(font)
    
    app.exec_()
    sys.exit(0)

if __name__ == '__main__':
    main()
```

dybber, 2020-12-06:
Could it be because for `QFontDatabase.addApplicationFontFromData` "Currently only TrueType fonts and TrueType font collections are supported"? https://doc.qt.io/qt-5/qfontdatabase.html#addApplicationFontFromData
We're using OpenType Fonts.

For `QFontDatabase.addApplicationFont` they write: "Currently only TrueType fonts, TrueType font collections, and OpenType fonts are supported."
https://doc.qt.io/qt-5/qfontdatabase.html#addApplicationFont

dybber, 2020-12-06:
The problem also appears when using `QFontDatabase.addApplicationFont`:

```python
from PyQt5.QtWidgets import (QApplication, QMainWindow, QPlainTextEdit)
from PyQt5.QtGui import (QFontDatabase, QFont)
import sys
import os

def main():
    # Initialize application
    app = QApplication(sys.argv)

    mainwindow = QMainWindow()
    textedit = QPlainTextEdit(mainwindow)
    textedit.setPlainText("Example _-_-_ |il!")
    mainwindow.setCentralWidget(textedit)
    mainwindow.show()

    # Create font database w. Source Code Pro
    font_filename = os.path.abspath("mu/resources/fonts/SourceCodePro-Regular.otf")
    fontDB = QFontDatabase()
    fontDB.addApplicationFont(font_filename)

    # Change QPlainTextEdit font
    font = fontDB.font("Source Code Pro", "Regular", 14)
    textedit.setFont(font)
    
    app.exec_()
    sys.exit(0)

if __name__ == '__main__':
    main()
```

dybber, 2020-12-09:
I just tried figuring out why it works in the QtConsole, and it seems that in QtConsole we're actually not using Source Code Pro, but instead ".SF NS Text", which I believe is the default monospace font.

This doesn't explain why it works for the main editor widget. However, the editor widget is implemented using QsciScintilla, and when zooming out on QsciScintilla, we're not just changing font sizes, but using instead a QsciScintilla specific zoom function: https://docs.huihoo.com/pyqt/QScintilla2/classQsciScintilla.html#ab33ae790e43f4b856e44d202b9b3ce07

dybber, 2021-01-26:
I will suggest we move this to Milestone 1.2, as it really doesn't seem to be a problem on our end, and thus not much we can do about it other than mention it to PyQt devs

carlosperate, 2021-06-15:
**Merging https://github.com/mu-editor/mu/issues/1353 here:**

[image] [image]

Mac OS X 11.2.2, MacBook Pro (15inch, 2018)
Mu 1.1.0.alpha.2
Window in default zoom level.

The CircuitPython REPL text renders well on my laptop screen. When I move the window to an external monitor (2560 x 1440) then in the REPL, underscores are not rendered at all.

_Originally posted by @BiffoBear in https://github.com/mu-editor/mu/issues/1353#issue-823518702_

<br>

[image] [image]

This sounds like the same issue as #1098. Could you try if the issue persist between zoom-levels?

I've tried debugging it previously, but it seems more like a problem with the underlying libraries we use (Qt) or the font (Source Code Pro), as I can replicate it with very small Qt-programs.

_Originally posted by @dybber in https://github.com/mu-editor/mu/issues/1353#issuecomment-791935237_

<br>

[image] [image]

Yes, it seems to be something to do with the way Mac OS X renders  fonts on different display resolutions. The REPL font renders poorly on my 2560 x 1440 external monitor at all sizes. The underscore is missing only at the default size on the external monitor. It is there in other sizes.

Would you prefer me to close this issue and comment on #1098?

_Originally posted by @BiffoBear in https://github.com/mu-editor/mu/issues/1353#issuecomment-792213433_

<br>

[image] [image]

I have several other external monitors on macOS (M1 Mac mini), and yes, the underscores are a problem on all three. (But NOT on any native resolution of a M1 MacBook Air). 

Is there any kind of way to change the font used? we might try some others, and see if any of them are better in these cases?

_Originally posted by @xt0fer in https://github.com/mu-editor/mu/issues/1353#issuecomment-861594790_

xt0fer, 2021-06-15:
Yeah, I'm thinking it is at least a Qt rasterization from the text machinery. But could it also be a font hinting problem from the default font used by the text widget? If the outlines don't align with the raster grid? are the same fonts used on all platforms?

carlosperate, 2021-06-20:
> are the same fonts used on all platforms?

Yeah, SourceCodePro font is packaged and used (except for the apt/deb package): https://github.com/mu-editor/mu/tree/master/mu/resources/fonts

raquo, 2022-01-10:
Just wanted to say that I also ran into this bug, exactly as it's described in the issue, also with Macos Catalina. Thankfully Mu remembers the zoom level, so I'll be using a different one as a workaround.