mu-editor/mu #1099: mu on Linux Mint 20 - in case someone will have a problem

abixadamj, 2020-09-17:
This is only **INFORMATION**
_mu 1.0.3 is able to install on Linux Mint 20._

Before I have installed some packages:
```
apt install --yes python3-freetype libfreetype-dev libfreetype6-dev
apt install --yes python3-pygame-sdl2 python3-dev
apt install --yes libsdl-image1.2-dev libsdl-mixer1.2-dev libsdl-ttf2.0-dev libsdl1.2-dev libsmpeg-dev
apt install --yes libportmidi-dev ffmpeg libswscale-dev libavformat-dev libavcodec-dev libfreetype6-dev
```

After that, I run the following:

```
**root@zs-ceglow:~# pip3 install mu-editor==1.0.3**
Collecting mu-editor==1.0.3
  Downloading mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 1.2 MB/s 
Collecting pyqt5==5.14.1
  Downloading PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
     |████████████████████████████████| 63.5 MB 26.1 MB/s 
Requirement already satisfied: Pillow>=5.2.0 in /usr/lib/python3/dist-packages (from mu-editor==1.0.3) (7.0.0)
Collecting nudatus>=0.0.3
  Downloading nudatus-0.0.4.tar.gz (4.5 kB)
Collecting semver>=2.8.0
  Downloading semver-2.10.2-py2.py3-none-any.whl (12 kB)
Collecting gpiozero>=1.4.1
  Downloading gpiozero-1.5.1-py2.py3-none-any.whl (138 kB)
     |████████████████████████████████| 138 kB 64.8 MB/s 
Collecting pycodestyle==2.4.0
  Downloading pycodestyle-2.4.0-py2.py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 947 kB/s 
Collecting qscintilla==2.11.4
  Downloading QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
     |████████████████████████████████| 1.9 MB 35.9 MB/s 
Collecting qtconsole==4.3.1
  Downloading qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
     |████████████████████████████████| 108 kB 41.0 MB/s 
Collecting pyflakes==2.0.0
  Downloading pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
     |████████████████████████████████| 53 kB 2.0 MB/s 
Collecting pgzero==1.2
  Downloading pgzero-1.2-py3-none-any.whl (69 kB)
     |████████████████████████████████| 69 kB 7.3 MB/s 
Collecting appdirs>=1.4.3
  Downloading appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting guizero>=1.1.0
  Downloading guizero-1.1.0-py3-none-any.whl (41 kB)
     |████████████████████████████████| 41 kB 686 kB/s 
Requirement already satisfied: requests>=2.19.1 in /usr/lib/python3/dist-packages (from mu-editor==1.0.3) (2.22.0)
Collecting matplotlib==2.2.2
  Downloading matplotlib-2.2.2.tar.gz (37.3 MB)
     |████████████████████████████████| 37.3 MB 54 kB/s 
Collecting PyQtChart==5.14.0
  Downloading PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 2.1 MB/s 
Collecting pigpio>=1.40.post1
  Downloading pigpio-1.46-py2.py3-none-any.whl (39 kB)
Requirement already satisfied: pyserial==3.4 in /usr/local/lib/python3.8/dist-packages (from mu-editor==1.0.3) (3.4)
Collecting PyQt5-sip<13,>=12.7
  Downloading PyQt5_sip-12.8.1-cp38-cp38-manylinux1_x86_64.whl (293 kB)
     |████████████████████████████████| 293 kB 40.3 MB/s 
Collecting colorzero
  Downloading colorzero-1.1-py2.py3-none-any.whl (33 kB)
Requirement already satisfied: ipython-genutils in /usr/local/lib/python3.8/dist-packages (from qtconsole==4.3.1->mu-editor==1.0.3) (0.2.0)
Requirement already satisfied: jupyter-core in /usr/local/lib/python3.8/dist-packages (from qtconsole==4.3.1->mu-editor==1.0.3) (4.6.3)
Requirement already satisfied: jupyter-client>=4.1 in /usr/local/lib/python3.8/dist-packages (from qtconsole==4.3.1->mu-editor==1.0.3) (6.1.7)
Requirement already satisfied: traitlets in /usr/local/lib/python3.8/dist-packages (from qtconsole==4.3.1->mu-editor==1.0.3) (5.0.4)
Requirement already satisfied: ipykernel>=4.1 in /usr/local/lib/python3.8/dist-packages (from qtconsole==4.3.1->mu-editor==1.0.3) (5.3.4)
Requirement already satisfied: pygments in /usr/lib/python3/dist-packages (from qtconsole==4.3.1->mu-editor==1.0.3) (2.3.1)
Requirement already satisfied: numpy in /usr/local/lib/python3.8/dist-packages (from pgzero==1.2->mu-editor==1.0.3) (1.19.2)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.8/dist-packages (from matplotlib==2.2.2->mu-editor==1.0.3) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.8/dist-packages (from matplotlib==2.2.2->mu-editor==1.0.3) (1.2.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor==1.0.3) (2.4.6)
Requirement already satisfied: python-dateutil>=2.1 in /usr/local/lib/python3.8/dist-packages (from matplotlib==2.2.2->mu-editor==1.0.3) (2.8.1)
Collecting pytz
  Downloading pytz-2020.1-py2.py3-none-any.whl (510 kB)
     |████████████████████████████████| 510 kB 46.6 MB/s 
Requirement already satisfied: six>=1.10 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor==1.0.3) (1.14.0)
Requirement already satisfied: tornado>=4.1 in /usr/local/lib/python3.8/dist-packages (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (6.0.4)
Requirement already satisfied: pyzmq>=13 in /usr/local/lib/python3.8/dist-packages (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (19.0.2)
Requirement already satisfied: ipython>=5.0.0 in /usr/local/lib/python3.8/dist-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (7.18.1)
Requirement already satisfied: decorator in /usr/local/lib/python3.8/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (4.4.2)
Requirement already satisfied: backcall in /usr/local/lib/python3.8/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (0.2.0)
Requirement already satisfied: prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0 in /usr/local/lib/python3.8/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (3.0.7)
Requirement already satisfied: pexpect>4.3; sys_platform != "win32" in /usr/lib/python3/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (4.6.0)
Requirement already satisfied: jedi>=0.10 in /usr/lib/python3/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (0.15.2)
Requirement already satisfied: setuptools>=18.5 in /usr/lib/python3/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (45.2.0)
Requirement already satisfied: pickleshare in /usr/local/lib/python3.8/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (0.7.5)
Requirement already satisfied: wcwidth in /usr/local/lib/python3.8/dist-packages (from prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0->ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor==1.0.3) (0.2.5)
Building wheels for collected packages: nudatus, matplotlib, pygame
  Building wheel for nudatus (setup.py) ... done
  Created wheel for nudatus: filename=nudatus-0.0.4-py3-none-any.whl size=4985 sha256=dde65714982f27ab5fc1e6ba4ee8b2f2ebefec4666210293052e8a9e2da7b081
  Stored in directory: /root/.cache/pip/wheels/67/6b/17/c0a49d82be8b8d0ca790d0c62dea713b49c2408dbb82025347
  Building wheel for matplotlib (setup.py) ... done
  Created wheel for matplotlib: filename=matplotlib-2.2.2-cp38-cp38-linux_x86_64.whl size=11398383 sha256=e02efd0bb46a386c2a1fda996e27c7949d903c066bf0deb32aa377ef1da51e3d
  Stored in directory: /root/.cache/pip/wheels/5b/99/36/584361dc7788ea9af2bf4dc89881e8c4cc251c15c8dd16dcf6
  Building wheel for pygame (setup.py) ... done
  Created wheel for pygame: filename=pygame-1.9.6-cp38-cp38-linux_x86_64.whl size=3833021 sha256=ee70848a7541460423d9e47282ca9ef4f3de9e38b7d3906aebeff792aa8f6356
  Stored in directory: /root/.cache/pip/wheels/5c/3a/34/864e618a7e57b5df7cc5f1d1338aa43e7d566a4749a540c9b7
Successfully built nudatus matplotlib pygame
Installing collected packages: PyQt5-sip, pyqt5, nudatus, semver, colorzero, gpiozero, pycodestyle, qscintilla, qtconsole, pyflakes, pygame, pgzero, appdirs, guizero, pytz, matplotlib, PyQtChart, pigpio, mu-editor
  Attempting uninstall: qtconsole
    Found existing installation: qtconsole 4.7.7
    Uninstalling qtconsole-4.7.7:
      Successfully uninstalled qtconsole-4.7.7
  Attempting uninstall: pygame
    Found existing installation: pygame 2.0.0.dev10
    Uninstalling pygame-2.0.0.dev10:
      Successfully uninstalled pygame-2.0.0.dev10
  Attempting uninstall: matplotlib
    Found existing installation: matplotlib 3.3.2
    Uninstalling matplotlib-3.3.2:
      Successfully uninstalled matplotlib-3.3.2
Successfully installed PyQt5-sip-12.8.1 PyQtChart-5.14.0 appdirs-1.4.4 colorzero-1.1 gpiozero-1.5.1 guizero-1.1.0 matplotlib-2.2.2 mu-editor-1.0.3 nudatus-0.0.4 pgzero-1.2 pigpio-1.46 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pyqt5-5.14.1 pytz-2020.1 qscintilla-2.11.4 qtconsole-4.3.1 semver-2.10.2

```

dybber, 2020-09-17:
Is this something you suggest should be added to one of the howto guides on the Mu website? https://codewith.mu/en/howto/

What error did you experience, if you tried installing without installing the packages with `apt` beforehand?

Are we sure that all those packages are necessary to run Mu? (Does Mu really depend on `ffmpeg`?)

carlosperate, 2020-09-17:
Is it possible the `matplotlib` and `pygame`were build from source and all the apt dependencies are only needed to build them?
The wheels for the dependences hard-pinned in Mu 1.0.3 are not compatible with Python 3.8, so it's quite likely that this is what happened.
The easiest way to install Mu is with Python 3.6 or 3.7, we should update https://codewith.mu/en/howto/1.0/install_with_python to include that.

carlosperate, 2021-03-16:
This should have been fixed in the latest release 🎉 
- https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 
- https://pypi.org/project/mu-editor/1.1.0b2/

I'll close this as resolved, but please do reopen it if you are still having this issue with the v1.1.0-beta.2.

closed 2021-03-16 by carlosperate