mu-editor/mu #1103: Issues installing Mu in Raspbian Stretch

carlosperate, 2020-09-20:
If anybody tries to install Mu using the [dev instructions](https://mu.readthedocs.io/en/latest/setup.html) on Raspberry Pi with Raspbian Stretch, there are a few python packages that will fail to install when running `pip install -e ."[dev]"`.

This is because the `twine`, `traitlets` and `ipython` packages are not hard pinned (these are dependencies of Mu dependencies), so it installs the latest available version instead of the latest compatible version, and these are not compatible with Python 3.5, the python3 version in Stretch.

This will be fixed soon as PR https://github.com/piwheels/piwheels/pull/254, the implementation of https://github.com/piwheels/piwheels/issues/208, is merged and deployed in piwheels. @bennuttall is already on this (thanks Ben!).

But in the meantime these packages will have to be manually installed first:

```
$ pip install twine==1.15.0
$ pip install traitlets==4.3.3
$ pip install ipython==7.9.0
$ pip install -e ."[dev]"
``` 

It's probably not worth updating the docs if these steps won't be needed for long, but will be good to have it here in case anybody encounters this issue.

bennuttall, 2020-09-20:
Fix should be deployed this week. I'm addressing @waveform80's PR review comments as we speak!

carlosperate, 2020-09-21:
That's great news, thanks @bennuttall and @waveform80!

bennuttall, 2020-09-25:
Done! https://blog.piwheels.org/requires-python-support-new-project-page-layout-and-a-new-json-api/

carlosperate, 2020-09-27:
Thanks Ben!
I can confirm that a pip install in Raspbian/Raspberry Pi OS now correctly installs the package version needed for the running Python version. 🎉 

Now that the requires-python check has been implemented in piwheels the instructions we have in the Mu docs work correctly again 👍

closed 2020-09-27 by carlosperate