mu-editor/mu #1105: Random failures on the CI

carlosperate, 2020-09-24:
For quite a long time now it has been quite common having to restart Travis CI builds due to random failures.
In these cases it is always it always fails the same way with some package with a C extension that causes Python to abort the process.

I had a quick check changing the pytest flags to get more info out of the Travis failures and this is what I've found so far:
- It takes about 6 or 7 individual restarts to get all the 3 Linux builds to fail in a Travis job:
    - https://travis-ci.org/github/mu-editor/mu/builds/729939378
    - https://travis-ci.org/github/mu-editor/mu/builds/730072364
    - So roughly a Travis Linux build has 1/3 chances of failing due to this
- It turned out to be PyQt causing the issue: `QWidget: Must construct a QApplication before a QWidget`
- I've been able to replicate the same issue in Ubuntu 20.04 with Python 3.7 and macOs Mojave with 3.6
    - However, it took more attempts locally than in Travis, about 10 to 15 restarts to get it to fail again

I'll look into what we might need to do to fix this.

dybber, 2020-09-25:
I've seen that before! 

Apparantly there's some tests in `test_logic.py` that also involves Qt, where I forgot to apply the fix I found.

The fix is to make all unit tests that does anything with Qt to depend on the `qtapp` fixture, which is defined globally for all our tests in tests/conftest.py:
 https://github.com/mu-editor/mu/blob/master/tests/conftest.py

These few lines creates a global QApplication instance that will be used by all tests depending on that fixture.

Here's an example from `test_main.py` that shows how it's used:
```python

def test_Window_zoom_out(qtapp):
    """
    Ensure the correct signal is emitted.
    """
    w = mu.interface.main.Window()
    w._zoom_out = mock.MagicMock()
    w.zoom_out()
    w._zoom_out.emit.assert_called_once_with("s")
```
`qtapp` just needs to be added as argument where ever we make a test that instantiates parts of the UI

carlosperate, 2020-09-25:
Oh, that's great, thanks Martin! That'll save me the PyQt debugging 😁 

I found a couple of test cases needing that, I believe the issue happens when they are the first test running from their file, as we have a randomiser, that also explains why it only fails sometimes.

dybber, 2020-09-25:
That might be the case, yes, that the order of tests is giving the issues.

I don't know what the best way to avoid this occuring again. Let's say a contributor adds some tests without `qtapp`, then the error might not occur immediately. Perhaps it would be better, if we instead just made sure _always_ to have a QApplication() initialized before running any tests?

Alternatively, if we could force the errors to occur on the tests that actually need a QApplication, by not reusing the same environment for individual tests in the same file, that would also be an option, but I'm not sure if that's possible.

dybber, 2020-09-25:
### Making `qtapp` implicit for all tests
Seems easy to just make `qtapp` an implicit fixture for all our tests. We just need to add  `autouse=True`:

```python
@pytest.fixture(scope="session", autouse=True)
def qtapp():
    ...
```

Will that give any problems? (I guess not)

### Attempt at forcing the errors to occur
I can't figure out how to force the errors to occur instead of showing up randomly. I tried to add some teardown code to the `qtapp` fixture, but I couldn't get it to work. This is what I tried:
```python
@pytest.fixture()
def qtapp():
    app = QApplication([])
    yield app
    app.exit()
```

```python
@pytest.fixture()
def qtapp(request):
    app = QApplication([])
    request.addfinalizer(app.exit)
    return app
```

tjguk, 2020-09-25:
My only concern (and it's not a big one) is how expensive that fixture is, if we're going to run it against every test. (But, presumably, we can include it as `autouse` only in those `test_xxx.py` files which principally test Qt functionality.

dybber, 2020-09-25:
It will only be run once, as long as we set `scope=“session”`

Martin Dybdal

carlosperate, 2020-09-25:
Yeah, I can confirm with the session scope the fixture only runs once.

I've run several timings of the tests with the `qtapp` as test cases arguments vs setting the fixture to `autouse`, and I haven't noticed any difference in my laptop. Running the tests 10 times in a row in quiet mode took 75s +/- 5%  in all cases.

I think `autouse` is a good call, saves us from having to think about what might or might not need the fixture when writing tests.

dybber, 2020-09-25:
Should we then remove the explicit `qtapp` everywhere again? Almost just one or two inplace `sed` operations

carlosperate, 2020-09-25:
Yeah, even though it's not necessary it's one less thing to think about, specially for new contributors if it's not clear what it is.
I've update https://github.com/mu-editor/mu/pull/1106 and ready for review :)

closed 2020-09-26 by dybber