mu-editor/mu #1107: Docs: Third Party Packages

carlosperate, 2020-09-27:
Logging this here as a "todo".
When the next version of Mu is ready for release we need to add a tutorial in the docs about installing third party packages.

[image]

carlosperate, 2020-09-28:
The Automate The Boring Stuff with Python had to add this to the appendix: https://automatetheboringstuff.com/2e/appendixa/

carlosperate, 2021-03-15:
@ntoll there is already a PR from the great @martinohanlon in https://github.com/mu-editor/mu-editor.github.io/pull/51

martinohanlon, 2021-03-16:
Thesw instructions should still be relevant. I don't think anything has changed since I created them.

carlosperate, 2021-06-14:
PR https://github.com/mu-editor/mu-editor.github.io/pull/51 has been merged, thanks @ntoll! 🎉 
And @martinohanlon thank you for submitting the PR!

closed 2021-06-14 by carlosperate