mu-editor/mu #1109: supported usb id's .. please add USB_PRODUCT = "Feather STM32F405 Express"

CarlFK, 2020-09-27:
mu has a list of ID's it supports:
```
# List of supported board USB IDs.  Each board is a tuple of unique USB vendor
# ID, USB product ID.
BOARD_IDS = [
    # VID  , PID   , manufact., device name
    (0x0D28, 0x0204, None, "micro:bit"),
    (0x239A, 0x800B, None, "Adafruit Feather M0"),  # CDC only
    (0x239A, 0x8016, None, "Adafruit Feather M0"),  # CDC + MSC
    (0x239A, 0x8014, None, "Adafruit Metro M0"),
    (0x239A, 0x8019, None, "Circuit Playground Express M0"),
    (0x239A, 0x8015, None, "Circuit Playground M0 (prototype)"),
    (0x239A, 0x801B, None, "Adafruit Feather M0 Express"),
]
```
https://github.com/mu-editor/mu/blob/master/mu/modes/base.py#L42-L53

Which does not include:
239a:805a 
USB_VID = 0x239A
USB_PID = 0x805A
https://github.com/adafruit/circuitpython/blob/main/ports/stm/boards/feather_stm32f405_express/mpconfigboard.mk#L1-L2

or the MP version: 
f055:9800 
https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/ports/stm32/usb.h#L34-L35

dybber, 2020-09-27:
That list is actually left-over that we've forgotten to remove and is not used anymore! 

The actual lists used are those specified inside the various modes. I believe the first one you mention is actually supported, but perhaps not in any official Mu release yet: https://github.com/mu-editor/mu/blob/master/mu/modes/circuitpython.py#L43

The other one isn't on the list, so yeah I guess it should be added. https://github.com/mu-editor/mu/blob/master/mu/modes/esp.py#L49
Which hardware device is it that get's that VID/PID? Is that the pyboard? 

There's some work on a pyboard mode over at #650, as the libraries are different than plain MicroPython, and it should supply different documentation. But just want to make sure I understand this correctly, and this is not a different board.

CarlFK, 2020-09-27:
oh gosh, please remove that code before someone else goes bonkers trying to add a board.  I figured I wasn't getting the description string right or something.

both sets of ID's are from the same hardware:
https://learn.adafruit.com/adafruit-stm32f405-feather-express 

in addition, when I jumper and put it into boot loader mode:
`0483:df11 STMicroelectronics STM Device in DFU Mode`

Which might be useful to tell the user "Yes a recognized board is plugged in, but no it isn't going to work right now."

dhalbert, 2020-09-28:
All VID=0x239a are now detected as CircuitPython boards by https://github.com/mu-editor/mu/pull/721. I agree this confusing should be removed. I'm sure a PR would be welcome.

dybber, 2020-09-29:
The alternative is what @CarlFK suggests in #1110, where users can add additional devices e.g. through some kind of configuration option (e.g. adding a line in a configuration file)

CarlFK, 2020-09-29:
>     in addition, when I jumper and put it into boot loader mode:
>     0483:df11 STMicroelectronics STM Device in DFU Mode
> 
>     Which might be useful to tell the user "Yes a recognized board is plugged in, but no it isn't going to work right now."
> 
> I'm not sure I completely understand this part, @CarlFK, could you try to explain it in other words?

to flash a new CP or MP onto the board,  "..connect the BOOT0 (B0) pin to 3.3V logic. "
https://learn.adafruit.com/adafruit-stm32f405-feather-express/dfu-bootloader-details

Similar to double pressing the reset button on the CPX, which changes the id from
239a:8019
to
239a:0018