mu-editor/mu #1110: ER: option to add usb id

CarlFK, 2020-09-27:
I would like a command line option to specify a usb id to be added/used for:

# List of supported board USB IDs.  Each board is a tuple of unique USB vendor
https://github.com/mu-editor/mu/blob/master/mu/modes/base.py#L42

So that when I plug in a board not on the current list, I can set the ID of the board I want to work with.

dybber, 2020-09-27:
I've thought about this as well, I think one option would be to make it part of the configuration file, or a separate additional configuration file where you can list additional devices. Actually, the current list of devices could just as well be a file that was read instead of hard-coded in the .py file.

Another option, would be to allow interacting with unknown devices. E.g. if any serial device is detected, it will always show up in the drop-down, but I'm not sure if there's all kinds of serial devices, which would show up and be confusing for learners using Mu.

CarlFK, 2020-09-27:
I like 2 files.  It means mu updates can replace the base files, and user specified things aren't stepped on.

I love that mu is easy for beginners.  I don't mind suffering in order to make it easier for new people to get started. 

unknown devices - hide it behind an advanced mode option.   it may get tricky trying to associate serial ports with drives (usb storage I think it is called?)  but as soon as you have enabled advanced mode, the user can be responsible for all that.