mu-editor/mu #1111: ModuleNotFoundError: No module named 'openpyxl'

shartrooper, 2020-09-27:
Hi. 
Recently, I've installed the **openpyxl** module through **pip3 install --user -U openpyxl==2.6.2**; then added the **import openpyxl** statement and the SheBang line **#! python3** at the start of my script to use the module. 

However, for some reason, whenever I run it on my **Mu 1.1.0.alpha.2 build**, the following error log is thrown: 
```
Traceback (most recent call last):
  File "c:\users\marko\documents\varios\python scripts\documents manipulation\updateproduce.py", line 4, in <module>
    import openpyxl
ModuleNotFoundError: No module named 'openpyxl'
```
As reference, I already tried running the script from IDLE,  It loads and use the module as intended. Not error log. 

My operating system is Win7.

dybber, 2020-09-28:
Hi @shartrooper. Mu runs it's own Python installation, with it's own set of packages, so it's not able to access packages installed outside Mu with pip in a terminal. The intension is to make it easy for beginners to get started, as they don't have to set up the editor and Python installation separately.

If you install the newest alpha release however, there's the option to install third party packages within Mu's isolated environment. You have to push the settings button in the bottom right corner, and type in `openpyxl` on a separate line in this dialog:
[image]

The same issue was reported earlier in #1084, so you are not the only one who finds this confusing. We will try to document this better in the future on the website as per #1107

shartrooper, 2020-09-28:
Oops, my mistake then. I've been using Js/php with vscode for a while and assumed package management was similar to this IDE.

closed 2020-09-28 by shartrooper