mu-editor/mu #1117: switching between devices does't switch default file system

CarlFK, 2020-09-29:
version: today's git master 

This is messy.  I suspect getting things to work right is going to be quite a bit of work as it isn't easy to figure out what usb device provides what tty and storage device.

I suspect the solution is make it easy for one device so that beginners have a good experience,  and consider 2 devices "expert mode" and give the user more options.

I'm not even sure if this should be 1 issue or many.  
so, here is one of the current problems:  

2 CPX boards running CircuitPython, mu sees 2, but switching between the 2 doesn't change the default when I click "Load":

log shows:
```
2020-09-29 11:27:04,825 - mu.logic:833(check_usb) INFO: circuitpython device connected on port: /dev/ttyACM2(VID: 0x239A, PID: 0x8019, manufacturer: 'Adafruit Industries LLC')
2020-09-29 11:27:04,826 - mu.logic:833(check_usb) INFO: circuitpython device connected on port: /dev/ttyACM0(VID: 0x239A, PID: 0x8019, manufacturer: 'Adafruit Industries LLC')
```

Select the ACM0 device, click load:
`2020-09-29 11:29:56,722 - mu.logic:1198(get_dialog_directory) INFO: Using path for file dialog: /media/carl/CIRCUITPY1
`
Select the ACM2 device, click load:
`2020-09-29 11:32:32,422 - mu.logic:1198(get_dialog_directory) INFO: Using path for file dialog: /media/carl/CIRCUITPY1`

In both cases, the dialog shows the same CPX.

It does switch between the 2 device's serial (good).

dybber, 2020-09-29:
Oh yeah, that can indeed get complicated. It's probably not easy to figure out which serial device and which volume corresponds to the same device. I guess you can't tell a CircuitPython or micro:bit (e.g. through serial) to change its volume name to something more specific?

I guess we could change the CircuitPython and micro:bit modes to transfer files through the serial interface, like in the ESP MicroPython mode, then we wouldn't have that problem. I'm no expert in CircuitPython and micro:bit, however, so perhaps someone else should chip in with ideas for possible solutions.

Currently everything should work for only one device, so at least the large majority will not have any problems.

CarlFK, 2020-09-29:
> Oh yeah, that can indeed get complicated. It's probably not easy to figure out which serial device and which volume corresponds to the same device.

Basically.  it is possible, but likely difficult to work on most os/distro/platform reliably. 

> I guess you can't tell a CircuitPython or micro:bit (e.g. through serial) to change its volume name to something more specific?

Not really, nothing I would want new users to be exposed to.  I don't know if it can be done over repl/serial.  Here is what I do know:
The volume name is a file system thing, the fatlabel command can set the label.
```
$df
...
/dev/sdc1             94        34        60  37% /media/carl/PYBFLASH
/dev/sdd1           2024        32      1992   2% /media/carl/CIRCUITPY

$ pumount /dev/sdc1
$ pumount /dev/sdd1 

$ fatlabel /dev/sdc1
PYBFLASH   
$ fatlabel /dev/sdd1
CIRCUITPY  

$ fatlabel /dev/sdc1 CIRCUITPY

$ fatlabel /dev/sdc1
CIRCUITPY  
(un/plug to get automounter to mount it)

$ df
/dev/sdc1           2024        32      1992   2% /media/carl/CIRCUITPY
/dev/sdd1             94        34        60  37% /media/carl/CIRCUITPY1
```

> I guess we could change the CircuitPython and micro:bit modes to transfer files through the serial interface, like in the ESP MicroPython mode, then we wouldn't have that problem. I'm no expert in [...]

I am not aware of being able to do that.

> 
> Currently everything should work for only one device, so at least the large majority will not have any problems.

:+1:

dybber, 2020-10-06:
One hacky option would be to write a file using serial connection with a unique identifier, and then read that file through the mountpoint (and delete it again), then we would be able to match serial connection with mountpoint.

I don't have any other ideas right now, and as I don't have multiple of the same devices right now, it's unclear for me what their specific behavior are when multiple devices are connected. I will try to get hands on one more micro:bit to test with.

CarlFK, 2020-10-06:
I see 2 ways of solving this:

1. Ask the OS for information about usb devices, figure out what devices have both serial and storage and look Micro Python-ish.
This is going to be a bit of work to work on various OSs, and may break the next time an OS changes something like permissions required for such tings.

2. give up. ;)

I'm in favour of 2 as I don't think there are enough (any) mu users that need to switch between devices using the UI.   

being able to pick what is needed on the fly would be fine.  Either a com port picker and fs navigator,  or command line parameters: which I started to work on in #1119

dybber, 2020-11-23:
I also think we should wait with this until more people experience the problem, as it might be very complicated with potentially very little gain.

aivarannamaa, 2020-12-18:
You can query (and change) the mount label from CircuitPython code: https://learn.adafruit.com/welcome-to-circuitpython/the-circuitpy-drive#renaming-circuitpy-through-circuitpython-3014813-19

dybber, 2020-12-18:
nice find! I wonder if this is also the case for micro:bit's and the Pyboard.

carlosperate, 2020-12-18:
Not on the micro:bit, as there are programmes (and tutorials) that depend on the drive name to find the micro:bit.
Not sure about the Pyboard.