mu-editor/mu #1120: Hacktoberfest proposals

carlosperate, 2020-09-29:
As [Hacktoberfest](https://hacktoberfest.digitalocean.com/) is about to start we could select some issues to tag in case people are interested to contribute to Mu.

I've been through the issue tracker and selected these as they seemed more or less "self contained". While some of this have more discussions that others, I think most of them would probably need a comment to summarise or conclude the next steps.

While I can try to summarise some of these tickets, that might take time quite a bit of time and effort, so I suggest to create a new [issue to pin](https://docs.github.com/en/free-pro-team@latest/github/managing-your-work-on-github/pinning-an-issue-to-your-repository) with the final list, and add some instructions (apart from the general contributing guidelines), to ask contributors to first indicate what they would like to work on, and we could then make sure the next steps are refined.

- Zooming should change the width of the line number column #238
    - It would mostly need changes in the qscintilla editor and logic.py
- Optimising serial REPL processing, to avoid hangs when the serial throughoutput is large
    - PR #1026 already improved things a bit, but  it's possible more can be done
    - Program hangs unresponsive during loop in REPL. #447
- Add a reset Mu option to the admin pane. #466
    - Related to Feature request: Add an 'amnesia' mode so that Mu starts without opening previous project #690
    - I've always found it tedious to find the settings.json file, we could also add a button to open the folder containing the settings?
        - Alternatively we could just show this "show settings" button and note that deleting the file resets Mu
    - Related: Request: clear the logs in the dialog window #545
- Add to the settings panel more options
    - Start-up Switches #693
- Allow users to override locale setting #692
- Investigate / Implement Anaconda/Conda packaging of Mu #535
- Problem running from a Google File stream #574
- Debug show variable of the previous frame #658
- Snap:
    - Snap install mu-editor #718
    - Snap Info Suggestion #719
    - Snap: No CircuitPython on Xubuntu 18.10 #734
    - Snap: mu-editor only works in admin account. (Ubuntu 18.04) #935
- Web Mode:
    - EP2019 BD | Webmode fails when file saved elsewhere #890
        - Related: Flask mode does not copy over example files when path changes. #923
    - EP2019 BD | Webmode hard crash on macOS #889
    - Webmode hard crash when another server is listening on localhost:5000 #897
    - Flask server not shut down on quit #1004
- Mu copies junk to device on startup #872
- make.py vs. make | Inconsistent behaviour #942
- If you stop a running Python3 program from mu, atexit-registered code is not run #984
- Docs: Third Party Packages #1107
- mu on Linux Mint 20 - in case someone will have a problem #1099
- True dark mode #970

carlosperate, 2020-09-29:
Oh, and if anybody has any other suggestion for things to include please mention them here as well!

dybber, 2020-09-30:
Looks good, doesn't conflict with any of my work.

tmontes, 2020-09-30:
...thanks for that @carlosperate, looking great.

I just logged #1122 which may be a good fit too, given that is is UI related and relatively independent of the "internals" that have been changing quite a bit as of recently. What do you say?

carlosperate, 2020-09-30:
> I just logged #1122 which may be a good fit too, given that is is UI related and relatively independent of the "internals" that have been changing quite a bit as of recently. What do you say?

Yeah, that looks good, thanks! I'll add it to the list.

carlosperate, 2020-10-01:
@tjguk have you had a chance to look at the list in case something else might significantly clash with your work?

dybber, 2020-10-03:
Hacktoberfest is now opt-in: https://github.com/digitalocean/hacktoberfest/pull/596

dybber, 2020-10-05:
> Hacktoberfest is now opt-in: [digitalocean/hacktoberfest#596](https://github.com/digitalocean/hacktoberfest/pull/596)

That is, we need to add the repository topic "hacktoberfest", see also https://hacktoberfest.digitalocean.com/hacktoberfest-update

tjguk, 2020-10-05:
Honestly, at this stage, I'm lukewarm at best about this. I appreciate that several of you have taken the trouble to pull out likely issues, and if people want to go ahead then I'm not going to put up a roadblock. But it seems to me that we're not well set up to support any influx of PRs atm. If there's a feeling that the benefit of getting some exposure -- and hopefully some useful contributions -- to Mu is greater, then please carry on. But I would rather give this a miss this time around.

dybber, 2020-10-05:
I still think we should go ahead, I don't think the risk is that bad after what I've read, and my guess is we'll mostly get a few new contributors through our own channels, when sharing this on Twitter, or personal networks.

I don't think it will become unmanageable, I'm happy to help new contributors and dismiss spammers.

tjguk, 2020-10-05:
@dybber That's fine by me. Certainly not against it in principle, and thanks for stepping up to help :)

carlosperate, 2020-10-05:
Yeah, I think the most likely scenario is that we won't get large PRs from this, but hopefully maybe some of the contributors get their feet wet and continue contributing, even if it's just a bit here and there.

Worse case scenario the overhead of preparing stuff is larger than the return, but useful conversations about some features has already emerged from this, and the rest is useful clean up to have done anyway.

Thank you for offering more help @dybber!

We haven't heard from @ntoll yet?

tjguk, 2020-10-05:
@carlosperate He'll probably pick up your mention there, but just in case I've pinged him offline to ask if he wants to weigh in...

ntoll, 2020-10-05:
Ugh... I'm so sorry. My alerts from GitHub (due to "real work") have all been recently siphoned off into a catch-all "github" folder in my email and this thread got lost in there. Thanks @tjguk for the ping. Let me catch up with :arrow_up: .

ntoll, 2020-10-05:
OK... so "Hacktoberfest" has been a bit of a mixed blessing for me... when it kicked off, for another of my projects, I got inundated with around 22 PRs in a single working day, all of which were spammy and, at best, spurious. I have to admit, after that terrible experience and Digital Ocean's, "it's up to the maintainers" stance I have quite a negative outlook on this whole process. I also don't appreciate my repos, time and effort being used in a PR stunt by a company who won't actually support my projects in any meaningful way. At least it's now opt-in, but I'm don't feel that great about being a part of this given the above.

However, helping folks contribute to Mu, mentoring new participants and curating our tickets to make our project friendly to collaborators is definitely a GOOD THING :tm:. Thank you so much @carlosperate for all the work you've done so far on this, and to everyone else for the support and help you've provided.

So the question is, how do we proceed? I'm reticent to support the behaviour of Digital Ocean in hijacking so many open source contributors' time and efforts for their own PR gain. Yet I believe we should do all the things listed above in this issue.

I'd be interested to hear folks take on this.... may I suggest the following "middle way"..? I can easily create a "Not Hacktoberfest" blog post on https://MadeWith.Mu directing folks to the work we're doing on helping folks contribute to Mu. The main https://codewith.mu website and developer docs on https://mu.rtfd.io/ can also be updated to reflect this change.

I hope you agree, we should be living these collaborative/open/supportive practices all months of the year.

Thoughts..?

dybber, 2020-10-05:
Perhaps, it's a good idea to pursue an alternative to achieve the same, as you suggest @ntoll, a blogpost on how to get started contributing to Mu with a short roadmap of what we want achieve in 1.1, but perhaps also marking all issues in this post with a "Good First Issue"-label (and keep using that label in the future). Maybe also inviting people to communication channels such as gitter chat or video meetings, which might be a bit hidden.

carlosperate, 2020-10-05:
I'm happy to continue work in preparation for supporting new contributors and it not being part of Hacktoberfest. As Nicholas has mentioned, all these tasks are a GOOD THING ™️, but realistically it's hard to keep it up all year, since our limited free time fluctuates, and in that regard I have to admit that  Hacktoberfest has been a good kick in the butt to put things in order.

I have indeed also noticed the influx of spam PRs in a few of the repo's I watch (and a couple I manage), and I agree that DO went completely the wrong way by putting all the responsibility on the maintainers. But I have also heard good things from other projects that have had good and valuable contributions as well, so like everything else in life, there is good and bad.

I don't mind either way, I'm happy to do either (or both), but I do think that it'll be easy to run out of steam through the year, so we need to be mindful of that.

ntoll, 2020-10-07:
Hi folks... so given a Hacktoberfest kick in the butt, I guess a positive outcome is the curating from @carlosperate (many thanks!!!).

However, I feel uncomfortable supporting Hacktoberfest. As educators we have to live the lessons we want to teach and practice what we preach. Otherwise, how can we expect learners to follow our example and trust us when we try to get them to do X, Y or Z. It's not a good look to inundate volunteer maintainers of various open source projects with (as far as I can tell) either spammy, needlessly trivial or ill thought out PRs.

I believe we should foster collaborative, friendly and technically coherent practices:

* Before submitting code for a new feature, let's talk about it in an issue (so folks don't waste their time with things that'll never make it into Mu).
* Core maintainers or more experienced developers should mentor first time submitters of a PR with helpful and constructive criticism offered in a friendly manner.
* We should celebrate not only a persons contributions to Mu, but also how they were made: in a friendly and supportive environment that expects and maintains high quality standards of coding. This is done via the [Made with Mu](https://madewith.mu/) blog.

Thoughts..?

carlosperate, 2020-10-11:
Sounds good to me 👍
Do we have any next steps, or just business as usual?
We can probably start tagging "good first contribution" issues, and maybe updating issue/PR templates with this info (we could create a "feature suggestion" issue template). But I assume the rest is just what we've been doing before?

carlosperate, 2020-11-10:
I'll close this issue as it is no longer relevant.

closed 2020-11-10 by carlosperate