mu-editor/mu #1122: Mu should remember UI pane positions and sizes

tmontes, 2020-09-30:
(I could have sworn this topic has been raised in a past issue, but could not find it, so here it goes)

## Facts

* Depending on how it is used, one or more UI panes are displayed within Mu's window: REPL, debugger, charts, and maybe more.
* Qt (and thus Mu) allow users to resize these panes and also re-attach them to differente window sides (example: REPL is shown attached to the bottom but can be dragged to the right in order for it to become more like a "column", instead of a "row").
* In general, such user-driven UI adjustments are not remembered by Mu (some, across restarts, others, across pane close/open cycles).

## Reproducing | Attachment

* Lanch Mu in Python 3 mode.
* Bring up a Python 3 REPL by clicking the REPL button.
* The REPL pane is displayed, attached to the bottom of the window.
* Drag that pane (from its "title bar") towards the right-hand side of the window until it attaches there.
* The result is now a UI with two large columns: editor on the left, REPL on the right.
* Close the REPL and bring it up again by clicking twice on the REPL button.
* The REPL pane is shown attached to the bottom of the window.

### Expected behaviour
* The REPL pane should have been displayed attached to right-hand side and with the same width as when it was closed.

Notes:
* Suspect such re-attachment mis-behaviours are observed for all of Mu's UI panes.

## Reproducing | Sizing

* Launch Mu in Python 3 mode.
* Create a new file with, say, a few `print` statements.
* Initiate a Python 3 debugging session by clicking the DEBUG button.
* Two panes are displayed:
  * The running pane, attached to the bottom.
  * The debug inspector pane attached to the right.
* Adjust the width of the debug inspector pane.
* Adjust the width of the *name*/*value* columns within the debug inspector pane.
* Stop debugging.
* Start debugging again.
* The debug inspector pane is still the same width... (Good!)
* ...the *name*/*value* column widths are some how reset.

### Expected behaviour
* The  *name*/*value* column widths shouln't have been changed from what we have just set.

### Continuation

* Quit Mu.
* Relaunch Mu.
* Initiate the same debugging session as before.
* Two panes are displayed, again, but the width of the debug inspector pane was forgotten.

Expected behaviour:
* The width of the debug inspector pane should have been remembered, along with the width of the  *name*/*value* column within it.

Notes:
* Suspect such sizing mis-behaviours are observed for all of Mu's UI panes.

## Epiloge

The behaviours I'm describing above may not be considered wrong/undesirable by some, even though I can't imagine why. Shall we fix this, or is it ok as is? :)

carlosperate, 2020-09-30:
In the past during workshops it has been incredibly useful that the application panels have stayed the same in between sessions, as a lot of times instructions and screenshots point to specific parts of the editor. If a previous user moves things around it could cause a bit of confusion.

I would guess some classrooms might share the same preference, so I would probably suggest that by default panel position/size changes only last during the current session by default.

tmontes, 2020-10-01:
> In the past during workshops it has been incredibly useful that the application panels have stayed the same in between sessions, as a lot of times instructions and screenshots point to specific parts of the editor. If a previous user moves things around it could cause a bit of confusion.

Agreed, that makes sense and is very important.

> I would guess some classrooms might share the same preference, so I would probably suggest that by default panel position/size changes only last during the current session by default.

Not so sure I agree here. I think consistency should be a top priority from a user experience standpoint.
Along those lines, I'd say:

* If users can move things around, they should stay around across sessions...
* ...otherwise, it might just be better to prevent users from moving them.

Maybe Mu could remember attachment sides, sizes, etc. (much like it remembers its own window size and position) across sessions, as long as we have the "reset Mu button" in #466, which would also reset these user preferences, bringing back Mu to the "standard screenshot size and layout".

Thoughts? :)

dybber, 2020-10-01:
If it's possible to break something for the next one using a computer, someone somewhere will do it. What I'm trying to say is that if it's possible to move panels around, then 1) it shouldn't be possible to completely remove the panels, and 2) it should be easy for the next user to get it back to original settings.

Perhaps remembering attachment side, with only two options is okay (e.g. REPL below editor, REPL to the right of editor), but I'm not even sure I like it if it remembered the size, as someone might make it very tiny, and the next one using it wouldn't recognize that something was wrong, but would just live with terrible ergonomics (e.g. only seeing 3 lines in the REPL)

tmontes, 2020-10-01:
> If it's possible to break something for the next one using a computer, someone somewhere will do it. What I'm trying to say is that if it's possible to move panels around, then 1) it shouldn't be [...]

Agree with everything, especially 0) if its breakable, someone will break it! :)

> Perhaps remembering attachment side, with only two options is okay (e.g. REPL below editor, REPL to the right of editor), but I'm not even sure I like it if it remembered the size, as someone might [...]

Agreed again. Maybe, given that as of now panels are sizeable, we could prevent them from being sized too small/large adjusting current limits, much like it's not possible to size the Mu Editor window down to an unusable size, IIRC. :)

carlosperate, 2020-10-01:
> Not so sure I agree here. I think consistency should be a top priority from a user experience standpoint.
> Along those lines, I'd say:
> 
>     * If users can move things around, they should stay around across sessions...
> 
>     * ...otherwise, it might just be better to prevent users from moving them.

I only partially agree with that.

I think we all agree that keeping the configuration during the same session is really useful, even if we disagree on saving the state between sessions.

Some editors do **not** save panel size on exit, at the very least I can confirm VSCode and PyCharm only save changes made via settings, like moving the panel to different side of the window, but the size of the panel itself is reset between sessions.

It's quite possible the reason i don't mind this specific behaviour is precisely because I got used to it, and I do see the argument for keeping this between sessions, but I really don't feel strongly about this part specifically.

The part I do think it's quite useful, is always having a clearly marked position for things, so that workshops and classrooms don't have to worry about this when explaining what's on everyone's screens. 

If I had to chose between:
- a) Freedom to **move** panels and saving panel **position** between sessions
or
- b) Locking the panel position on the window (not the size),

Then I would probably vote to lock the position as long as it allowed resizing.

Ideally I would love if we could get more teacher feedback on this specific area, as think my reasoning falls in a similar category that the zoom buttons: Great for teachers to have it, while not that important for the rest of users.

Do people agree that, either way, we should break this down in three different tasks?:
- 1 . Remember the changes during the session
- 2 . Easily reset the settings/session 
- 3 . Save the panel positions/dimensions in the session.json file

dybber, 2020-10-01:
My thoughts are similar to @carlosperate, it's nice to be able pull out the school laptops or go to the computer room, and then know that everything will look the same for all computers, without having to instruct individual students to move things around.

> Do people agree that, either way, we should break this down in three different tasks?:
> 
>    1 . Remember the changes during the session
>    2 . Easily reset the settings/session
>    3 . Save the panel positions/dimensions in the session.json file

Yes, I think that's a good solution. At least we aren't disagreeing on the first two, and can go ahead with those.

For the third, perhaps we could start with a boolean in the .json to control where the REPL will be (bottom or to the right), where users manually had to open the settings file to set it.

tmontes, 2020-10-01:
> Do people agree that, either way, we should break this down in three different tasks?:
> 
> * 1 . Remember the changes during the session
> * 2 . Easily reset the settings/session
> * 3 . Save the panel positions/dimensions in the session.json file

Great breakdown. Let's go for 1. and 2. leaving 3. TBD... :)

devdanzin, 2021-01-22:
#1244 allows remembering debugger column width (and which items were expanded) during the session. Should I propose a simple "reopen pane in similar position" PR to move this forward, or would it be better to have a more complete well-designed fix?

Should we try to remember panes grouped as tabs too?
[image]

dybber, 2021-01-26:
Hi @devdanzin, if you mean "reopen pane in similar position" is only for the current session, then yes I think we all agree that would be nice if you moved the REPL, then close and open it again it will open in the same location.

The doubt is whether it should be saved for next session, where it might be another teacher, with another group of kids.