mu-editor/mu #1125: Only show administration tabs relevant for the current mode

dybber, 2020-10-01:
To simplify the administration panel, I suggest that only tabs relevant for the current mode are shown. We currently do this for the ESP Firmware flasher, but might also be relevant to do for:

 - micro:bit tab only shown in micro:bit mode
 - Python3 Environment + Third Party Packages, only in Python 3 and Web mode

At least that's my suggestions, but I'm not sure if that would introduce any problems to hide them? In the future when more and more modes are added it could be come unwieldy.

tjguk, 2020-10-01:
I'm broadly in agreement, but can we hold off on this for now, please? I'm juggling with enough changes to the admin panel as it is for the venv stuff

dybber, 2020-10-01:
Okay, no problem!

commit 3788ae48 (2021-02-14): Merge pull request #1278 from dybber/only_show_admin_tabs_for_current_mode

Hide admin tabs not relevant for the current mode. Fixes #1125
commit e2db37e3 (2021-02-14): Hide admin tabs not relevant for the current mode. Fixes #1125
closed 2021-02-14 by ntoll