mu-editor/mu #1127: Drop Python 3.5 support.

tmontes, 2020-10-04:
(leaving this here as a reminder for future consideration, after having read that 3.5 is no longe supported by the core dev team - not that it needs to be done “now”)

IIRC, supporting 3.5 is (was, hopefully) fundamental in order to allow Mu to run on the previous version of Raspbian. Current Raspbian ships with 3.6, again, IIRC.

If that’s the case, I wonder when we could do this. IOW, how many Mu users are still running an older Raspbian version?

Dropping 3.5 would allow:
* Softening CI load.
* Progressively move to f-strings.

Thoughts? :-)

ZanderBrown, 2020-10-04:
```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
$ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 

```

The final stretch (so 3.5) release was 2019-04-08, and didn't support the Pi 4, is 18 months (or is it 18 years....) long enough? Not sure but given 3.5 stopped getting security fixes last month I would suggest it's fairly reasonable to jump to 3.7

ZanderBrown, 2020-10-04:
> Progressively move to f-strings.

Just on this though it's worth noting we can't use f-strings with gettext