mu-editor/mu #1131: Save As

supuflounder, 2020-10-07:
What were you trying to do?

Save my file after disconnecting the microcontroller

What steps you took to make this happen?

I wanted a "save as", but there is no menu, and only a very limited set of capabilities.

What you expected to happen?

I would find an option to save the file under a new name on a different device.

What actually happened?

I could not save my file, which is exceedingly annoying.

Why is this difference problematic?

Because I wanted to save the source file and the computer was no longer connected.  I should be able to save any file, at any time, any where I want to save it.  This should not be something I should ever be prevented from doing.

Mu 1.0.3, Windows 10.  Formerly connected to a CircuitPython board, board was removed for work.  

This is a serious design error.  There is never any reason to keep me from saving a file under whatever name I want to save it as, whereever I feel like saving it.  In fact, there should always be an option to simultaneously save the file to my hard drive; if the processor is disconnected or dead, I still get my source saved.  Furthermore, I can edit it at any time, whether or not the board is connected, and in a sane world I could, when prompted for a load, load the file from my hard drive, and it it was newer than the file on the board, would replace it, or ask me if I wanted to replace it.  Making the only storage device be one that might vaporize at any instant with no backup is a colossal failure mode.  I want to *always* and *with no exception* have it keep a file on my hard drive, and if no board is attached, save to my hard drive anyway.  Anything less than this is insanity in its worst form: the naive assumption that nothing will ever fail.  Ever.  This is a fantasy world that none of us have ever encountered.

If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

dybber, 2020-10-07:
Thanks for reporting this. I will just add that this relates to previously mentioned issue #724.

carlosperate, 2020-10-07:
In the meantime you can double click on the editor tab and that should offer the "save as" dialogue.

dybber, 2020-11-09:
To make it more discoverable, we could perhaps add a right click context menu, to the Save button, with a "Save as" option? What do you think?

devdanzin, 2020-12-02:
Could this help solve #745 ? Maybe a button with an optional dropdown as in attached image is more discoverable than just a on-hover tip.
[image]

dybber, 2020-12-02:
Sounds like a good idea, but I guess that will require a redesign of the buttons to make room for such dropdown-arrows.

devdanzin, 2020-12-17:
Adding a fancy button seems doable, but (specially since the Save As feature is already implemented) maybe we could try a simple approach first. We could:
* add tooltips to tab titles to the effect of "Double click to save with another name";
* add a mention to renaming to the 'cannot write file' and 'shadow module' error messages, citing the Ctrl+Shift+S shortcut;
* maybe change the file dialog title when called from rename_tab to "Save as";
* maybe add a 'shortcut cheatsheet' modal (from a small button with interrogation mark near the gear?) to improve discoverability of shortcuts.

It would add different paths to discovering the feature, including when it's needed the most.

If any of these is deemed desirable, I can create an issue for each of those, or group them, and submit PRs. Except for the modal which requires a bit of design, others are trivial.

The fancy button route seems to consist of adding a `QToolButton` to our `ButtonBar` via `addWidget`, so it would replace the save action. It would have `ToolButtonPopupMode(1)` so the arrow and dropdown come as part of the package and then... we'd somehow create then attach a menu to it?, that's around when I lost the plot.

supuflounder, 2020-12-17:
The problem is that anyone who uses a computer knows that “save as” is a first-class file option.  Hiding it in obscure places or under double-clicks seems rather user-hostile.  “Renaming” is not seen as the same idea; generally, you rename a file in-place, and if that place has disappeared, you have no way to “rename” it, but you could do a “save as” to another place.  In an effort to simplify things, you should not try to build convoluted substitutions for simple ideas (think of Occam’s Razor).  Simple things should be simple and obvious.  “Save as” is a simple concept.  Double clicking a tab title is not a simple concept, and adds gratuitous complexity to what should be a simple task.  It is not an “intuitively obvious” interface.

 

I’d love to put a JSON-like comment in the front that would allow for all kinds of extensions, such as “keep a shadow copy on my hard drive, here”.  I gave up, though, because the instructions on how to build mu make assumptions that I know subtle details of Python, such as the instruction “create a virtualenv”.  I have no idea what that means, and shouldn’t have to know.

closed 2021-04-10 by carlosperate