mu-editor/mu #1132: Feature request: csv logger

rmhfl, 2020-10-12:
One of the functions i use most in mu (connected to a micro:bit) is the plotter's ability to write data into a `.csv`-file. It would be very nice to be able to switch mu into a mode in which it only writes to a `.csv`, without the need to open and close the plotter. This would be especially useful for tuples that contain a timestamp in the range of milliseconds, which renders the plotter's graphical feedback useless within seconds because the graph representing time grows too fast. 
I think the code for this to be implemented is basically already written in the plotter functions, but since I am pretty new to programming, I lack the abillity to make these bits into a full-blown module.

dybber, 2020-12-07:
It's pretty simple to write directly to a .csv file in your own code. I don't have my micro:bit with me today, but something along the following lines should do it:

```python
from microbit import *

file_handle = open("data.csv", "a")
file_handle.write("header1;header2;header3")

# Log until both buttons are pressed simultaneously
while True:
  x, y, z = accelerometer.get_values()
  file_handle.write("{};{};{}\n".format(x, y, z)

  if button_a.is_pressed() and button_b.is_pressed():
    break

file_handle.close()
```

I agree in the sense that I believe it would be better that logging to a file didn't happen automatically just when the plotter is opened, but a file only is created when requested by the user. My current thought on it would be to add a "Record" button to the plotter. 

I think the issue with displaying time values is a different issue. The plotter UI should perhaps show a checkbox for each "line", which would allow a user to hide lines which aren't interesting (such as your timestamp).

carlosperate, 2021-03-03:
Moving this as a feature request in the GH Discussions section.

closed 2021-03-03 by carlosperate