mu-editor/mu #1134: 在 MAC 10.5.7 上 安装后,启动失败!

yvan918, 2020-10-13:
在 MAC 10.5.7 上 安装后,启动失败! 反复测试都是如此, 点击启动图标后没有任何反应.

angilent, 2020-10-14:
$ cd /Applications
$ sudo open mu-editor.app
试试用上面的方式打开

yvan918, 2020-10-14:
非常感谢回复,按照您的方法终于可以在mac上运行打开 mu-editor了,但是,无法输入固件程序,提示: 没有MICROBIT文件夹的写入权限,但是,我已经授权 mu-editor 软件权限了,不知道是何缘故,请继续指教,我该怎么手动用命令给 mu-editor 授权访问固件? 谢谢

yvan918, 2020-10-14:
补充: 但是: 网页版 mu-editor 却是可以正常刷入 microbit 程序,并且可以正常运行的.

yvan918, 2020-10-14:
非常感谢回复,按照您的方法终于可以在mac上运行打开 mu-editor了,但是,无法输入固件程序,提示: 没有MICROBIT文件夹的写入权限,但是,我已经授权 mu-editor 软件权限了,但是: 客户端软件的 REPL 模式可以链接 microbit , 测试程序可以正常运行的.不知道是何缘故,请继续指教,我该怎么手动用命令给 mu-editor 授权访问固件?  谢谢

angilent, 2020-10-15:
mu-editor 需要设置隐私-> 完全磁盘访问权限
可以试下先链接micro:bit ，然后sudo 启动 mu-editor

我还做的尝试包括：mu-editor 保存一个文件到 MICROBIT 盘里。（确认那个盘是可写的）
我也是这两天开始用mu-editor，没找到问题的根本。跟拍电视📺一样，反正就可以了。 🤪

yvan918, 2020-10-16:
谢谢回复, 现在的问题就是无法授权给 mu-editor 完全的磁盘访问权限, 尝试了多个方法都不行.

carlosperate, 2021-03-03:
Based on google translate it looks like some of the issues here were resolved and the outstanding macOS problems are already fix in the Mu project, but we just haven't had a chance to do a release yet.
I'll close this for now, as everything should be resolved in the next release.

closed 2021-03-03 by carlosperate