mu-editor/mu #1136: Unable to Run/View files on microcontroller when I upload a script that has a bug

dehmoubed, 2020-10-17:
I am teaching a course on Micropython and I am using ESP8266/ESP32 board.

I want to teach my students that they can upload a main.py script onto their microcontrollers by clicking on the files tab on their IDE and dragging and dropping a selected script from “files on your computer” tab to the “files on your device” tab. My goal is to teach students that main.py will automatically run when their board is powered on.  

I am trying to anticipate possible errors that my students will make, so I intentionally uploaded a main.py script that has a bug in it (wrong GPIO pin initialization) to see how the Mu IDE can handle it. Unfortunately, when I flash the code the microcontroller, I get a warning notification window pop up with the following message:

“There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.”

** What Steps you took to make this happen **

1. Write the following code in the editor: 

 (Here is the code that I uploaded):

# Write your code here 
￼
from machine import Pin, PWM
import time

led = Pin(3, Pin.OUT)

while True:
    led.value(not led.value())
    time.sleep(2)

2.  Press the files button on the IDE and drag and drop the file to your microcontroller. 

3. Board is now unrecoverable and unusable in mu-editor. The only way to fix this is to manually flash default MicroPython firmware onto the device (There ins’t a simple and intuitive  option for this on the mu-editor).

* What did you expected to happen:

I was expecting there to be no actuation on the microcontroller when power is connected since there is a bug in the script that is uploaded, but MU IDE would still allow me to remove the script from the microcontroller and proceed normally.

* What actually happened:

The board cannot be connected to, so I can’t even view different files that are on my computer or on my microcontroller. 

* Why this difference is problematic (it may not be a bug!):

If you do this mistake, your board cannot work anymore. Mu does not have an option to reset our firmware to the original micropython firmware.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

I am using MacOS Catalina with the mu-editor version 1.1.0-alpha.2

2020-10-16 21:07:47,402 - mu.logic:1203(sync_package_state) INFO: Synchronize package states...
2020-10-16 21:07:47,403 - mu.logic:1204(sync_package_state) INFO: Old: set()
2020-10-16 21:07:47,403 - mu.logic:1205(sync_package_state) INFO: New: set()
2020-10-16 21:07:55,934 - mu.modes.base:290(toggle_repl) INFO: Toggle REPL off.
2020-10-16 21:07:56,762 - mu.modes.base:261(find_device) INFO: Found device on port: cu.SLAB_USBtoUART
2020-10-16 21:07:56,762 - mu.modes.base:262(find_device) INFO: Serial number: 0001
2020-10-16 21:07:56,775 - mu.modes.esp:181(toggle_files) INFO: Toggle filesystem on.
2020-10-16 21:07:58,911 - mu.modes.base:435(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/base.py", line 432, in ls
    result = tuple(microfs.ls(self.serial))
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/contrib/microfs.py", line 188, in ls
    ], serial)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/contrib/microfs.py", line 139, in execute
    raw_on(serial)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/contrib/microfs.py", line 91, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/contrib/microfs.py", line 72, in flush_to_msg
    raise IOError('Could not enter raw REPL.')
OSError: Could not enter raw REPL.
2020-10-16 21:07:58,916 - mu.interface.main:763(show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.
2020-10-16 21:07:58,916 - mu.interface.main:764(show_message) DEBUG: None
2020-10-16 21:08:01,722 - mu.modes.esp:185(toggle_files) INFO: Toggle filesystem off.
2020-10-16 21:08:02,272 - mu.modes.base:261(find_device) INFO: Found device on port: cu.SLAB_USBtoUART
2020-10-16 21:08:02,273 - mu.modes.base:262(find_device) INFO: Serial number: 0001
2020-10-16 21:08:02,285 - mu.modes.esp:181(toggle_files) INFO: Toggle filesystem on.
2020-10-16 21:08:08,819 - mu.interface.panes:381(dropEvent) INFO: Copying '/Users/philipperivard/mu_code/main.py' to micro:bit.
2020-10-16 21:09:05,286 - mu.modes.esp:185(toggle_files) INFO: Toggle filesystem off.
2020-10-16 21:09:13,120 - mu.modes.base:261(find_device) INFO: Found device on port: cu.SLAB_USBtoUART
2020-10-16 21:09:13,120 - mu.modes.base:262(find_device) INFO: Serial number: 0001
2020-10-16 21:09:13,197 - mu.modes.base:312(add_repl) INFO: Started REPL on port: /dev/cu.SLAB_USBtoUART
2020-10-16 21:09:13,197 - mu.modes.base:293(toggle_repl) INFO: Toggle REPL on.
2020-10-16 21:09:20,615 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/philipperivard/Library/Logs/mu/mu.log
2020-10-16 21:09:20,615 - mu.logic:163(installed_packages) INFO: Packages found: []

Some solutions that I thought would significantly enhance user experience for students would be to have a reset button that would simply reset the firmware of the connected board. For example if you had an ESP32, the IDE would register it and it would flash the ESP32 firmware by default. Similarly, for the ESP8266, it would register the board and reset the firmware accordingly. 

Additionally, I am really curious as to when the new update would be released. MU is the cleanest IDE out there and I would love the possibility of introducing my students to Micropython using MU by the beginning of December.

dybber, 2020-12-07:
To me this seems more like a MicroPython issue, than a Mu issue. When executing a MicroPython program like yours, it's seemlingly not possible to abort it and get back to the REPL (which is required for transferring files).

The problem seems to be this:
 - when executing a program, MicroPython is in raw mode
 - when in raw mode you normally get back to normal mode through CTRL+B
 - when a program is executing, however, this doesn't work
 - only when in normal mode does MicroPython respond to CTRL-C (abort)

carlosperate, 2020-12-07:
I'm not too familiar with the MicroPython ESP port, but when an error is encountered and an exception thrown, doesn't it automatically drop into the REPL if a serial connection is present?

dybber, 2020-12-08:
Either it doesn't drop back to the REPL when a KeyboardInterrupt exception is thrown (CTRL-C) in raw mode, or it just doesn't handle CTRL-C while in raw mode. It also doesn't answer to CTRL-D (soft-reboot), so I suspect it's the latter.

Just checked and I can't replicate the problem on a micro:bit. I will try with an earlier MicroPython firmware, because I don't remember this being a problem a year ago.

dybber, 2020-12-08:
Ah!! Now I found the problem. It's a problem with the code. Pin 3 is the UART RX channel, so when using that for an LED, the ESP can't receive input anymore.

@dehmoubed: Could you try using another pin for the LED?

ntoll, 2020-12-08:
Aha..! :slightly_smiling_face: :microscope: :snake:

carlosperate, 2021-03-04:
@dehmoubed have you been able to try the code changes suggested by @dybber?