mu-editor/mu #1137: Instructions do not work

supuflounder, 2020-10-19:
If you are reporting a bug, we would like to know:
I went to the developer page, and found that I should do

* What you were trying to do,
Follow the developer setup instructions

* What steps you took to make this happen,
Copied a command from the Web page, pasted it into cmd

* What you expected to happen,
It would work

* What actually happened,
The first command worked.  The second command told me that pip did not exist

```
Microsoft Windows [Version 10.0.18362.1139]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\admin>git clone https://github.com/mu-editor/mu.git
Cloning into 'mu'...
remote: Enumerating objects: 10227, done.
remote: Total 10227 (delta 0), reused 0 (delta 0), pack-reused 10227
Receiving objects: 100% (10227/10227), 16.84 MiB | 11.17 MiB/s, done.
Resolving deltas: 100% (7145/7145), done.

C:\Users\admin>pip install -e ".[dev]"
'pip' is not recognized as an internal or external command,
operable program or batch file.
```
* Why this difference is problematic (it may not be a bug!),
If you want me to be a developer, you have to give me instructions that work.  

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Windows 10.  See output from cmd, above.  Version of Mu is irrelevant.

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)
I can't contribute.  If you want people to contribute, give us instructions that *work*.

carlosperate, 2020-10-23:
Hi @supuflounder 

Do you have Python installed in the system already?
There is also Mu installer for Windows that should be a lot more easy to use: https://codewith.mu/en/download

supuflounder, 2020-10-24:
Ah.  The instructions did not say to “install Python”, so in the absence of those instructions I assumed that whatever I did would cause that installation to happen.   I will check and make sure I have a working Python installed.

dybber, 2020-11-23:
I guess it's this page that needs to be updated? https://mu.readthedocs.io/en/latest/

It already mentions "Mu works with Python 3.6 or above.", but I guess it should be more clear that you need to install Python first.

devdanzin, 2020-12-02:
Also https://mu.readthedocs.io/en/latest/setup.html (which mentions **3.5**) goes clone->we work with Python->pip. Maybe we should give a short description of installing Python and creating a venv.

carlosperate, 2020-12-02:
The problem is that there are so many different ways to install Python and pip that it would be impossible to cover all bases.
Apart from that, keeping instructions like that up to date should also be outside of the scope for this project, as it can be a lot of work, and let's remember that these are the **developer** docs and we need to assume some level of familiarisation with the programming environment.

On the other hand, if there is a good resource online that explains all this, then we could just link to that.

devdanzin, 2020-12-02:
I believe making a couple of small helpful fixes gives a good ROI, if we can save time for newcomers without burdening current contributors.

I've found some usable guides, but as you say they're not exhaustive and might ignore nice options like pyenv, chocolatey, etc. So I suggest simply stating "install or make sure you have Python 3.6 or 3.7 installed (versions above 3.7 aren't supported yet).",  probably linking "install" to a [good enough guide](https://realpython.com/installing-python/).

The Quickstart states "Make a virtualenv, then install the requirements:", which I think is enough since there are also many methods of making a venv. Maybe "make and activate it"?

And either in the Quickstart or the Developer Setup page, "use en_GB.utf8 as locale when executing tests or expect failures due to non-matching messages".

Should I propose a PR like that?

dybber, 2020-12-03:
> And either in the Quickstart or the Developer Setup page, "use en_GB.utf8 as locale when executing tests or expect failures due to non-matching messages".

This part would be unnecessary when PR  https://github.com/mu-editor/mu/pull/1151 is merged, I will take a second look at it this weekend and then we can probably merge it.

The rest sounds good to me.

PR #1697 (2021-07-12): Install Python to develop Mu
Fixes #1137 

Documentation update to make it obvious that you need to install Python if you want to actually develop Mu (along with a link to some comprehensive instructions). Also added a link to the `venv` documentation as a bonus.
closed 2021-07-12 by ntoll