mu-editor/mu #1138: Update translations for the v1.1 release

carlosperate, 2020-10-26:
This is to be done **only** when we are hitting the beta releases and the strings are stable. This is to minimise changing text and hopefully simplify the work of the translators.

We might need to reach out to our wonderful community to have a look at the translations at that point.

To help the process when the strings are ready we could also update all the po files to include the updated strings as that could save everybody a step and reduce the room for error.

carlosperate, 2020-10-26:
Related issues:
- Check Chinese translation. #534
- Turkish Translation #598
- Swiss german translation (de_CH) #785
- A french spelling mistake in mu.mo "Ecris ton programme" #967

Related PRs:
- Locale SK #322
- Create mu.po #962 (Ukrainian)
- update of slovak translation #990
- fr translation typo #1041
- Fixes for French localization #1043

So if we do this:

> We might need to reach out to our wonderful community to have a look at the translations at that point.

It might be better to merge some of these PRs first as that would have less conflicts at this stage, then update the translation files, and then ask translators to review them?

xbecas, 2021-09-09:
Hi all. This is my first contribution attempt. @tmontes has been assisting me at setting up a development enviroment so I could start contributing with the Portuguese translation update.

I'm facing the issue that @carlosperate mentioned above, namely trying to come up with an updated .po file.

When I run the `make translate` I get an empty .po file. When I run the `make translateall` I get the following error:
```
(.venv) C:\Docs\GIT\mu\mu.git>python make.py translateall
*** mu\modes\web.py:176: Seen unexpected token "+"
*** mu/modes\web.py:176: Seen unexpected token "+"
Traceback (most recent call last):
  File "c:\users\vasco.mora\appdata\Local\Programs\Python\Python38\tools\i18n\pygettext.py", line 626, in <module>
    main()
  File "c:\users\vasco.mora\appdata\Local\Programs\Python\Python38\tools\i18n\pygettext.py", line 619, in main
    eater.write(fp)
  File "c:\users\vasco.mora\appdata\Local\Programs\Python\Python38\tools\i18n\pygettext.py", line 460, in write
    print('msgid', normalize(k, encoding), file=fp)
  File "c:\users\vasco.mora\appdata\Local\Programs\Python\Python38\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u03c0' in position 480: character maps to <undefined>

New messages.pot file created.
Remember to update the translation stringsfound in the locale directory.
```

Can anyone please assist? My environment is W10 + Python 3.8.10.
Thanks in advance.

carlosperate, 2022-02-23:
V1.1 has been released and we have this GitHub discussion for translations:
- https://github.com/mu-editor/mu/discussions/1931

So we can close this one.

closed 2022-02-23 by carlosperate