mu-editor/mu #1139: Minor bug on Mu 1.1.0.alpha.2 active file tab names get changed  with another tab

shartrooper, 2020-10-27:
[image]
I highlighted the issue with red frames.

I have to clarify this doesn't overwrite actual file's name in any way. In fact, saving changes fix the issue, tab's text gets updated with the corresponding file's name. However, despite I have seen it often, I can't tell what does trigger the issue since I mostly ignore the tabs while I'm writing code.

I'll let you know If I find out how It happens.

My SO is W7 Ultimate edition.
Python version is 3.8.6

carlosperate, 2020-11-03:
Thanks for the report @shartrooper, do you have a way to easily replicate?
I think this might have been fixed in https://github.com/mu-editor/mu/pull/1063, but it's not included in the latest alpha yet.

You can also try the latest builds from master (very alpha, it could contain other bugs!) to see if that resolved this issue: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/

shartrooper, 2020-11-04:
Alright then. I will try the latest build for a while for writing code and then give you feedback whether It happens again or not!.

shartrooper, 2020-11-06:
Three days in a row coding with the latest alpha build so far and seems like the bug isn't there anymore. So I think it's fixed.

closed 2020-11-06 by shartrooper

carlosperate, 2020-11-09:
Great, thanks @shartrooper!

Derek-microbit, 2020-11-09:
SHUT UP AND STOP SPAMMING ME. I KEEP UNSUBSCRIBING BUT U KEEP SENDING ME
STUFF!!!!!!!

Derek-microbit, 2020-11-09:
how do I unsubscribe from u cause I keep doing it and it doesn't doesn't it cause ur blowing up my phone like crazy and it's annoying

Derek-microbit, 2020-11-09:
no offense