mu-editor/mu #1142: The compatibility of micropython pyboard

CaptainJackey, 2020-11-10:
Why Mu still not compatible with micropython (pyBoard) ? Will fix it soon?

dybber, 2020-11-10:
Hi @CaptainJackey 
Thank you for the reminder, there's already a pull request at #650, which I've promised to look at. I will take a look this week and see if we can get it in a state where we can merge it.

CaptainJackey, 2020-11-10:
Great, waiting for your good news!!

dybber, 2020-11-10:
Seems like just some very small changes are necessary, to get it running and then we can merge it into Mu. See https://github.com/kyoungmeyer/mu/pull/1

closed 2020-12-05 by dybber

CaptainJackey, 2020-12-07:
@dybber Already add to the latest ALPHA  version?

dybber, 2020-12-07:
Hi @CaptainJackey, you can try it through the most recent development build: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=

CaptainJackey, 2020-12-08:
> Hi @CaptainJackey, you can try it through the most recent development build: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=

Love you so much!