mu-editor/mu #1143: Can we support VIM?

yingshaoxo, 2020-11-10:
The only drawback of this IDE that I can think of is that it's missing a vim plugin.

Or, at least support hot reload if that file get changed outside.

For example, use vscode to write code, use mu to upload or run code.

dybber, 2020-11-10:
Mu is intended to be a good beginner editor, e.g. for school children, not an editor for experts, so I think VIM style keybindings will just introduce confusion.

As I understand it, it seems like what you really need is a plugin for VSCode for interacting with whatever MicroPython/CircuitPython-device you are using.

yingshaoxo, 2020-11-10:
https://marketplace.visualstudio.com/items?itemName=RT-Thread.rt-thread-micropython

This plugin does well at interacting with micropython boards, but it is closed-source.

ntoll, 2020-11-10:
For what it's worth, I use vim to develop Mu. Vim key-bindings may (seriously) make a cute "Easter egg" feature only available on 1st April. :-)

dhalbert, 2020-11-10:
Another VSCode plugin, specifically for CircuitPython: https://marketplace.visualstudio.com/items?itemName=joedevivo.vscode-circuitpython

CaptainJackey, 2020-11-11:
Keep Mu Simple, Please!

dybber, 2020-11-11:
Agree, @CaptainJackey!

yingshaoxo, 2020-11-11:
> Another VSCode plugin, specifically for CircuitPython: https://marketplace.visualstudio.com/items?itemName=joedevivo.vscode-circuitpython

To be honest, I can't understand how that plugin works after reading its document.

It's quite complex without UI support.

So I'll still use this for now: https://marketplace.visualstudio.com/items?itemName=RT-Thread.rt-thread-micropython

yingshaoxo, 2020-11-11:
> Keep Mu Simple, Please!

Support for vim is something unnecessary.

But support hot-loading when the file got changes by other editors, it's acceptable.

dybber, 2020-11-11:
Okay, could you change the title of the issue then?