mu-editor/mu #1145: App doesn't open on macOS

ajs256, 2020-11-14:
* What you were trying to do **Open the app**
* What steps you took to make this happen **Open the `Contents/MacOS/mu-editor` executable from the Terminal**
* What you expected to happen **The app to open**
* What actually happened **I got the following traceback**
```
greeble:~ $ /Applications/mu-editor.app/Contents/MacOS/mu-editor
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')()
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 480, in load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2324, in load
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2330, in resolve
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/QtCore.abi3.so, 2): Symbol not found: _futimens
  Referenced from: /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore
```

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
 * macOS version: **10.12.5**
 * Mu version: **1.0.2**

tmontes, 2020-11-17:
@ajs256,

Thanks for using Mu (or trying to, per your report)!

It turns out that Mu requires macOS High Sierra (10.13) or later. That's why you're getting that failure (note the third line from the bottom saying "which was built for Mac OS X 10.13").

We're sorry that it hasn't worked for you. Do you think you can try it out on a more recent macOS version? (note: the most recent one, Big Sur, is not expected to work, either, being less than a week old -- but we are keeping an eye on that!)

Thanks again.

ajs256, 2020-11-17:
Oops, between the time I opened this issue and your reply, I upgraded to Big Sur! I no longer need this issue.

closed 2020-11-17 by ajs256

ajs256, 2020-11-18:
FYI, I'm now watching #1147 instead of this one.