mu-editor/mu #1146: Line too long (xx > 79 characters) after click button tidy.

fraramon, 2020-11-15:
Hi, just when I click on tidy, I get a lot of line overrun errors that I can't eliminate.
Thanks for everything...

devdanzin, 2020-12-02:
Hello, thank you for the report! 
I have suffered from this, but it turned out that clicking the "Check" button again made the errors disappear. Took me a while to figure it out, then later I saw another user with the same issue.

Can you test whether clicking the "Check" button makes the error messages disappear (and reappear if  clicked again)? Does clicking on Tidy make the errors disappear?

If you case is the same as mine, I can open an issue about making easier to understand the button behavior (like change colour when Check is active).

dybber, 2020-12-07:
@fraramon Could you post some example code? 

I can't replicate this behaviour. For me it seems tidy does indeed work, unless I do something like the following, where tidy is unable to find a place to wrap the line:

```
a_very_long_variable_name_which_tidy_is_unable_to_shorten_to_something_below_79_chars = 1
```

devdanzin, 2020-12-08:
It's possible to get check to complain and tidy not to help with this code, which would seem to have two kinds of wrappable sites:

```python
import random
colors = ["red", "orange", "yellow"] + ["gray", "black", "blue", "purple", "green"]
color = random.choice(colors)
```

Oh, it's from the settings we use for black in tidy_code, this fixes it:

```diff
diff --git a/mu/logic.py b/mu/logic.py
index bdc4cfb..a43647a 100644
--- a/mu/logic.py
+++ b/mu/logic.py
@@ -1839,7 +1839,7 @@ class Editor(QObject):
             source_code = tab.text()
             logger.info("Tidy code.")
             logger.info(source_code)
-            filemode = FileMode(target_versions=PY36_VERSIONS, line_length=88)
+            filemode = FileMode(target_versions=PY36_VERSIONS, line_length=79)
             tidy_code = format_str(source_code, mode=filemode)
             # The following bypasses tab.setText which resets the undo history.
             # Doing it this way means the user can use CTRL-Z to undo the
```

While 88 was a deliberate choice in #687, maybe check should accept longer lines to avoid a gap between check and tidy.

dybber, 2020-12-08:
Ah, good find! 

I think I agree, instead of your patch we should increase max line length for Pycodestyle to allow up to 88 chars, and let the number be configured as a single variable used for both tidy and check, so we don't get this problem again if we redecide and want to set it to 79 at some point and forget to change it at both locations.

/Martin

PR #1189 (2020-12-19): Make check and tidy use the same maximum line lenght. Fixes #1146.
Add another constant to logic.py and use it to configure maximum line lengths for both black and pycodestyle, attempting to fix #1146. This is a clear candidate for going into conf.py and settings in the future. Might be a little test-heavy.

Open to and welcoming any suggestions, criticisms, alternative designs and requests.
commit 76fa6063 (2020-12-19): Make check and tidy use the same maximum line lenght. Fixes #1146. (#1189)

* Keep maximum line length in sync for check and tidy.

* Move MAX_LINE_LENGTH to config.py, skip tests in Python <3.6 because we need newer version of mock.
closed 2020-12-19 by tjguk