mu-editor/mu #1147: mu-editor hangs on macOS Big Sur

ged99, 2020-11-16:
I have been using **mu-editor** successfully for a while under **Catolina** on my Mac. Today I upgraded the OS to **Big Sur**. Now the mu-editor does not open. When I launch it by clicking on the icon, the app just hangs. There are no error messages. 

I have to close the application through the "Force Quit Applications". In this window it shows that mu-editor is running but not responding.

I re-installed the editor with the latest version from the website, but this did not work either.

If you are reporting a bug, we would like to know:

* What you were trying to do: Launch the mu-editor
* What steps you took to make this happen: click on its icon
* What you expected to happen: Editor to open
* What actually happened: editor just hung - no error messages
* Why this difference is problematic (it may not be a bug!): The pp no longer works and I cannot run Python!!
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
macOS Big Sur V 11.0.1
mu_editor - the latest version (1.0.3? - hard to tell!)

Can't provide a log file as the editor does not open.

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

ged99, 2020-11-17:
i have attached a screen shot showing that it is using cpu time while hanging. Also the Mac Dump Log that is generated when I do a **Forced Quit**
[image]
[Mu_dump.txt](https://github.com/mu-editor/mu/files/5551513/Mu_dump.txt)

aivarannamaa, 2020-11-17:
Could it be because of OCSP servers overload (https://arstechnica.com/gadgets/2020/11/macos-big-sur-launch-appears-to-cause-temporary-slowdown-in-even-non-big-sur-macs/) ?

ajs256, 2020-11-17:
I have this issue too. 
<details>
<summary>My logs</summary>
Starting Mu 1.0.3
2020-11-17 11:24:25,880 - root:113(run) INFO: uname_result(system='Darwin', node='greeble-mi.local', release='20.1.0', version='Darwin Kernel Version 20.1.0: Sat Oct 31 00:07:11 PDT 2020; root:xnu-7195.50.7~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2020-11-17 11:24:25,880 - root:114(run) INFO: Python path: ['/Users/ajs256/Applications/mu-editor_stable.app/Contents/MacOS', '/Users/ajs256/Applications/mu-editor_stable.app/Contents/Resources/python/lib/python36.zip', '/Users/ajs256/Applications/mu-editor_stable.app/Contents/Resources/python/lib/python3.6', '/Users/ajs256/Applications/mu-editor_stable.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Users/ajs256/Applications/mu-editor_stable.app/Contents/Resources/python/lib/python3.6/site-packages', '/Users/ajs256/Applications/mu-editor_stable.app/Contents/Resources/app', '/Users/alex/Applications/mu-editor_stable.app/Contents/Resources/app_packages', '/Users/ajs256/Applications/mu-editor_stable.app/Contents/Resources/app_packages/IPython/extensions']
2020-11-17 11:24:25,880 - root:115(run) INFO: Language code: en_US
2020-11-17 11:24:26,136 - mu.logic:543(__init__) INFO: Setting up editor.
2020-11-17 11:24:26,136 - mu.logic:562(__init__) INFO: Settings path: /Users/ajs256/Library/Application Support/mu/settings.json
2020-11-17 11:24:26,136 - mu.logic:563(__init__) INFO: Session path: /Users/ajs256/Library/Application Support/mu/session.json
2020-11-17 11:24:26,136 - mu.logic:564(__init__) INFO: Log directory: /Users/ajs256/Library/Logs/mu
2020-11-17 11:24:26,136 - mu.logic:565(__init__) INFO: Data directory: /Users/ajs256/Library/Application Support/mu
2020-11-17 11:24:26,149 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2020-11-17 11:24:26,288 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/ajs256/mu_code
2020-11-17 11:24:26,289 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/ajs256/Library/Application Support/mu/session.json
2020-11-17 11:24:26,289 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'circuitpython', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': 146, 'y': 90, 'w': 1152, 'h': 720}}
2020-11-17 11:24:26,289 - mu.logic:1107(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
</details>

carlosperate, 2020-11-18:
Thanks for the reports everybody!

Unfortunately the log attached doesn't show any issues.

Could anybody experiencing this issue try to run Mu from source code? https://mu.readthedocs.io/en/latest/setup.html
This way we should hopefully see any errors and traces in the terminal.

In the past macOS updates usually break PyQt5, so I wouldn't be surprised if this is the culprit in the case as well. We should have a look at the mailing list.

In the meantime I found this: https://forums.macrumors.com/threads/pyqt5-and-big-sur.2260773/
If somebody is having this issue and can run Mu from source, can you try running adding this env variable first and then running Mu?
```
$ export QT_MAC_WANTS_LAYER=1
$ python run.py
```

ajs256, 2020-11-20:
@carlosperate Setting `QT_MAC_WANTS_LAYER` worked!

ged99, 2020-11-20:
I tried the suggested given above, but did not get anywhere. If someone would like to help me, then I will try to run this and provide whatever diagnostics I can. Please see my efforts below.

Running 
$ export QT_MAC_WANTS_LAYER=1
in a terminal and then trying to run the mu-editor from an I can still leaves it hanging.

[image]

ajs256, 2020-11-20:
@ged99 A few tips: (no offense intended)
* `git clone` put everything into `~/mu`. Move there with `cd mu`, then try using `pip`. That's why `pip`'s returning an error.
* The `$` is used to indicate something should be run at the shell. It should not be used as part of your shell command.

kurbonas, 2020-11-21:
@ajs256 having the same issue. 

Your instruction is unclear. Could you write a more detailed guide?

Thank you!

ajs256, 2020-11-21:
Here’s my best guide for everything you need to do:

1. Download Xcode and agree to the license. (Some of the tools you need are bundled with Xcode.)
2. In a terminal, type `git clone https://github.com/mu-editor/mu.git`
3. Move to the folder Mu downloaded to with `cd mu`
4. Type `python3 -m venv -p /usr/bin/python3 ~/mu-venv` to create a virtual environment.
5. Type `source ~/mu-venv/bin/activate` to activate the venv you created
6. Install the dependencies with `pip3 install -e ".[dev]"`
    7. If that returns an error about SDL, follow these instructions:
    8. Install Homebrew following the instructions at [brew.sh](brew.sh)
    9. Run `brew install sdl sdl_gfx sdl_image sdl_mixer sdl_ttf` (I’m not sure which of those it actually needed)
    10. Try running the pip3 command above again
11.  Set the variable Qt needs with `export QT_MAC_WANTS_LAYER=1`
12. Run Mu with `python3 run.py`

If anything was unclear there, let me know. If you want to exit the virtual environment, type `deactivate`.

kurbonas, 2020-11-21:
@ajs256 

Thanks, mate!

You are the best!

keyra-coder, 2020-11-23:
Can reproduce the issue above. Managed to  start Mu from the terminal after setting Qt variable. None of the previously successful programs run in Pygame Zero mode with the following error:
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
libpng warning: iCCP: known incorrect sRGB profile

BobCochran, 2020-11-29:
I have this same issue as well on Big Sur. I think the code base should be refactored to avoid this type of breakage each time Apple updates Mac OS.

ged99, 2020-11-29:
> 4\. python3 -m venv -p /usr/bin/python3 ~/mu-venv

Trying to follow your instructions. Latest Xcode installed. Here is a screen shot of where I got to
[image]

ajs256, 2020-11-29:
@ged99 Try running it like `python3 -m venv ~/mu-venv` (without the -p argument).

ged99, 2020-11-29:
Making progress! But now it complains that 'setup.py' does not exist. But there it is, you can see it in the mu directory.
[image]

ps - thanks for your help! Really appreciate it.

ged99, 2020-11-29:
Making progress. Got as far as running 
pip3 install -e ".[dev]"
[image]

Then it stopped at this point.
[image]

ged99, 2020-11-29:
@ajs256 Making progress bit-by-bit. I got as far as this before I got an error.

BTW, thanks for your help! Much appreciated 

[image]

ajs256, 2020-11-29:
@ged99, try following steps A, B, and C to install SDL. That's what fixed that error for me.

ntomlin, 2020-12-05:
Thanks @carlosperate, was getting ready to try the steps above even though way over my head, but then saw your post and just switched to the alpha version - problem solved!

ged99, 2020-12-06:
The Alpha Version!!! Yes, this works. Shame that I didn't find out about this until I saw @ntomlin comment. I'd been battling with this for **weeks**.

Anyway, happy once again!

dybber, 2020-12-09:
As pointed out by @ZanderBrown on Gitter, the newest PyQt5 version sets the `QT_MAC_WANTS_LAYER=1` variable automatically: https://riverbankcomputing.com/news/PyQt_v5.15.2_Released

However, upgrading PyQt also means that we will have to say farewell to macOS 10.12 Sierra, as PyQt 5.15.2 only supports 10.13, 10.14, 10.15 (Mojave, Catalina, Big Sur)

carlosperate, 2020-12-09:
I thought the version of PyQt5 we have right now already dropped support for Sierra? And it's only 10.13 High Sierra or newer?

Either way, unless we find any bugs, I think I would prefer to stay in an older version of PyQt5 that supports an older macOS and manually add the env variable in the Mu source code. At least until we are force to update for any other reasons.

We get a few issues opened for Mu not being able to run in older macOS versions, and I think it's worth the trouble to try to maximise the versions Mu supports.

dybber, 2020-12-09:
This archived website says that 10.13 supports 10.12, 10.13, 10.14, but it might be outdated: https://doc.qt.io/archives/qt-5.13/supported-platforms.html

I agree that we should try adding the variable ourselves. I haven't dared upgrading to Big Sur, so I'm not able to test whether it works if I try to fix it.

ged99, 2020-12-09:
@dybber - I'm already fully committed to Big Sur. And it's been frustrating getting Mu to work on it. The Alpha version runs ok, but there is a problem importing libraries (as you know). I would be happy to help test anything out within reason.

I am having to use Mu on my Windows m/c (which is only until I get the Mac sorted)

BobCochran, 2020-12-10:
Like @ged99, I am fully committed to Big Sur and would appreciate a
convenient solution which would allow Mu to work properly on this OS. I buy
Adafruit Industries products and have been dabbling with CircuitPython. The
company urges customers new to CircuitPython to use the Mu editor. I would
guess there are a lot of people who are getting their first programming
experience using Python, and many will be on an up-to-date Mac OS. Mu
should be fully supported on all versions of MacOS.

dybber, 2020-12-10:
@ged99 @BobCochran @ntomlin @kurbonas @ajs256: Could any of you test this development version of Mu? https://github.com/mu-editor/mu/actions/runs/413065090 

It incorporates the `QT_MAC_WANTS_LAYER=1`, but as I haven't upgraded to Big Sur myself, I'm not able to confirm whether it works as expected.

ged99, 2020-12-10:
Thankyou Martin. I appreciate your efforts.

I downloaded the file, but cannot run it!!! Arrrgh. It says I do not have the permissions to run the app, but I am the sys admin! Here are the screen shots - all looks OK but it refuses to run.

dybber, 2020-12-10:
That's another known issue, could you try opening with Right-click -> Open? (or CTRL-click -> Open)

ajs256, 2020-12-10:
@dybber, that didn't work for me. What _did_ fix that problem was using `chmod +x` on the executables inside the Mu application. However, that still doesn't fix the big problem of Mu not running. Here's what happens when I try to run from the terminal:
```sh
% pwd
/Users/ajs256/Downloads/mu-editor 2.app/Contents
% ./MacOS/mu-editor 
./MacOS/mu-editor: line 2: /Users/ajs256/Downloads/mu-editor 2.app/Contents/MacOS/../Resources/python/bin/python3: Permission denied
./MacOS/mu-editor: line 2: exec: /Users/ajs256/Downloads/mu-editor 2.app/Contents/MacOS/../Resources/python/bin/python3: cannot execute: Undefined error: 0
% sudo !!
sudo ./MacOS/mu-editor
./MacOS/mu-editor: line 2: /Users/ajs256/Downloads/mu-editor 2.app/Contents/MacOS/../Resources/python/bin/python3: Permission denied
./MacOS/mu-editor: line 2: exec: /Users/ajs256/Downloads/mu-editor 2.app/Contents/MacOS/../Resources/python/bin/python3: cannot execute: Undefined error: 0 
```

dybber, 2020-12-10:
That's strange. It works on Mojave, but Apple have added various security protocols in Catalina and Big Sur, that makes it much harder to ship software for Mac, and which are probably causing this. 

These issues are being worked on, but I'm not directly involved, so I can't be of much help here. Perhaps you could try also doing a `chmod +x ./Resources/python/bin/python3`, but I'm not sure that's enough either.

ajs256, 2020-12-10:
That worked 🤦 

I think the build process just has to make sure the executables are, well, executable and it'll work!

dybber, 2020-12-10:
@ajs256: They are actually `chmod +x` on my machine when I download it (I'm not on Big Sur), what I suspect is that Big Sur has a security mechanism that does a `chmod -x` on all executables when downloaded, if they aren't properly signed and notarized.

dybber, 2020-12-10:
But it's great news that it works though, I think this means that when the signing/notarization process is in place and my PR is merged this will also be fixed.

ajs256, 2020-12-11:
Huh...I'm not sure that's what's causing it.
I tried with a different app (PiBakery):
[image]
and [WhatsYourSign](https://objective-see.com/products/whatsyoursign.html) reports it's unsigned, but:
```sh
% cd ~/Downloads/PiBakery-darwin-x64.app
% ls -l Contents/MacOS/PiBakery # the executable
-rwxr-xr-x@ 1 alex  staff  8608 Jun 29  2018 Contents/MacOS/PiBakery
```
It's still executable! Must be something weird with Github CI.

dybber, 2020-12-11:
Weird. I'm not sure either, but whatever is causing it, I'm pretty confident that it will be solved when we will shift our packaging infrastructure to use @tmontes' work on https://github.com/mu-editor/pup

JakeWoodwardIRL, 2020-12-15:
Is this bug still under review? I haven't found that any of these workarounds have stopped Mu from "hanging" whenever it is launched. Currently running MacOS Big Sur 11.0.1.

ged99, 2020-12-15:
> Is this bug still under review? I haven't found that any of these workarounds have stopped Mu from "hanging" whenever it is launched. Currently running MacOS Big Sur 11.0.1.

The "alpha" version on the mu website definitely works for me. The main stable version hangs. 

The only problem for me is that I cannot get it to import external modules, in particular Matplotlib which I need for one application I am working on. These are known bugs but not sure when they will be fixed.

I must admit that I am now turning to the 'dark side' and using another IDE which works perfectly under Big Sur.

dybber, 2020-12-15:
> Is this bug still under review? I haven't found that any of these workarounds have stopped Mu from "hanging" whenever it is launched. Currently running MacOS Big Sur 11.0.1.

Yes, it is still not closed, and we're waiting for confirmation that the proposed solution works (see https://github.com/mu-editor/mu/issues/1147#issuecomment-742558052). However, as people seems to have trouble opening that version and confirming that it solves the problem, we aren't moving forward with this before the Mac OS signing/notarization/packaging issues are resolved (which is a different issue entirely)

ajs256, 2020-12-26:
Hey there @dybber (or anyone else), maybe you can help me with this.

I tried to `git pull` and then update the dependencies, but it seems that the version of `PyQt5` it requires is no longer on PyPI.

Here's my output:
```
ajs256@greeble mu % pip3 install -e ".[dev]"
Obtaining file:///Users/ajs256/code/mu
Collecting qtconsole==4.7.4
  Using cached qtconsole-4.7.4-py2.py3-none-any.whl (118 kB)
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting flake8>=3.8.3
  Using cached flake8-3.8.4-py2.py3-none-any.whl (72 kB)
Collecting appdirs>=1.4.3
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting semver>=2.8.0
  Using cached semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting virtualenv
  Using cached virtualenv-20.2.2-py2.py3-none-any.whl (5.7 MB)
Requirement already satisfied: wheel in /usr/local/lib/python3.7/site-packages (from mu-editor==1.1.0a3) (0.34.2)
ERROR: Could not find a version that satisfies the requirement PyQt5==5.13.2 (from mu-editor==1.1.0a3) (from versions: 5.14.0, 5.14.1, 5.14.2, 5.15.0, 5.15.1, 5.15.2)
ERROR: No matching distribution found for PyQt5==5.13.2 (from mu-editor==1.1.0a3)
```

What should I do here?

dybber, 2020-12-30:
@ajs256 Good that you found a solution to your problem. Can you confirm that the solution in #1147 works?

ajs256, 2020-12-30:
I think I'm running into the same problem as @ged99, where the executables are not executable and need a `chmod +x` to run. I can confirm that running `chmod +x` on the files in `[APP]/Contents/MacOS/*` and  `APP/Resources/python/bin/*` fixes the problem.

kyleplo, 2020-12-31:
The alpha version on the download page works for me

dwestbury, 2021-01-25:
@ajs256 I was able to get mu-editor to work based on the advice above:
https://github.com/mu-editor/mu/issues/1147#issuecomment-731609561

But, my setup on macOS Big Sur v11.1 required a couple additional steps:
1) `brew install pyqt5` this was missing
2) `brew install python@3.7` I was running 3.9, which was causing problems
3) `brew unlink python@3.9 && brew link python@3.7` changed my default python3 version to 3.7 

Additionally, I still can't get it to launch from the mu-editor desktop icon, instead I had to write a quick shell script, which seems to work reliably, once all your aforementioned preparation steps have been completed:

```
## launching Adafruit Mu-Editor

cd ~/mu
python3 -m vent ~/mu-venv
source ~/mu-venv/bin/activate
export QT_MAC_WANTS_LAYER=1
python3 run.py

```

Frederic-soft, 2021-01-26:
I just installed mu-editor on Big Sur using `brew install mu-editor`, I edited the file `/Applications/mu-editor.app/Contents/MacOS/mu-editor` to change the second line into:

`"exec" "env" "QT_MAC_WANTS_LAYER=1" "`dirname $0`/../Resources/python/bin/python3" "$0" "$@"`

and it works fine.

Frederic-soft, 2021-01-26:
Sorry, in my previous post, the backticks around:
```
`dirname $0`
```
have been removed by the markdown rendering process.

The line should read:
```
"exec" "env" "QT_MAC_WANTS_LAYER=1" "`dirname $0`/../Resources/python/bin/python3" "$0" "$@"
```

carlosperate, 2021-01-26:
> brew install mu-editor

This is new to be, didn't realised there is a brew formula for Mu: https://formulae.brew.sh/cask/mu-editor
Does anybody know how is the maintainer?

dwestbury, 2021-01-26:
Terrific. Didn't even consider installing it as a Homebrew managed package. That approach worked well.
I did have to make the changes `/Applications/mu-editor.app/Contents/MacOS/mu-editor`, as suggested above by @Frederic-soft, but that was quick and painless enough.

Cheers!

dybber, 2021-01-29:
The problem was resolved by #1171, which I tested and confirmed yesterday on a machine running Big Sur.

To everyone here: thanks for the reports, the problem should be solved in the next release.

closed 2021-01-29 by dybber

DrNebin, 2021-02-08:
This does not seem to be fixed. I am on OS Big Sur and downloaded and tried to run Mu today. The program hangs when it attempts to load and never actually starts. I will try attaching two screenshots if I can.... also note there seems to be no way for me to bypass the security denial (right clicking does not bypass it) and I have to go to settings and get to the "open anyway" button to have it make an attempt to load.... then it hangs and the program doesn't open.

[image]
[image]
[image]

dybber, 2021-02-09:
The issue is solved, we just haven't been able to release a new version of Mu for Mac since the fix was made, I'm sorry.

skarlekar, 2021-02-21:
If you are running into Mu Editor hanging on Big Sur, all you have to do is set the environment variable QT_MAC_WANTS_LAYER=1 at the terminal and open the mu-editor app from the terminal as follows:

```
export QT_MAC_WANTS_LAYER=1
open /Applications/mu-editor.app/
```

Just place the above in a shell script, make it executable and run the script from the terminal.

BananaDok, 2021-02-23:
@dybber When can we expect the release with the resolved issue? I am just wondering if I should try to do a work around or wait, thanks!

beebopkim, 2021-03-04:
> If you are running into Mu Editor hanging on Big Sur, all you have to do is set the environment variable QT_MAC_WANTS_LAYER=1 at the terminal and open the mu-editor app from the terminal as [...]

I just put a line - os.environ.putenv('QT_MAC_WANTS_LAYER', '1') - in mu-editor.app/Contents/MacOS/mu-editor and it works very well in Big Sur.
[image]

iadesdragon, 2021-03-10:
@beebopkim Yep, adding that line to mu-editor works like a charm! Thanks a lot for this helpful LiveHack :-)

alsauter, 2021-05-31:
> @beebopkim Yep, adding that line to mu-editor works like a charm! Thanks a lot for this helpful LiveHack :-)

Hi, could you please go through the process step by step? is it just a line on terminal...? or where? appreciate detail help as I am new to mu editor and using the terminal to run it...but need to get it to work on Mac OSX Sur
Thank you

Is it the same as this? @skarlekar 

> If you are running into Mu Editor hanging on Big Sur, all you have to do is set the environment variable QT_MAC_WANTS_LAYER=1 at the terminal and open the mu-editor app from the terminal as [...]

carlosperate, 2021-05-31:
Hi @alsauter,
These workaround should no longer be neccesary if you are using the latest beta releases: https://github.com/mu-editor/mu/releases

If you are still having issue in Big Sur with beta.4 could you open a new issue detailing the problem?
Thanks!

alsauter, 2021-05-31:
Thanks @carlosperate ::

so that means I only need to download: Mu.Editor.1.1.0b4.dmg and it should work fine on Big Sur? To tell you the truth I am not on Big Sur, I have to fix this problem for a student (that does use Big Sur) using Mu Editor to program the Circuit Playground Express with Circuit Python. So I am trying to go ahead and know as much as I can before helping the student remotely.

Thank you so much.

carlosperate, 2021-05-31:
Yes, using beta.4 and these steps should get you set up in Big Sur: https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546