mu-editor/mu #1148: editor won't open, says i don't have python 3.6 but i have 3.9

Clerudx, 2020-11-20:
hello, i am extremely sorry if this a stupid question but i am minutes new to programming.
i just downloaded both mu and python and whenever i try to open mu it says "requested python version 3.6 is not installed" but i have 3.9 python. does this mean only ver 3.6 is allowed? or is there a step that i jumped.

tjguk, 2020-11-20:
@Clerudx can you explain how you're using Mu, please? I don't think we have a check for a Python version -- although I could be wrong -- but I'm not sure whether you're running Mu from a downloaded version (ie from https://codewith.mu/en/download) or whether you're running from source inside a virtual environment.

It might also help to know what platform you're on (eg Windows, Mac, Linux)

Clerudx, 2020-11-20:
I have windows 10. 
I use mu by going to where i downloaded it and then running mu launcher.

I did download it on code with mu

abixadamj, 2020-11-21:
Try to reinstall MU, but place exactly directory for example `C:\MU_TEST `

carlosperate, 2020-11-27:
> I don't think we have a check for a Python version

@tjguk Not at the moment, but this might change soon (at the `pip install` stage) since we have quite a few tickets trying to run in newer Python versions with incompatible pinned dependencies: https://github.com/mu-editor/mu/pull/1108/

>I have windows 10.
> I use mu by going to where i downloaded it and then running mu launcher.
> 
> I did download it on code with mu

@Clerudx If you download mu from https://codewith.mu it doesn't need Python to be installed as well, as it comes with it's own Python.

abixadamj's suggestion might be the fastest way to fix it, but if you could take a screenshot of the error you are getting before doing that, that would help us.

@abixadamj is there a specific reason you suggest to installed on the C:\ root? Have you seen similar issues with Mu?

dybber, 2021-02-01:
My guess is that @abixadamj was running the `Mu.launch.pyw`, from the install location, but not using the Desktop shortcut. The Python version to use is AFAIK defined within the desktop shortcut, not the launcher, so when double-clicking the `Mu.launch.pyw` it will instead look for the system installed Python version.

I've had this problem with students where the Desktop-shortcut wasn't created, or where they had deleted it. It's a messy process to recreate that shortcut, where I had to find another Windows computer and copy the "Target" for the shortcut from there. It would be nice if there just was a .exe file in the Mu install location, because then it would also be easy to recreate the shortcut (right click -> Create shortcut)

mercatian, 2021-02-17:
Hi, I am also new to programming and I am having these issues as well. I installed 3.9, then went online to install mu, then when I launch it, it says "requested python version (3.6) not installed" and all you can click is ok. I first downloaded everything to my desktop, then uninstalled and downloaded to a folder, tried to launch from the folder, then tried to create a desktop shortcut of the launcher and launch with that, but everything resulted in the same error. I tried to read the suggestions, but i am afraid i am still a little lost. any help would be appreciated.

carlosperate, 2021-02-17:
Hi @mercatian,
> hen when I launch it, it says "requested python version (3.6) not installed" and all you can click is ok. 

To better understand the issue, how exactly did you launch Mu? Was it from a desktop shorcut, a start menu shortcut, or something else?

> I first downloaded everything to my desktop, then uninstalled and downloaded to a folder, tried to launch from the folder

Is this the "portable" version? What did you do to try to launch it from the folder?

> then tried to create a desktop shortcut of the launcher and launch with that, but everything resulted in the same error. 

What was the shortcut pointing to?

carlosperate, 2021-02-17:
> I've had this problem with students where the Desktop-shortcut wasn't created, or where they had deleted it. It's a messy process to recreate that shortcut, where I had to find another Windows [...]

This is a good point. I guess even a simple bat file could already be useful in this case.
@tmontes is this something you've considered in pup?

EvF, 2021-02-19:
Same problem here. Reading the comments above I can see this is because the installer did not create a desktop icon. Nothing added to Windows start menu either.
So.. I go looking in the c:\ProgramFiles\Mu" directory and the only launchable thing there looks like Mu.launch.pyw  and when I run that I get the error message about "Requested python versions (3.6) isnot installed. I do however have Python 3.9.1 installed.

So... what should I run?  What sort of shortcut should I create on my desktop?

EvF, 2021-03-12:
Hello, anyone?
Can someone just give me an idea how to create this missing shortcut?
What should it be running if not Mu.launch.pyw ?
I tried installing on two other Win10 x64 PCs with same result; None of them get a Desktop shortcut.

carlosperate, 2021-03-12:
Can you find a shortcut in the start menu?

EvF, 2021-03-12:
No, there is no sign of it in the start menu; and typing "mu" into the start menu search box doesn't find it either.

EvF, 2021-03-12:
Oh - it does seem to have a tile in the Windows 10 Tiles (ie. Windows Metro) interface.
I hadn't thought to go looking there, I never use that.

So I can copy the Mu shortcut from C:\ProgramData\Microsoft\Windows\Start Menu\Programs
onto my desktop. That works.

carlosperate, 2021-03-12:
Glad you were able to resolve it!

What was the name of the shortcut? Was it just "Mu"? It should have appeared in the start menu search when typing "mu", but that search has been getting worse in Windows 10.

EvF, 2021-03-13:
It maybe because I have "Classic Shell" installed on my Windows 10. It was does not appearing in the Windows-7 style start menu or being found by the "Search programs and files" box on the start menu.
It does however get found if I type "mu" in the standard Windows 10 task-bar search box (I hadn't thought to try that).

I see the shortcut is using Mu's own python install rather than my Python3.9.2
"C:\Program Files\Mu\Python\pythonw.exe" "C:\Program Files\Mu\Mu.launch.pyw"

carlosperate, 2021-04-13:
@Clerudx @mercatian @EvF We've released a new beta version using a different installer. Would you be able to install the latest version to see if you still experience similar problems?
https://github.com/mu-editor/mu/releases

Thank you again for all the testing!

EvF, 2021-04-14:
>Would you be able to install the latest version to see if you still experience similar problems?

Hi Carlos

I just tried Mu-Editor-Win64-1.1.0b3.msi  (after first uninstalling).
It's the same for me. No desktop icon, and nothing found from Win7 style "All Programs" menu.

I was able to find it by typing Mu into the Win10 task bar search box. Alternatively, I was also able to find it on the Windows 10 start Menu (Metro interface) where a right-click context menu will allow me to "pin to Start" or "pin to taskbar". From there I could copy it and paste a shortcut onto my desktop that I could then drag anywhere else I needed it (in my case, into a "Dev Tools" folder on my desktop).

It's the same on my Win10 Laptop. All my PC's are running "Open Shell" (a Win7 start menu, used to be called "Classic Shell" for Windows 10...   https://open-shell.github.io/Open-Shell-Menu    ).
I mainly use this because I find the Win10 Metro interface very irritating.

carlosperate, 2021-04-16:
Thanks @EvF, but once you found the Mu shortcut it all worked correctly, right?
We are tracking the changes to the shortcut here: https://github.com/mu-editor/mu/issues/1282

EvF, 2021-04-16:
Yes, it worked correctly.
It only failed in the earlier examples because I was launching Mu.launch.pyw directly (double click) so it was running with the default python (which in my case was v3.9.1)... hence the error.
The official shortcut (which I failed to find before) was launching it with it's own v3.6 instead so was OK,

carlosperate, 2021-04-16:
Great, thanks for all the testing!

@Clerudx if this is fixed for you as well we could close this issue.

carlosperate, 2021-08-11:
I'll close this issue as resolved, but if anybody is still having this problem please feel free to reopen!

Thanks everyone!

closed 2021-08-11 by carlosperate