mu-editor/mu #1149: flash error

Denforth5, 2020-11-20:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2020-11-27:
Hi @Denforth5,

The error checker built into Mu is good, but won't be able to catch all possible errors. What was scrolling on the micro:bit screen?

dybber, 2021-02-01:
Hi @Denforth5, we haven't heard back from you, could you supply us with more information about the issue. Such as:

 - error shown on the microbit
 - code you used
 - log file

Denforth5, 2021-02-02:
I am sorry Dennis cannot make a reply at present - has been in Watford
Hospital for quite a while and is home now, self isolating for a while.  
Once he is more mobile and stronger, he can respond, but I will ask him your
questions and send his answers if possible.

Audrey Pitchforth (Dennis's wife)
 
 
 
 
-------Original Message-------
 
From: Martin Dybdal
Date: 01/02/2021 22:03:45
To: mu-editor/mu
Cc: Denforth5;  Mention
Subject: Re: [mu-editor/mu] flash error (#1149)
 
Hi @Denforth5, we haven't heard back from you, could you supply us with more
information about the issue. Such as:
error shown on the microbit 
code you used 
log file

ntoll, 2021-02-02:
Please send our best wishes to Dennis and our hopes for his speedy recovery.

carlosperate, 2021-02-02:
Hi Audrey,
I'm really sorry to hear that, please do send our best wishes for a speedy recover to Dennis.

Denforth5, 2021-02-10:
Hi Carlos
At long last I am sat at my computer replying to your earliier e-mail to
Audrey my wife in my abscense. 
I have been in hospital for over a Month and this is my first sitting before
the Screen.

I have been home for for two weeks but have had two weeks solitary
isolatation due to having picked up the
Corona Virus in hospital..

I cannot remember the reason I e-mailed you in the first place.    Perhaps
you would be good enough to
Remind me of the topic  about which  I e-mailed you and we could take I from
there.

Regards
Dennis Pitchforth
_____________________________________________________________________________
_____
 

-------Original Message-------
 
From: Carlos Pereira Atencio
Date: 02/02/2021 15:53:17
To: mu-editor/mu
Cc: Denforth5;  Mention
Subject: Re: [mu-editor/mu] flash error (#1149)
 
Hi Audrey,
I'm really sorry to hear that, please do send our best wishes for a speedy
recover to Dennis.

carlosperate, 2021-02-10:
Hi Dennis,

This email thread originates from a GitHub issue report ( https://github.com/mu-editor/mu/issues/1149 ) in the Mu Editor ( https://codewith.mu ) GitHub repository. Even though we haven't really met outside this, I'm really glad to hear you are doing better, it has brighten my day to hear the good news.

In case you are not aware, even if these replies go via email, this gets posted in a public thread on GitHub, so if you would like to delete it let us know by replying here and I will do that right away.

The original bug report was about some errors that were not captured by the Mu editor and appeared in the micro:bit running the code. If you had the original code that would help us replicate the issue, although it's worth noting that the Mu integrated checker has some limitations and some genuine errors in the code cannot be detected until it is run.

But most importantly, you should concentrate on your recovery, so don't worry about his issue at all, you can always come back to it in the future.

Cheers,
Carlos

ntoll, 2021-02-14:
@Denforth5 best wishes and hoping for a speedy and full recover from everyone involved in Mu.