mu-editor/mu #1152: Dark theme patch needs to check for macOS Catalina

frenoy, 2020-11-23:
**Issue:**
Dark mode doesn't seem to be working as intended on macOS Catalina (10.15.7). I believe this was first observed on Mojave (#552) and recently for Catalina (#1027).

[image]

[image]

**Fix:**
A temporary fix (#567) was used to resolve this for Mojave which checks the OS version (Darwin-18.) This needs to be extended to check for Catalina (Darwin-19.). I don't have access to a machine running Big Sur to check the behaviour there.

The following screenshots were taken after updating line 43 of themes.py (local copy) from:

`return platform.platform().startswith("Darwin-18.")` 

to:

`return platform.platform().startswith("Darwin-18.") or platform.platform().startswith("Darwin-19.")`

[image]

[image]

P.S. Thank you for the great work! Please accept my apologies if I haven't followed proper etiquette - I'm pretty new to software development.

carlosperate, 2020-11-27:
Thanks for the report @frenoy, have you tried the v1.0.3 release? I think this issue was solved in that release.

frenoy, 2020-11-27:
@carlosperate Thanks for responding. I was using v1.1.0 alpha.2 which I thought was the latest. Just installed v1.0.3 and I can confirm that the issue doesn't appear there.

Closing this now.

closed 2020-11-27 by frenoy