mu-editor/mu #1157: Checker raises error in all unused imports from a "import *"

carlosperate, 2020-11-27:
I've fairly sure this wasn't an issue before? If that's the case, then this was likely introduced in a recent update of pyflakes.

[image]

While this a fair warning, most new users will still want to do `from microbit import *`, and these errors are not very helpful.

```
$ pip list 
Package                       Version
----------------------------- ----------
alabaster                     0.7.12
appdirs                       1.4.4
appnope                       0.1.0
arrow                         0.17.0
attrs                         20.3.0
Babel                         2.9.0
backcall                      0.2.0
binaryornot                   0.4.4
black                         20.8b1
bleach                        3.2.1
boto3                         1.15.18
botocore                      1.18.18
briefcase                     0.2.9
certifi                       2020.11.8
chardet                       3.0.4
click                         7.1.2
colorama                      0.4.4
cookiecutter                  1.7.2
coverage                      5.3
dataclasses                   0.8
decorator                     4.4.2
dmgbuild                      1.4.2
docutils                      0.15.2
ds-store                      1.3.0
flake8                        3.8.4
Flask                         1.1.2
idna                          2.10
imagesize                     1.2.0
importlib-metadata            3.1.0
iniconfig                     1.1.1
ipykernel                     5.3.4
ipython                       7.16.1
ipython-genutils              0.2.0
itsdangerous                  1.1.0
jedi                          0.17.2
Jinja2                        2.11.2
jinja2-time                   0.2.0
jmespath                      0.10.0
jupyter-client                6.1.7
jupyter-core                  4.7.0
keyring                       21.5.0
mac-alias                     2.1.0
MarkupSafe                    1.1.1
mccabe                        0.6.1
mu-editor                     1.1.0a2
mypy-extensions               0.4.3
nudatus                       0.0.4
numpy                         1.19.4
packaging                     20.4
parso                         0.7.1
pathspec                      0.8.1
pexpect                       4.8.0
pgzero                        1.2
pickleshare                   0.7.5
pip                           20.2.4
pipdeptree                    1.0.0
pkginfo                       1.6.1
pluggy                        0.13.1
poyo                          0.5.0
prompt-toolkit                3.0.8
ptyprocess                    0.6.0
py                            1.9.0
pycodestyle                   2.6.0
pyflakes                      2.2.0
pygame                        1.9.6
Pygments                      2.7.2
pyparsing                     2.4.7
PyQt5                         5.13.2
PyQt5-sip                     12.8.1
PyQtChart                     5.13.1
pyserial                      3.4
pytest                        6.1.2
pytest-cov                    2.10.1
pytest-faulthandler           2.0.1
pytest-random-order           1.0.4
python-dateutil               2.8.0
python-slugify                4.0.1
pytz                          2020.4
pyzmq                         20.0.0
QScintilla                    2.11.3
qtconsole                     4.7.4
QtPy                          1.9.0
readme-renderer               28.0
regex                         2020.11.13
requests                      2.25.0
requests-toolbelt             0.9.1
rfc3986                       1.4.0
s3transfer                    0.3.3
semver                        2.13.0
setuptools                    50.3.2
six                           1.15.0
snowballstemmer               2.0.0
Sphinx                        3.3.1
sphinxcontrib-applehelp       1.0.2
sphinxcontrib-devhelp         1.0.2
sphinxcontrib-htmlhelp        1.0.3
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.3
sphinxcontrib-serializinghtml 1.1.4
text-unidecode                1.3
toml                          0.10.2
tornado                       6.1
tqdm                          4.54.0
traitlets                     4.3.3
twine                         3.2.0
typed-ast                     1.4.1
typing-extensions             3.7.4.3
urllib3                       1.23
voc                           0.1.6
wcwidth                       0.2.5
webencodings                  0.5.1
Werkzeug                      1.0.1
wheel                         0.35.1
zipp                          3.4.0
```

dybber, 2020-12-06:
I've added a fix in PR #1161, it was simply a regex that needed to be changed a bit. 
The current solution to this seems *flaky* (pun intended).

It's also not very clean to do this only for the microbit module, and not for anything else. At least, it would be nice if we could move this into the Micro:bit mode at some point, or allow other modes access to the same functionality, but let's wait and see when/if it becomes necessary.

dybber, 2021-01-26:
Solved by #1161, closing.

closed 2021-01-26 by dybber