mu-editor/mu #1158: INstalling fails - Ubuntu 18.04

TheFwGuy, 2020-11-29:
Hi !
I'm trying to install Mu editor on my machine with Ubuntu 18.04 - 64 bit.
I did install multiple version of Python, from 2.7 to 3.8 and all fails.
The error is the same no matter the Python version used and I did use Python 3.8.6.

To install I'm using pip3 :

pip3 install mu-editor --no-cache

```
    ERROR: Command errored out with exit status 1:
     command: /home/steve/.linuxbrew/opt/python@3.8/bin/python3.8 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-2svq53kh/matplotlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-2svq53kh/matplotlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-okne5j3r
         cwd: /tmp/pip-install-2svq53kh/matplotlib/
    Complete output (61 lines):
    ============================================================================
    Edit setup.cfg to change the build options
    
    BUILDING MATPLOTLIB
                matplotlib: yes [2.2.2]
                    python: yes [3.8.6 (default, Oct 22 2020, 11:35:12)  [GCC
                            5.5.0 20171010]]
                  platform: yes [linux]
    
    REQUIRED DEPENDENCIES AND EXTENSIONS
                     numpy: yes [not found. pip may install it below.]
          install_requires: yes [handled by setuptools]
                    libagg: yes [pkg-config information for 'libagg' could not
                            be found. Using local copy.]
                  freetype: no  [The C/C++ header for freetype2 (ft2build.h)
                            could not be found.  You may need to install the
                            development package.]
                       png: yes [version 1.6.34]
                     qhull: yes [pkg-config information for 'libqhull' could not
                            be found. Using local copy.]
    
    OPTIONAL SUBPACKAGES
               sample_data: yes [installing]
                  toolkits: yes [installing]
                     tests: no  [skipping due to configuration]
            toolkits_tests: no  [skipping due to configuration]
    
    OPTIONAL BACKEND EXTENSIONS
                    macosx: no  [Mac OS-X only]
                    qt5agg: no  [PySide2 not found; PyQt5 not found]
                    qt4agg: no  [PySide not found; PyQt4 not found]
                   gtk3agg: no  [Requires pygobject to be installed.]
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-2svq53kh/matplotlib/setup.py", line 197, in <module>
        msg = pkg.install_help_msg()
      File "/tmp/pip-install-2svq53kh/matplotlib/setupext.py", line 592, in install_help_msg
        release = platform.linux_distribution()[0].lower()
    AttributeError: module 'platform' has no attribute 'linux_distribution'
                 gtk3cairo: no  [Requires cairocffi or pycairo to be installed.]
                    gtkagg: no  [Requires pygtk]
                     tkagg: yes [installing; run-time loading from Python Tcl /
                            Tk]
                     wxagg: no  [requires wxPython]
                       gtk: no  [Requires pygtk]
                       agg: yes [installing]
                     cairo: no  [cairocffi or pycairo not found]
                 windowing: no  [Microsoft Windows only]
    
    OPTIONAL LATEX DEPENDENCIES
                    dvipng: no
               ghostscript: yes [version 9.26]
                     latex: no
                   pdftops: yes [version 0.62.0]
    
    OPTIONAL PACKAGE DATA
                      dlls: no  [skipping due to configuration]
    
    ============================================================================
                            * The following required packages can not be built:
                            * freetype
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

```
Any suggestion about what I could do ?
Thanks

devdanzin, 2020-11-30:
Seems like you should install the Ubuntu freetype dev package?

abixadamj, 2020-11-30:
`sudo apt install python3-freetype`

carlosperate, 2020-11-30:
Due to some pinned dependencies Mu currently doesn't work in Python 3.8. This should be fixed for the next release, but at the moment the easiest workaround would be to use Python 3.6 or 3.7

In Ubuntu something like the deadsnakes PPA can be used:

```bash
$ sudo add-apt-repository ppa:deadsnakes/ppa
$ sudo apt-get update
$ sudo apt-get install python3.7
# Ideally install in a virtualenv, this will create it in the current wording directory, in a folder called "mu_venv"
$ python3.7 -m venv mu_venv
$ source mu_venv/bin/activate
$ python -m pip install mu-editor
$ python -m mu
```

I know having to add a PPA is not great, but these steps should be fairly pain-free.

carlosperate, 2021-03-24:
Thanks everyone! The latest beta release has updated the dependencies to be compatible with Python 3.5 to 3.8, so it should work now in this cases.

I'll close this issue, but if anybody is still having issue please feel free to reopen it or to open a new one.

closed 2021-03-24 by carlosperate