mu-editor/mu #1160: Support .pyw files

tjguk, 2020-12-01:
We don't currently support .pyw files when opening a file loader

devdanzin, 2020-12-02:
The change to make it work is trivial indeed, seen below. But some choices aren't as clear to an outsider like me, such as:
* should we have a single test for a .pyw file, or somehow run (a subset of?) normal .py tests for .pyw files too?
* should we hardcode .pyw along with .py (which arguably is the base file extension supported) or put it into Python3 mode (which maybe could also grow, say, .pxi someday) ?
* what about saving a new file as .pyw, is it in scope (might lead to file saved in new version not opening in old version)?

```diff

diff --git a/mu/logic.py b/mu/logic.py
index a96a36f..675e8d3 100644
--- a/mu/logic.py
+++ b/mu/logic.py
@@ -1105,7 +1105,7 @@ class Editor(QObject):
                 return
         name, text, newline, file_mode = None, None, None, None
         try:
-            if path.lower().endswith(".py"):
+            if path.lower().endswith(".py") or path.lower().endswith(".pyw"):
                 # Open the file, read the textual content and set the name as
                 # the path to the file.
                 try:
@@ -1204,7 +1204,7 @@ class Editor(QObject):
         extracts a Python script from a hex file.
         """
         # Get all supported extensions from the different modes
-        extensions = ["py"]
+        extensions = ["py", "pyw"]
         for mode_name, mode in self.modes.items():
             if mode.file_extensions:
                 extensions += mode.file_extensions
```

devdanzin, 2020-12-08:
Tim, adding .pyw as a file extension to Python3 mode is not trivial because some of what we'd need to handle it is directly implemented in logic.py instead, so the basic editor always works with Python files. Maybe adding it to places like mentioned above is good enough, but I can work on moving logic into Python mode if it's the better path.

Tidy, check and serving from web mode also do a check for `endswith(".py")`, so we should add .pyw there too.

Saving with any extension is already supported, so that's covered. Starting the REPL and importing from .pyw files works. 

Should I submit an exploratory PR from the changes I tried?

tjguk, 2020-12-08:
Thanks for giving this more consideration than I obviously have, @devdanzin  :)

I would happily look at a tentative PR moving the file-extension logic into a mode -- although I think there are several plans in the air around making modes more... well, modular / pluggable etc. So if you want to move things around, don't go *too* far. ;)

dybber, 2020-12-08:
I'm not sure why it would be necessary to move it into the modes? Do you mean that modes supply a list of supported extension `self.file_extensions = ["py", "pyw"]`?

carlosperate, 2020-12-08:
Yeah, I guess the pyw extension is pretty much a Windows specific feature and doesn't make a lot of sense for MicroPython/CircuitPython boards (would that file even be recognised as a Python script?).

Adding file extensions per mode seems sensible, but at the same time it means having to implement a way to enforce it, which will be more involved. Or at the very least a way for some modes to reject or rename flashing pyw files if they won't work on the board.

If we were to ever implement a system to have different modes per tab, as it has been discussed in a couple of other issues, then this is something that could go as part of that. But outside of that it might be better to do something a lot simpler for v1.1.

devdanzin, 2020-12-09:
That's really good news, because the exploratory changes I made were in the 'add .pyw where we have .py' style :)

So I'll leave thinking about what could/should be moved  into Python mode for later. WIP simple take in #1169.

dybber, 2020-12-10:
Thank you @devdanzin for the PR on this!

closed 2020-12-10 by dybber