mu-editor/mu #1162: When will Mu be compatible with micro:bit V2.0

chenjiegd, 2020-12-07:
I tried to program the micro:bit V2 with Mu 1.0.3, but this prompted an error in the micro:bit. I hope that developers can adapt to the micro:bit V2.0 as soon as possible.

carlosperate, 2020-12-08:
Is that when pressing the "Flash" button?

If you have MicroPython running already the Flash button should work, but I think there is a problem in MicroPython/CODAL (https://github.com/microbit-foundation/micropython-microbit-v2/issues/35 https://github.com/lancaster-university/codal-microbit-v2/issues/45) that might be interfering with the Flash process.

If you open the REPL once or twice (until the REPL works) and then press the Flash button, does that work?

If you could attach a Mu log here that would be really useful as well.

AllegroGavin, 2020-12-10:
A few experiences with a V2 and Mu 1.1.0.alpha.2 on Windows 10:

1) with micropython 2.0.0.beta.2 on the MBV2, If I have Mu running and connect the V2 to USB, Mu is unable to talk to it (Files gives an error dialog, Repl goes into a weird state) until I press the reset button on the V2 after which it can do those things without issue.

2) In the Mu BBC micro:bit Settings I set micropython-microbit-v2.0.0-beta.2.hex as the override runtime. Pressing Flash appears to download the entire .hex (I believe the beta.2 is a universal hex with both V1 and V2 micropythons in it?) and then reports an error:

[image]

However after resetting the V2 the micropython got flashed but there is no main.py on the micro:bit.

I note in the Mu log:

2020-12-10 11:39:40,726 - mu.modes.microbit:266(flash) INFO: Board MicroPython: 2.0.0-beta.2
2020-12-10 11:39:40,727 - mu.modes.microbit:268(flash) INFO: Mu MicroPython: 1.0.1
2020-12-10 11:39:40,727 - mu.modes.microbit:296(flash) INFO: Flashing new MicroPython runtime onto device
2020-12-10 11:39:53,998 - mu.modes.microbit:420(copy_main) INFO: Copying main.py onto device
2020-12-10 11:39:53,998 - mu.modes.microbit:430(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'# Write your code here :-)\\ndef hello():\\n    print("Hello V2 worl\')', 'f(b\'d!")\\n\\ndef bench(n):\\n    x=7.0\\n    from microbit import running_t\')', 'f(b"ime\\n    print(\'Running...\')\\n    s=running_time()\\n    for i in ra")', 'f(b"nge(n):\\n        x *= 1.3\\n    print(\'Elapsed:\', running_time()-s)")', 'f(b\' # for 100,000 on V1.5: 7836 V2: 2093\\n\\n\\nprint("Hello V2 World!")\')', "f(b'\\n')", 'fd.close()']
2020-12-10 11:39:55,409 - mu.modes.microbit:447(flash_failed) ERROR: Could not enter raw REPL.
2020-12-10 11:39:55,410 - mu.interface.main:763(show_message) DEBUG: There was a problem flashing the micro:bit.
2020-12-10 11:39:55,410 - mu.interface.main:764(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.

In the first couple lines of that it seems to be trying to compare the MP versions and it always downloads the full MP .hex if the version does not match what it thinks it should be. There should be some way to tell Mu "Just assume it's a good version" so it's not constantly re-flashing the whole device.

3) The repl is broken with respect to <backspace> when talking to a V2. See:

https://github.com/microbit-foundation/micropython-microbit-v2/issues/39

where I reported it as possibly an MP 2 issue, but other clients are not seeing the issue so maybe it's a Mu thing?

4) A workaround to the flashing issues is to get the new micropython hex onto the device either through the above abortive Mu download or by dropping it onto the device in file manager, then after a reset you can use the Files manager in Mu to move programs onto the V2 and immediately import them etc. through the repl and that works fine with no issues (apart from the intermittent backspace problem noted above).

carlosperate, 2020-12-21:
>With micropython 2.0.0.beta.2 on the MBV2, If I have Mu running and connect the V2 to USB, Mu is unable to talk to it (Files gives an error dialog, Repl goes into a weird state) until I press the reset button on the V2 after which it can do those things without issue.

Yes, this was due to an issue with serial, which will be fix in the next release that will be done today or tomorrow.

The problem is that there was some unexpected additional serial data sent from the micro:bit on start-up, so Mu/microFS was unable to enter the raw REPL and send the files.

Another workaround to this issue was to open the REPL in Mu a couple of times, this normally was enough to clear the UART buffers.

> In the Mu BBC micro:bit Settings I set micropython-microbit-v2.0.0-beta.2.hex as the override runtime. Pressing Flash appears to download the entire .hex (I believe the beta.2 is a universal hex with both V1 and V2 micropythons in it?) and then reports an error:

This is due to the same issue, this method will work with the new MicroPython hex for V2. After flashing the MicroPython hex then Mu/microFS tries to send the file via serial and fails as being unable to enter the raw REPL.

carlosperate, 2020-12-21:
Fresh from the oven: https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.3

carlosperate, 2021-01-31:
Implemented in https://github.com/mu-editor/mu/pull/1257 🎉

closed 2021-01-31 by carlosperate

linguistdan, 2022-02-23:
> I tried to program the micro:bit V2 with Mu 1.0.3, but this prompted an error in the micro:bit. I hope that developers can adapt to the micro:bit V2.0 as soon as possible.

Hello, is it ok for me ask what happened to this question? We just got a Micro:bit v2 and are running Mu 1.0.3 on a Rasberry Pi but it tells us to update to Mu 1.1.1 Typing sudo apt-get update and sudo apt-get dist-upgrade still only gets us Mu 1.0.3 - how can we download 1.1.1 to the Pi? Thanks, Daniel.

carlosperate, 2022-02-23:
Hi @linguistdan,

The Debian package that then feeds the Raspbian package is not managed by the Mu team directly, and it is maintained by volunteer packagers like @knowledgejunkie (which we thank for all the hard work keeping those packages running!).

It will take some time for the latest release to be updated in Debian, and probably a bit longer to feed down to Raspbia/Raspberry Pi OS, so the easiest way for Raspberry Pi user right now might be to follow these instructions: https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi

linguistdan, 2022-02-23:
> Hi @linguistdan,
> 
> The Debian package that then feeds the Raspbian package is not managed by the Mu team directly, and it is maintained by volunteer packagers like @knowledgejunkie (which we thank for all the hard work keeping those packages running!).
> 
> It will take some time for the latest release to be updated in Debian, and probably a bit longer to feed down to Raspbian/Raspberry Pi OS, so the easiest way for Raspberry Pi user right now might be to follow these instructions: https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi

Oh thanks so much, really helpful - I'll try the instructions at that link.