mu-editor/mu #1167: Mu Editor can't flash the microbit under Big Sur

Villafruela, 2020-12-08:
Hi,

Under Mac OS Big Sur (v11.0.1), I've installed Mu editor 1.1.alpha, allowed it to launch and have disk access.
It launches ok, but when i try to flash my microbit with a simple program:
1. i got a popup window with a message "A problem occured while flashing the Micro:bit, do not disconnect the peripheral while flashing is not finished. Look for the log fore more information" (Translated from french to english, so this may not be the exact terms in the english popup).
2. i looked at the logs : 
`2020-12-08 11:52:27,257 - mu.logic:1192(show_admin) INFO: No admin settings changed.
2020-12-08 11:52:29,592 - mu.modes.microbit:154(flash) INFO: Preparing to flash script.
2020-12-08 11:52:29,593 - mu.modes.microbit:164(flash) DEBUG: Python script:
2020-12-08 11:52:29,593 - mu.modes.microbit:165(flash) DEBUG: b'# Ecrit ton programme ici ;-)\nfrom microbit import *\n\npin0.write_digital(1)\n'
2020-12-08 11:52:29,611 - mu.modes.microbit:212(flash) INFO: Path to micro:bit: /Volumes/MICROBIT
2020-12-08 11:52:29,615 - mu.modes.base:261(find_device) INFO: Found device on port: cu.usbmodem141302
2020-12-08 11:52:29,615 - mu.modes.base:262(find_device) INFO: Serial number: 9901000052324e450076800a0000004b0000000097969901
2020-12-08 11:52:29,615 - mu.modes.microbit:217(flash) INFO: Serial port: /dev/cu.usbmodem141302
2020-12-08 11:52:29,616 - mu.modes.microbit:218(flash) INFO: Device serial number: 9901000052324e450076800a0000004b0000000097969901
2020-12-08 11:52:29,616 - mu.modes.microbit:243(flash) INFO: Checking target device.
2020-12-08 11:52:30,166 - mu.modes.microbit:247(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.1', 'version': 'micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2020-12-08 11:52:30,167 - mu.modes.microbit:266(flash) INFO: Board MicroPython: 1.0.1
2020-12-08 11:52:30,167 - mu.modes.microbit:268(flash) INFO: Mu MicroPython: 1.0.1
2020-12-08 11:52:30,167 - mu.modes.microbit:420(copy_main) INFO: Copying main.py onto device
2020-12-08 11:52:30,167 - mu.modes.microbit:430(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'# Ecrit ton programme ici ;-)\\nfrom microbit import *\\n\\npin0.write')", "f(b'_digital(1)\\n')", 'fd.close()']
2020-12-08 11:52:31,209 - mu.modes.microbit:365(flash) WARNING: Could not copy file to device.
2020-12-08 11:52:31,209 - mu.modes.microbit:366(flash) ERROR: Could not enter raw REPL.
2020-12-08 11:52:31,210 - mu.modes.microbit:367(flash) INFO: Falling back to old-style flashing.
2020-12-08 11:52:31,216 - mu.modes.microbit:75(run) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2020-12-08 11:52:31,270 - mu.modes.microbit:447(flash_failed) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2020-12-08 11:52:31,277 - mu.interface.main:763(show_message) DEBUG: Un problème est survenu pendant le flashage de la micro:bit.
2020-12-08 11:52:31,277 - mu.interface.main:764(show_message) DEBUG: Ne pas déconnecter le périphériques tant que le flashage n'est pas terminé. Regarde les logs pour plus d'informations.
2020-12-08 11:52:35,103 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/naimericvillafruela/Library/Logs/mu/mu.log
2020-12-08 11:52:35,103 - mu.logic:163(installed_packages) INFO: Packages found: []`

I noticed the "2020-12-08 11:52:31,209 - mu.modes.microbit:365(flash) WARNING: Could not copy file to device." line.
I looks like the mu-editor doesn't have the rights to write on the microbit.

Any idea how to fix this ?

Thanks for your help!

carlosperate, 2020-12-08:
This is probably related to https://github.com/mu-editor/mu/issues/998, could you try the workarounds listed there to see if it works?

Villafruela, 2020-12-08:
You're right it's the same symptoms. (Sorry i didn't tought to watch on over os...)
but prompting `sudo -a open mu-editor.app didn't work on Big Sur (sudo doesn't have a -a option). `sudo open mu-editor.app` launches mu-editor but flashing the microbit still doesn't work.

In the meanwhile i use VSC+uflash but i use mu-editor with kids. They like it a lot ! Very easy to use ! The team as done a wonderfull work !

Hope it will be fixed soon...

carlosperate, 2020-12-08:
As a workaround, if you are familiar with Python you could install Mu from source code in a virtual environment (Python 3.6 or 3.7 at the moment) and that should work as well: https://mu.readthedocs.io/en/latest/setup.html

carlosperate, 2021-03-03:
Thanks for the report @Villafruela, I believe this is already fixed with the new pup packaging system, so the next release should work.
As this issue is already covered in #998, I'll close this one.

closed 2021-03-03 by carlosperate