mu-editor/mu #1168: Importing Matplotlib on mu 1.1.0 Alpha.2

ged99, 2020-12-08:
If you are reporting a bug, we would like to know:

* What you were trying to do: 

Trying to "import matplotlib"

* What steps you took to make this happen,

Installed it at the terminal level. This seems fine.

* What you expected to happen,

I expect my code to pick up the library

* What actually happened,
ModuleNotFoundError: No module named 'matplotlib'

* Why this difference is problematic (it may not be a bug!),
I need the module!

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

mu 1.1.0 Alpha 2 
MacBook Pro with Big Sur

(I need Alpha as the main release fails to run on Big Sur.)

This used to work under the old release. I can also get mu to find the module on my Windows m/c. I have imported matplotib thru the terminal using:

pip3 install matplotlib

This worked. If I execute Python in the terminal it can find the module. But the mu version does not. I have attached the screen shots for you:
[image]

devdanzin, 2020-12-09:
Hello, thank you for the report!

The Python version in your terminal is 3.9.0, while the one displayed in Mu's REPL is probably 3.6.x, because Mu is shipping its own interpreter. 

If you click the gear in the lower right corner, you'll get a dialog with a tab called 'Third Party Packages'. In it, you can add matplotlib and it will try to install the package for you. Worked for me on Windows, please report what results you get.

ged99, 2020-12-09:
Thanks for your help @devdanzin . Not seen that dialog tab before. Anyway it didn't work. Nothing to do with matpoltlib, but something to do SSL certificates. Any help would be appreciated.
[image]

dybber, 2020-12-09:
This is a known issue unfortunately, see #910, and a fix is in progress. I'm not sure there's any good workarounds.

carlosperate, 2020-12-09:
A possible workaround is to use brew to install openssl, but it might need the old v1.0, I think right now brew installs v1.1.

A tiny bit more info in https://github.com/mu-editor/mu/issues/910#issuecomment-558845231.

ged99, 2020-12-09:
@carlosperate - the only reason I'm using Alpha is that Mu 1.1 doesn't work on Big Sur. No sign of a fix yet, although it is being discussed.

carlosperate, 2020-12-09:
Right, the brew workaround would work with the alpha as well. Basically the Python packaged with Mu expects to find OpenSSL as a shared library available on the system (this was a bug, not by design), and if it is brew installed it should be able to find it and run as expected.

devdanzin, 2020-12-09:
The issue is that Mu cannot talk to pypi.org without SSL. But if you manually download the wheels for the packages you want to install, installing locally should work.

Matplotlib has these requirements:

- Requires-Dist: [cycler ](https://pypi.org/project/cycler/#files)(>=0.10)
- Requires-Dist: [kiwisolver ](https://pypi.org/project/kiwisolver/#files)(>=1.0.1)
- Requires-Dist: [numpy ](https://pypi.org/project/numpy/#files)(>=1.15)
- Requires-Dist: [pillow ](https://pypi.org/project/pillow/#files)(>=6.2.0)
- Requires-Dist: [pyparsing ](https://pypi.org/project/pyparsing/#files)(!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3)
- Requires-Dist: [python-dateutil ](https://pypi.org/project/python-dateutil/#files) (>=2.1)

If you download those as wheels, plus [matplotlib ](https://pypi.org/project/matplotlib/#files) itself, you can use [the script below](https://gist.github.com/devdanzin/be6eeab95bf592bc630cc66b70cec5b7) (save as e.g. install_mu.py) from inside Mu and it'll install to the right Python lib. It's probable that a missing sub-dependency will be identified when pip tries to install these, so the workflow is:

1. Open script in Mu
2. Open REPL,`from install_mu import install_wheels`
3. `install_wheels(PATH_TO_WHERE_YOU_DOWNLOADED)`
4. Read pip's error messages
5. If there's some missing dependency, download the wheel and goto 2
6. You might have matplotlib working

BTW, if anyone wants to test workarounds for this from Windows, renaming _ssl.pyd gets you the same pip error message.

```python
from subprocess import run, PIPE
import sys
import os, os.path
import appdirs

MU_PYTHON = sys.executable
DATA_DIR = appdirs.user_data_dir(appname="mu", appauthor="python")
MODULE_DIR = os.path.join(DATA_DIR, "site-packages")  #
WHEELS_DIR = os.path.expanduser("~/wheels")
ARGS = ["-m", "pip", "install", "--no-index", "--target", MODULE_DIR]

def install_wheels(path=WHEELS_DIR, extra_args=None):
    if extra_args is None:
        extra_args = []
    elif type(extra_args) == str:
        extra_args = extra_args.split()
    print("Looking for wheels in", path)
    files = os.listdir(path)

    wheel_names = [name for name in files if name.lower().endswith(".whl")]
    print("Found these wheels to install:\n\t" + "\n\t".join(wheel_names))
    wheel_names = [os.path.join(path, name) for name in wheel_names]

    do_the_pip = run(
        [MU_PYTHON] + ARGS + extra_args + wheel_names, stdout=PIPE, stderr=PIPE
    )

    print(do_the_pip.stdout.decode())
    print(do_the_pip.stderr.decode())

```

carlosperate, 2021-03-16:
The issue with pip not being able to download packages due to the missing openSSL library has been fixed in the latest release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 🎉 

I'll close this as resolved, but please do reopen it if you are still having this issue with the v1.1.0-beta.2.

closed 2021-03-16 by carlosperate